freifunk-gluon/gluon #2656: AX88179 (USB Network Adapter) stops working after a short while

herbetom, 2022-10-05:
### Bug report

**What is the problem?**

Since v2022.1 the "ASIX Elec. Corp. AX88179" attached to a node via USB only works for a short while and then there is a Stack Trace and after that the network connectiviy on that port stopps working.

```
  | [  611.185669] ------------[ cut here ]------------
-- | --
  | [  611.199522] NETDEV WATCHDOG: eth1 (ax88179_178a): transmit queue 0 timed out
  | [  611.220700] WARNING: CPU: 0 PID: 0 at dev_watchdog+0x1e9/0x1f0
  | [  611.238173] Modules linked in: ath9k ath9k_common mt76x0e mt76x0_common iptable_nat batman_adv ath9k_hw ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT smsc95xx rndis_host pl2303 nf_nat nf_conntrack mt76x2e mt76x2_common mt76x02_lib mt7615e mt7615_common mt7603e mt76_connac_lib mt76 mcs7830 mac80211 ipt_REJECT ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG via_velocity via_rhine usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm smsc sky2 sis900 r8169 r8152 pegasus pcnet32 nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 ne2k_pci iptable_mangle iptable_filter ipheth ip_tables hso forcedeth ebtables ebt_vlan ebt_stp
  | [  611.238356]  ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 e1000e e100 crc_ccitt compat bnx2 ax88796b 8390 8139too 8139cp sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_ipt act_csum libcrc32c sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact i2c_dev ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 3c59x e1000 dummy l2tp_eth l2tp_netlink l2tp_core ip_tunnel veth tun vfat fat ntfs nls_utf8 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp852 nls_cp850 nls_cp437 nls_cp1250 natsemi vxlan udp_tunnel ip6_udp_tunnel sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac uas leds_gpio xhci_plat_hcd sp5100_tco fsl_mph_dr_of ehci_platform ehci_fsl igb act_police cls_basic sch_tbf sch_ingress
  | [  611.499674]  gpio_button_hotplug button_hotplug gpio_amd_fch tg3 ptp realtek pps_core mii
  | [  611.783204] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.136 #0
  | [  611.801441] Hardware name: FUJITSU                          FUTRO S550-2                  /D2963-A1, BIOS 6.00 R1.24.2963.A1               02/17/2011
  | [  611.841498] EIP: dev_watchdog+0x1e9/0x1f0
  | [  611.853506] Code: 45 e8 e8 ba 41 16 00 8b 77 14 eb 9f 8b 75 e8 c6 05 10 cb cd c1 01 89 f0 e8 34 f3 fb ff 53 50 56 68 6c 46 bc c1 e8 68 f2 11 00 <0f> 0b 83 c4 10 eb c3 55 89 e5 57 89 cf 56 89 c6 53 89 d3 83 ec 0c
  | [  611.909673] EAX: 00000040 EBX: 00000000 ECX: f6eeff64 EDX: f6eeb144
  | [  611.928431] ESI: c3076000 EDI: c30762e8 EBP: c2015f68 ESP: c2015f3c
  | [  611.947188] DS: 007b ES: 007b FS: 00d8 GS: 00e0 SS: 0068 EFLAGS: 00210292
  | [  611.967502] CR0: 80050033 CR2: b7f63c40 CR3: 04514ae0 CR4: 000006f0
  | [  611.986256] Call Trace:
  | [  611.993581]  <SOFTIRQ>
  | [  612.000659]  ? pfifo_fast_enqueue+0x140/0x140
  | [  612.013710]  call_timer_fn.constprop.0+0x13/0x90
  | [  612.027533]  ? pfifo_fast_enqueue+0x140/0x140
  | [  612.040576]  run_timer_softirq+0x223/0x280
  | [  612.052846]  __do_softirq+0x9f/0x1b2
  | [  612.063552]  ? __entry_text_end+0x1/0x1
  | [  612.075038]  call_on_stack+0x40/0x50
  | [  612.085742]  </SOFTIRQ>
  | [  612.093068]  ? irq_exit_rcu+0x57/0xa0
  | [  612.104035]  ? sysvec_apic_timer_interrupt+0x27/0x40
  | [  612.118897]  ? handle_exception+0x10e/0x10e
  | [  612.131424]  ? amd_e400_idle+0x1b/0x50
  | [  612.142649]  ? sysvec_call_function_single+0x40/0x40
  | [  612.157508]  ? amd_e400_idle+0x1b/0x50
  | [  612.168732]  ? arch_cpu_idle+0xd/0x20
  | [  612.179697]  ? default_idle_call+0x2d/0x40
  | [  612.191963]  ? do_idle+0x175/0x1b0
  | [  612.202149]  ? cpu_startup_entry+0x1d/0x20
  | [  612.214416]  ? rest_init+0x90/0x96
  | [  612.224604]  ? arch_call_rest_init+0x8/0x13
  | [  612.237126]  ? start_kernel+0x404/0x416
  | [  612.248610]  ? i386_start_kernel+0x43/0x49
  | [  612.260876]  ? startup_32_smp+0x164/0x168
  | [  612.272882] ---[ end trace e8a4784e58c22df9 ]---

```

**What is the expected behaviour?**
It worked before with [v2021.1.2](https://github.com/freifunk-gluon/gluon/releases/tag/v2021.1.2).

**Gluon Version:**
[v2022.1](https://github.com/freifunk-gluon/gluon/releases/tag/v2022.1)

**Site Configuration:**
https://github.com/Freifunk-Rhein-Neckar/site-ffrn/tree/v2.0.0-20220905

**Custom patches:**

None

blocktrron, 2022-10-05:
There is this commit on the kernel bugzilla:

https://bugzilla.kernel.org/show_bug.cgi?id=216327#c20

The referenced revert is contained on the Linux-Stable branch OpenWrt 22.03 tracks:

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/drivers/net/usb/ax88179_178a.c?h=linux-5.10.y&id=e81046da1d9b2ffe8bb26a70871bebc281bcd06a

However this is contained in the kernel bump prior the release-tag which we missed by a few commits. Can you build from the v2022.1.x branch and check if the problem persists?

herbetom, 2022-10-08:
The Original Report and Stack Trace came from [kdkuschke](https://forum.ffrn.de/u/kdkuschke/summary).

I've now done some digging and found myself a USB Network Adapter with an AX88179 chip.

With the firmware based upon v2022.1 i could observe the issue with my adapter aswell, hoewer i didn't get an Stack Trace.

With a firmware based upon [v2022.1.x (ecd9f8c)](https://github.com/freifunk-gluon/gluon/commit/ecd9f8cfe8cc11d7b52a34858506093bc770e9e6) everything seems to be fine again. 

Thanks @blocktrron for digging this up! :+1:

closed 2022-10-08 by herbetom