freifunk-gluon/gluon #2662: Explain what Gluon is on the readme/readthedocs

RollingStar, 2022-10-10:
The readme says:

> Gluon is a firmware framework to build preconfigured OpenWrt images for public mesh networks.

The docs say:

> Gluon is a modular framework for creating OpenWrt-based firmware images for wireless mesh nodes. Several Freifunk communities in Germany use Gluon as the foundation of their Freifunk firmware.

Then with Google you can find: https://en.wikipedia.org/wiki/Freifunk

> This software, the Freifunk firmware, is based on [OpenWrt](https://en.wikipedia.org/wiki/OpenWrt) and other [free software](https://en.wikipedia.org/wiki/Free_software).

The project looks very cool. I can use my imagination and guess why a "public mesh network" might be valuable, but many projects evangelize their features to potential users and contributors. The problem I have now is that Gluon is marketed as a tool for public mesh networks, but there's no link or text explaining why public mesh matters.

AiyionPrime, 2022-10-11:
@RollingStar Maybe you could open a pullrequest with what info you were missing and we can go on from there together? I see you contributed to git based projects before. Thanks so far!

maurerle, 2023-08-28:
docs label..?

Mir fehlen hier ein paar Sätze in der README und in der `getting_started.rst`, die erkärt, was dieses Gluon jetzt wirklich tut:

Mein Vorschlag hierzu hab ich nachfolgend angehängt, falls wichtige Punkte fehlen, gerne ergänzen:

---

Generally, the firmware connects to a tunnel if a wired uplink is available.
It opens a public wifi which routes all traffic through the tunnel and creates a mesh network with surrounding gluon nodes with the same config.
This allows safe operation of separated networks without access to the uplink network.

Gluons features include
* easy configuration mode for less techy users
* community-specific customizations possible through `site.conf` and `site.mk`
* support for FastD/wireguard/tunneldigger as uplink tunnel
* support for mesh BATMAN/babel as mesh protocol
* usage of 802.11s for wifi meshing
* ecdsa signature-based autoupdater
* status-page
* node-data aggregation through respondd

T-X, 2023-08-29:
@maurerle I like the idea to list Gluon's main features on the starting pages, README and readthedocs, to better explain what Gluon does (and what it doesn't). And I like your text suggestion a lot, thanks for compiling this!

Three remarks/thoughts:

"usage of 802.11s for wifi meshing" might be misleading. 11s is used for direct node-to-node connections. The meshing feature of 11s is neither used nor supported in Gluon right now.

I'm also wondering if it's a good idea or not to start with the VPNs. As I have the feeling this often leads to misunderstandings on Gluon's architecture and abilities and limitations.

Last but not least I'm wondering if it would make sense to add a little bit of classification to the supported protocols? I like adding some hints like "partial" or "experimental" so that people won't get too surprised when they stumble over known issues or limitations (like missing IPv4 support with Babel and OLSRv2? and some missing bits with BATMAN V).

Here'd be a slightly modified feature introduction to Gluon based on your text:

---

Gluon provides an easy-to-use firmware for a public, decentral, WLAN and/or wire based mesh network. Which common network capable devices, like smartphones, laptops or desktop PCs can connect to and communicate over, without the need of passwords for access and without the need of installing special software. Additionally internet access and "cloud merging" can be provided over a WAN through VPN connected gateways.

Gluon's features include:

* a decentral mesh network
* easy (or optional) configuration mode for less techy users
* community-specific technical settings and customizations through a common site.conf and site.mk
* ecdsa signature-based autoupdater
* node status web page
* node information + statistics publication through respondd
* a variety of preconfigured mesh and VPN protocols:

Supported mesh protocols:

* batman-adv (BATMAN IV fully, BATMAN V partially)
* Babel (partially)
* OLSRv2 (partially)

Supported protocols for node-to-node connections:

* WLAN: 802.11s (with forwarding disabled)
* WAN: VPNs via fastd, Wireguard or Tunneldigger
* LAN: via VXLAN

PR #3035 (2023-12-28): docs: Add description about what gluon is
This fixes #2662 and uses the last proposal from T-X, which I think is quite fitting.
closed 2023-12-28 by blocktrron