freifunk-gluon/gluon #2663: v2022.1: Autoupdater from old release does not work for Loco M XW

goligo, 2022-10-10:
We are currently rolling out our first release based on v2022.1, and I noticed that Loco M XW is not updated, although a firmware image for the device exists. Reason seems to be a rename without proper alias in v2022.1:

In v2021.1.x, targets/ar71xx-generic:
```
device('ubiquiti-loco-m-xw', 'ubnt-loco-m-xw', {
	aliases = {
		'ubiquiti-nanostation-loco-m2-xw',
		'ubiquiti-nanostation-loco-m5-xw',
		-- 'ubiquiti-nanobeam-m5', -- untested
	},
})
```

In v2022.1.x, targets/ath79-generic
```
device('ubiquiti-nanostation-loco-m-xw', 'ubnt_nanostation-loco-m-xw', {
	manifest_aliases = {
		'ubiquiti-nanostation-loco-m2-xw', -- upgrade from OpenWrt 19.07
		'ubiquiti-nanostation-loco-m5-xw', -- upgrade from OpenWrt 19.07
	},
})
```

As far I understand the old name from v2021.1.x must be included as an alias that the firmware can be found by the autoupdater, like this:
```
device('ubiquiti-nanostation-loco-m-xw', 'ubnt_nanostation-loco-m-xw', {
	manifest_aliases = {
		'ubiquiti-loco-m-xw', -- upgrade from OpenWrt 19.07
	},
})
```

herbetom, 2022-10-11:
Support was readded in #2432 by @AiyionPrime.

@goligo I would propose you open a PR.

herbetom, 2022-10-11:
Relevant PRs for the migration seem to be #2562 and #2547.

AiyionPrime, 2022-10-11:
Busy at work, but I'd gladly review this later this evening.

goligo, 2022-10-11:
I have received the autoupdater output from one of the devices that didn't update and it confirmed what I wrote above.

```
root@Gauting_Bahnhof_6:~# autoupdater
Retrieving manifest from http://185.150.99.255/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: error downloading manifest: Connection failed
Retrieving manifest from http://firmware.ffmuc.net/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model ubiquiti-loco-m-xw)
Retrieving manifest from http://5.1.66.255/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: error downloading manifest: Connection failed
Retrieving manifest from http://[2001:678:e68:f000::]/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model ubiquiti-loco-m-xw)
Retrieving manifest from http://[2001:678:ed0:f000::]/experimental/sysupgrade/experimental.manifest ...
autoupdater: warning: no matching firmware found (model ubiquiti-loco-m-xw)
autoupdater: error: no usable mirror found
```

goligo, 2022-10-11:
Should I also open a PR for v2022.1.x branch? Or do you take care of this internally?

herbetom, 2022-10-11:
v2022.1.x is going to be an backport. Those should contain the commit hash from the master branch which is unknown before the merge.

Also there is an action which automatically takes care of creating backport PRs once a spefic label is added.

So i think this should be fine.

PR #2664 (2022-10-11): ath79-generic: Fix autoupdater for Loco M XW from v2021.1.x to v2022.1.x
Ubiquiti Nanostation Loco M XW was renamed in v2022.1.x, but the alias to the old name was missing, so devices running the old release did not update.

Fixes #2663
closed 2022-10-11 by blocktrron