freifunk-gluon/gluon #2665: cleanup /etc/opkg/keys on upgrades

rotanid, 2022-10-11:
we discussed in today's gluon meetup that we want to clear old opkg signing keys on each upgrade
currently, nodes that are around for years might have a load of old signing keys in that directory if the keys weren't re-used for new builds by the community.

maurerle, 2023-06-13:
That seems like a good way to do it:
I would add the above script right here:
https://github.com/freifunk-gluon/packages/tree/master/admin/autoupdater/files/usr/lib/autoupdater/upgrade.d

so that it gets picked up right after checking if the upgrade should actually be done here:
https://github.com/freifunk-gluon/packages/blob/master/admin/autoupdater/src/autoupdater.c#L420

neocturne, 2025-03-30:
As opkg is on the way out anyways (and I haven't check how this is going to be handled with apk), it may make sense to just patch the opkg package to remove `/etc/opkg/keys` from keep.d for now.

closed 2025-03-31 by rotanid