freifunk-gluon/gluon #2666: ipq40xx - DSA migration process

AiyionPrime, 2022-10-11:
Similar, yet incomparably simpler than #2413 this is tracking the DSA migration folowing https://github.com/openwrt/openwrt/pull/4721 and related patches.

The pattern stays same; find your device, produce a build on recent OpenWrt, follow the device integration checklist and report back what is working and what not.

**in progress**

| subtarget | tester         | PR      | device                              |
|-----------|----------------|---------|-------------------------------------|

**todo**

| subtarget | wanted to test | devices | device                              |
|-----------|----------------|---------|-------------------------------------|

//todo will follow up with this tomorrow

**done**

| subtarget | wanted to test | devices | device                              |
|-----------|----------------|---------|-------------------------------------|

**as tasklist for actual progress tracking**

**in progress**

**todo**

- [ ] aruba-ap-303 (aruba-instant-on-ap11)
- [ ] aruba-ap-303h (aruba-instant-on-ap11d)
- [ ] aruba-ap-365 (aruba_ap-365)
- [ ] avm-fritz-box-4040
- [ ] avm-fritz-box-7530
- [ ] avm-fritz-repeater-1200
- [ ] engenius-ens620ext
- [ ] gl.inet-gl-ap1300
- [ ] gl.inet-gl-b1300
- [ ] linksys-ea6350v3
- [ ] netgear-ex6100v2
- [ ] netgear-ex6150v2
- [ ] openmesh-a42
- [ ] openmesh-a62
- [ ] plasma-cloud-pa1200
- [ ] plasma-cloud-pa2200
- [ ] zyxel-nbg6617
- [ ] zyxel-wre6606
- [ ] 8devices-jalapeno
- [ ] mikrotik-sxtsq-5-ac-rbsxtsqg-5acd (mikrotik-discg-5acd)

**done**

blocktrron, 2022-10-11:
@AiyionPrime I think the AP-303 as well as 4040 are already migrated? A full re-validation is not necessary downstream, as it is performed upstream. We also didn't do it for neither mt7621 nor xrx200

ecsv, 2022-10-30:
* Pending Open-Mesh: https://github.com/openwrt/openwrt/pull/11078
* Pending PlasmaCloud: https://github.com/openwrt/openwrt/pull/11083

AiyionPrime, 2022-10-30:
I've got a notifier to monitor these changes in that part of openwrt's tree; once they make it to master, I'll mark them as done.

AiyionPrime, 2022-11-14:
updated the tracker, thanks @ecsv!

AiyionPrime, 2022-11-14:
Pending GL.iNet: https://github.com/openwrt/openwrt/pull/10909

rotanid, 2022-12-04:
i do own some netgear-ex6150v2, but i'm not sure i can help. maybe @mbaumga ?

AiyionPrime, 2022-12-28:
* Pending Aruba: https://github.com/openwrt/openwrt/pull/11579

JJWatMyself, 2022-12-29:
Aruba ap-303h tests are good and have reported this on the other thread a few minutes ago. There are two items I have identified to address in the future, but not specific to DSA.

I also flagged an invisible font/font color issue on the default theme, but will likely need to log a bug somewhere.

JJWatMyself, 2022-12-30:
Ok, DSA for Aruba AP-303H is live in the snapshot and I have verified.  I am running dumb AP with many VLANs and wired ports configured to do VLANs also.

https://github.com/openwrt/openwrt/pull/11579#issuecomment-1367888343

[image]

Djfe, 2023-02-06:
I'm going to fix this upstream for the `zyxel-wre6606`
Maybe also for the `netgear-ex6150v2`

AiyionPrime, 2023-02-24:
Hey there. Any chance you still have access to the device you added support for in OpenWrt? It currently doesn't get built, maybe you can help addressing the issue in an efficient manner.

- https://github.com/openwrt/openwrt/commit/300b7fe85a743ab0e039457a5a551819b49450da @blocktrron Aruba AP-365
- https://github.com/openwrt/openwrt/commit/dc4f6b896f009c24170e75af745fc636ef8166a8 @steveglennon EnGenius ENS620EXT
- https://github.com/openwrt/openwrt/commit/970f1914be3f299fb302ca74a214589397000858 @blocktrron Netgear EX6100v2/EX6150v2
- https://github.com/openwrt/openwrt/commit/4b280ad91a6b155ac71c417aaa7bb5f4e328712f @skorpy2009 ZyXEL WRE6606

Djfe, 2023-02-24:
I kinda stopped after the Zyxel didn't want to boot anymore and I wasn't able to interrupt boot anymore since then.
I'll try to get around to it this weekend but start with the EX6150v2 since it's easy to recover.

steveglennon, 2023-02-24:
I still have access. I will look into it.
Steve

AiyionPrime, 2023-03-08:
@blocktrron committed https://github.com/openwrt/openwrt/commit/bf055fcdca155993b9b6c82f951000eefb13288a

So 6100v2 and 6150v2 are done.

skorpy2009, 2023-04-11:
@AiyionPrime @blocktrron 
What do I have to do:

```
https://github.com/openwrt/openwrt/commit/4b280ad91a6b155ac71c417aaa7bb5f4e328712f @skorpy2009 ZyXEL WRE6606

    https://github.com/openwrt/openwrt/blob/master/target/linux/ipq40xx/image/generic.mk#L1194-L1195
```

Djfe, 2023-04-11:
test this https://github.com/Djfe/openwrt/commit/9c2353c61b9b9f2a7aa12c873a5f44b16f2b9fbb
I need to recover my device still, else I would have already done a PR.
One of the swports (switch ports) will come up as an interface, that's the one we need to keep then.

Djfe, 2023-04-28:
Update on my part:
I finally figured out what broke my Zyxel wre6606
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=45eb57f12f3a128a1822a20b7e536527ab92ca67
The kernel got too large and the bootloader doesn't load it fully.
I might've to increase the compat version so people don't accidentally brick their device by upgrading without adjusting bootcmd first.

Apart from that I still have that pesty issue that the wre6606 isn't reacting to any of my keyboard inputs.
While connecting tx to rx on my ttl directly with a cable works just fine.
I tested with screen, picocom and gtkterm.

Other devices I used my ttl adapter with so far didn't receive my tx either.
Next test: using an openwrt Router with ssh as my ttl Adapter.

mkg20001, 2023-05-21:
I tested hap ac2 (ipq40xx-mikrotik) as it gained dsa aswell. It has an increased compat version so I had to manually increment it in uci system compat_version to install the upgrade. Everything came up after upgrading. No new problems yet.

Djfe, 2023-05-21:
did you test openwrt or gluon?

mkg20001, 2023-05-21:
gluon, but installed image via sysupgrade and built from next branch.

AiyionPrime, 2025-06-27:
Thanks for everyone involved, we are done migrating all models, which currently have devices in the field.

closed 2025-06-27 by AiyionPrime