freifunk-gluon/gluon #2672: bridge MLD wakeup call feature was erroneously removed from stable branch

rotanid, 2022-10-14:
the "bridge MLD wakeup call feature" which was introduced by @T-X with PR #2076 and commit b2add48d2a7fa86c05eff3a2e6935168d60b2bf8 was accidently removed when switching the OpenWrt Base to 21.02 in commit aab2b914b8687fa91849998e27ed6b9cf27688db .
It was reintroduced in Gluon's master branch with #2365 commit 0530b06557ca2a7141e76d9f3c490f97383ae83c
Therefore, this feature was part of a Gluon stable release but is not part of the current stable release branch 2022.2.x as of now, which we also state in the "known issues" part of the release notes.
I suggest we backport it to fix this regression.

blocktrron, 2022-10-14:
It was removed because the patch did not apply and was not re-applied to v2022.1 because it was not applied to master back when we prepared the release.

rotanid, 2022-10-14:
> It was removed because the patch did not apply and was not re-applied to v2022.1 because it was not applied to master back when we prepared the release.

sure, i rephrased my issue to better suit the circumstances.

rotanid, 2022-10-23:
#2685

rotanid, 2022-12-18:
fixed by #2685 with next minor release

PR #2685 (2022-12-13): [Backport v2022.1.x] kernel: bridge: readding MLD wakeup call feature
With the update to OpenWrt 21.02 the bridge MLD wakeup call feature was
removed. However the issue with Android devices and MLD is still
present. Therefore readding the bridge MLD wakeup call patch and porting
it to Linux 5.10 / OpenWrt 22.03.

Link: https://issuetracker.google.com/issues/149630944
Fixes: aab2b914b868 ("modules: switch to OpenWrt 21.02")

closes #2672
closed 2022-12-18 by rotanid