freifunk-gluon/gluon #2679: Bridge Port Isolation not working on UBNT ER-X

mweinelt, 2022-10-18:
The current working theory is that this may be caused by offloading of the bridge port forwarding.

We would need someone to test this theory by disabling the relevant offloading features.

See https://github.com/freifunk-gluon/gluon/pull/2600#issuecomment-1245229432

AiyionPrime, 2022-11-25:
I flashed our current nightly based on gluons master; the problem persists.
This [meshlink](https://hannover.freifunk.net/karte/#/en/map/0418d668f3b4-44d1fa59f87c) should not be there.

On the ERX I performed the following actions:

```bash
root@erx-bridge-port-isolation-debugger:~#  opkg update; opkg install ethtool
```

```bash
root@erx-bridge-port-isolation-debugger:~#  ip a
```

```console
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1504 qdisc fq_codel state UP qlen 1000
    link/ether b4:fb:e4:53:ae:e8 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::b6fb:e4ff:fe53:aee8/64 scope link 
       valid_lft forever preferred_lft forever
3: eth0@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP qlen 1000
    link/ether b4:fb:e4:53:ae:e8 brd ff:ff:ff:ff:ff:ff
4: eth1@dsa: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-mesh_other state LOWERLAYERDOWN qlen 1000
    link/ether b4:fb:e4:53:ae:e9 brd ff:ff:ff:ff:ff:ff
5: eth2@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-mesh_other state UP qlen 1000
    link/ether b4:fb:e4:53:ae:ea brd ff:ff:ff:ff:ff:ff
6: eth3@dsa: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-mesh_other state UP qlen 1000
    link/ether b4:fb:e4:53:ae:eb brd ff:ff:ff:ff:ff:ff
7: eth4@dsa: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-mesh_other state LOWERLAYERDOWN qlen 1000
    link/ether b4:fb:e4:53:ae:ec brd ff:ff:ff:ff:ff:ff
8: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether fa:1d:95:4a:38:53 brd ff:ff:ff:ff:ff:ff
9: teql0: <NOARP> mtu 1500 qdisc noop state DOWN qlen 100
    link/void 
11: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether b4:fb:e4:53:ae:e8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.63/24 brd 192.168.178.255 scope global br-wan
       valid_lft forever preferred_lft forever
    inet6 2a02:560:529b:8c00:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 6915sec preferred_lft 3315sec
    inet6 fe80::b6fb:e4ff:fe53:aee8/64 scope link 
       valid_lft forever preferred_lft forever
12: local-port@local-node: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP qlen 1000
    link/ether b4:fb:e4:53:ae:e8 brd ff:ff:ff:ff:ff:ff
13: local-node@local-port: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether 16:41:95:40:f8:dc brd ff:ff:ff:ff:ff:ff
    inet 10.14.0.1/16 brd 10.14.255.255 scope global local-node
       valid_lft forever preferred_lft forever
    inet6 fdca:ffee:8:14::1/128 scope global deprecated 
       valid_lft forever preferred_lft 0sec
    inet6 fe80::1441:95ff:fe40:f8dc/64 scope link 
       valid_lft forever preferred_lft forever
14: br-client: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether b4:fb:e4:53:ae:e8 brd ff:ff:ff:ff:ff:ff
    inet6 2a02:790:ff:114:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 276sec preferred_lft 126sec
    inet6 2a02:790:ff:414:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 291sec preferred_lft 141sec
    inet6 2a02:790:ff:714:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 273sec preferred_lft 123sec
    inet6 2a02:790:ff:914:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 270sec preferred_lft 120sec
    inet6 2a02:790:ff:1014:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 239sec preferred_lft 89sec
    inet6 2a02:790:ff:514:b6fb:e4ff:fe53:aee8/64 scope global dynamic noprefixroute 
       valid_lft 277sec preferred_lft 127sec
[... 144 lines cut ...]
```
generic-segmentation-offload: on
generic-receive-offload: on
hw-tc-offload: on
```

I disabled them for the two lan interfaces that currently mesh with the other two routers.

```bash
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth2 generic-segmentation-offload off
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth2 generic-receive-offload off
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth2 hw-tc-offload off
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth3 generic-segmentation-offload off
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth3 generic-receive-offload off
root@erx-bridge-port-isolation-debugger:~# ethtool -K eth3 hw-tc-offload off
root@erx-bridge-port-isolation-debugger:~# /etc/init.d/network restart
```

Not sure about the error message, but restarting the network worked.

```console
Command failed: Not found
```

Both connected devices still see each other perfectly fine.

AiyionPrime, 2022-12-10:
Note: Bridge Port Isolation does not work on D-Link DGS-1210-10P either.

AiyionPrime, 2022-12-13:
While this just simply does not work in Hanover, it randomly breaks mesh connections in Darmstadt.
We concluded, either we get this resolved before an upcoming release or this (as well as it's backport will be reverted, as much as that'd suck).

@NeoRaider intends to experiment on the isolation feature on a FB4040 in the next days/weeks.

mweinelt, 2023-01-29:
The current thinking is that the DSA stack does not support bridge port isolation and fails to signal back its incapacity.

neocturne, 2024-06-15:
The bridge core supports passing these flags to DSA since Linux 5.19, however very few DSA drivers implement port isolation so far. I've added list of relevant drivers to the issue description.

olerem, 2024-06-15:
Hm.. there is still some work to do. I added isolation support for KSZ switches:
https://git.kernel.org/pub/scm/linux/kernel/git/netdev/net-next.git/commit/drivers/net/dsa/microchip?id=a7f08029e2e84ecafbfff50fcff976fafee72799

neocturne, 2024-06-15:
@AiyionPrime Missing bridge port isolation should not break any mesh connections, unless you build a ring or similar topology of multiple nodes and STP does not work to prevent a forwarding loop. I would be good to know if that is the case, or something else is going wrong in your deployment.

AiyionPrime, 2024-06-22:
I'm from Hanover, the finding was from FF Darmstadt, I think.
I only reported it here, in order not to lose the intel.

neocturne, 2025-03-29:
OpenWrt PR for qca8k: https://github.com/openwrt/openwrt/pull/18375

neocturne, 2025-04-08:
OpenWrt has been updated, so port isolation should work on qca8k now.

I guess we won't get port isolation support for lantiq_gswip any time soon (looking at the driver, this will be more complex to implement than for mt7530 and qca8k), so I'm moving this to the next milestone as a tracking issue.

rotanid, 2025-04-08:
@mweinelt @neocturne @AiyionPrime if i'm not mistaken, only broken devices/targets are left here, i suggest we close this issue?

Djfe, 2025-04-15:
> qca8k (ipq40xx, ipq806x)
I don't think this has been activated for ipq806x, has it? Does this only require testing on a supported device like the Netgear R7800 or would this require additional kernel changes?

Is support for port isolation on the mt7531 switch something we require? (for next gen devices on targets like mediatek-filogic and for last gen devices like mediatek-mt7622)

AiyionPrime, 2025-04-15:
I'm afraid we only've got one 550 left, which has been offline for the last 12 days.
https://hannover.freifunk.net/karte/#!/en/map/0019995f020a

Not sure whether we can acquire the owners help.
As it's just one out of 1.8K nodes, I'd say its not that much of a pressing issue for hanover.
I'll write to leon anyway.

Djfe, 2025-04-15:
I fail to see how x86 is related to this issue

neocturne, 2025-04-18:
> Is support for port isolation on the mt7531 switch something we require? (for next gen devices on targets like mediatek-filogic and for last gen devices like mediatek-mt7622)

@Djfe
The MT753x switch driver is used on all Mediatek SoCs, including filogic,
IPQ806x uses the regular upstream qca8k driver, so there is nothing more to enable, port isolation will just work.

All of this is done and should work in Gluon main (as noted by the checked boxes in the issue description).

Djfe, 2025-04-18:
That is great! 😊