freifunk-gluon/gluon #2687: renable ath10k-ct-smallbuffers for tiny devices to get Wifi 5 working

edeso, 2022-10-26:
i am using gluon on an TP-Link Archer C25v1 since years and 5GHz Wifi worked fine with  `ath10k-ct-smallbuffers` up until this was forced to use the mainline ath10k, 
https://github.com/freifunk-gluon/gluon/blob/master/targets/ath79-generic#L359-L363
which does not have a smallbuffers version. for now i resort to manually removing the enforcement.

would it be possible to remove the mainline `ath10k` enforcement from `tiny` devices in general in favor of the openwrt default, which is usually `ath10k-ct-smallbuffers`?  thanks!

blocktrron, 2022-10-27:
I've adjusted the title a bit.

> would it be possible to remove the mainline ath10k enforcement from tiny devices in general in favor of the openwrt default, which is usually ath10k-ct-smallbuffers?

The driver yes, the firmware no as far as I am mistaken, as ath10k-ct firmware does not support 11s on first generation chips.

Apart from that - go ahead. I don't necessarily think we should remove the broken flag, but it is primarily a task of doing research on a devices specifications and altering in the respective target file. A PR for that is very welcome!

edeso, 2022-10-30:
> I've adjusted the title a bit.

sure np 

> > would it be possible to remove the mainline ath10k enforcement from tiny devices in general in favor of the openwrt default, which is usually ath10k-ct-smallbuffers?
> 
> The driver yes, the firmware no as far as I am mistaken, as ath10k-ct firmware does not support 11s on first generation chips.

does ath10k-ct work with the mainline firmware? if so agreed, if not, considering it is broken completely with mainline ath10k i'd prefer the little bit more unbroken state, even not supporting 11s

> Apart from that - go ahead. I don't necessarily think we should remove the broken flag, but it is primarily a task of doing research on a devices specifications and altering in the respective target file. A PR for that is very welcome!

not sure how to proceed. i suggested to remove the `packages = ATH10K_PACKAGES_****,`  line. 
enforcing mainline non-ct firmware (assuming ct-driver works with it) will probably need a new variable definition, or?

blocktrron, 2022-10-30:
> does ath10k-ct work with the mainline firmware?

yes, firmware and driver are completely interchangable, ath10k-ct provides exclusive interfaces to -ct firmware

> if so agreed, if not, considering it is broken completely with mainline ath10k i'd prefer the little bit more unbroken state, even not supporting 11s

11s is a requirement for device integration, so this is not pushing forward by changing the firmware.

Remember, the host buffers are only influenced by the driver, not the firmware on the target. Also, there's a mainline ath10k smallbuffers variant available. So there is no need to switch to -ct firmware / driver.

edeso, 2022-10-31:
> > does ath10k-ct work with the mainline firmware?
> yes, firmware and driver are completely interchangable, ath10k-ct provides exclusive interfaces to -ct firmware

good to know
 
> > if so agreed, if not, considering it is broken completely with mainline ath10k i'd prefer the little bit more unbroken state, even not supporting 11s
> 11s is a requirement for device integration, so this is not pushing forward by changing the firmware.

well, as of now 5GHz is broken completely, oom killing/rebooting the whole box. but i see your point.
 
> Remember, the host buffers are only influenced by the driver, not the firmware on the target. 

looks like upstream Openwrt moved the buffer decrease patch to `ath10k-ct-smallbuffers` from `mac80211` earlier, where it "worked" for the mainline ath10k drivers, which explains why it used to work earlier, despite the mainline driver enforcement.

>Also, there's a mainline ath10k smallbuffers variant available. So there is no need to switch to -ct firmware / driver.

indeed, this commit added it
https://github.com/openwrt/openwrt/commit/694757a08f620a9f24b70003542d9dcd0abeac46
will try.
in case it works out would you ack something along the lines of ?

```
local ATH10K_PACKAGES_SMALLBUFFERS_QCA9887 = {
	'kmod-ath10k-smallbuffers',
	'-kmod-ath10k-ct',
	'-kmod-ath10k-ct-smallbuffers',
	'ath10k-firmware-qca9887',
	'-ath10k-firmware-qca9887-ct',
}

device('tp-link-archer-c25-v1', 'tplink_archer-c25-v1', {
	packages = ATH10K_PACKAGES_SMALLBUFFERS_QCA9887,
	broken = true, -- OOM with 5GHz enabled in most environments
	class = 'tiny', -- 64M ath9k + ath10k
})
```

blocktrron, 2022-10-31:
Yes, that looks good +1

edeso, 2022-11-02:
while i'm at it. any idea why there is this empty but regardless used `local ATH10K_PACKAGES_QCA9888 = {}` definition? 

also splitting firmware definitions from software packages would be more elegant (less duplicate definitions). e.g.
```
local ATH10K_PACKAGES = {
	'kmod-ath10k',
	'-kmod-ath10k-ct',
	'-kmod-ath10k-ct-smallbuffers',
}
local ATH10K_PACKAGES_SMALLBUFFERS = {
	'kmod-ath10k-smallbuffers',
	'-kmod-ath10k-ct',
	'-kmod-ath10k-ct-smallbuffers',
}
local ATH10K_FIRMWARE_QCA9887 = {
	'ath10k-firmware-qca9887',
	'-ath10k-firmware-qca9887-ct',
}
...
```
but is it even possible to merge them later to one list of packages? is that lua? should i use unpack?

blocktrron, 2022-11-02:
I don't see the benefit of doing so.

edeso, 2022-11-03:
@blocktrron np, i'll duplicate the arrays then

what about 
> while i'm at it. any idea why there is this empty but regardlessly used local ATH10K_PACKAGES_QCA9888 = {} definition?
?

blocktrron, 2022-11-05:
This is to preserve the Device <--> Wireless Chipset mapping for future use.

edeso, 2022-11-09:
PR above. didn't touch `ATH10K_PACKAGES_QCA9888` as it empty, not enforcing mainline ath10k and assuming that the openwrt default does already use smallbuffer ath10k driver.

edeso, 2022-11-14:
nope, all good. will close the issue as fixed. thanks again!

closed 2022-11-14 by edeso