freifunk-gluon/gluon #2711: naming of "prefix6" option is confusing

mkg20001, 2022-12-05:
so, what does it do? intuitively, no clue

documentation in the example says "Prefixes used within the mesh.", which isn't any bit more specific 

it appears that it's for the clients connected to client ap (correct me if i'm wrong), so it could be named "client_prefx6"

Djfe, 2023-02-06:
Does this help?
https://github.com/ffac/site/blob/v2022.1.x/site.conf#L19-L21

Djfe, 2023-02-06:
Maybe Aachen is also using it differently, this was configured years before I joined them.
It appears to me that the IPv6 range is used by radvd to give your client some sort of privat/local ipv6 address that's valid only within your Freifunk network (but not outside of it, this it "within the mesh" probably is supposed to imply). You can access a node via ssh for example using this ip or open it's http status page, but only while connected to Freifunk.
The ipv6 is also unique due to using your device's primary mac.
I'm therefore not sure how prefix4 would work.

here's the local node feature
https://github.com/ffac/site/blob/v2022.1.x/site.conf#L68-L72
My guess is, it has to use a unique ipv6 out of the range above.

mkg20001, 2023-02-06:
From what I've read in the source code, it appears layer3 mesh uses node_prefix6 for the actual node IP (that is the ipv6 that a babel node gets for itself) and prefix6 is just thrown in everywhere where there's the need for a client subnet

I've figured that prefix6 is client_prefix6... basicallly... in babel at least... and I feel like this should be renamed and documented properly instead of just being this magic.

Is the behaviour the same in batman? Then I guess it's better to go for client_prefix6 and alias and deprecate prefix6

If there's still the need for a special prefix6 for some magic then another solution would be needed

mkg20001, 2023-02-06:
When using next_node.ip4, prefix4 becomes important because of the netmask which is taken from the prefix (needs to be set to something reasonable so arp works). And then I made ddhcpd in community-packages use that as default for the range aswell, since it's basically client_prefix4 in my eyes.

With prefix6 no idea if it's actually a hard requirement for next_node.ip6, but radvd depends on it, so it's required regardless

NeoRaider, 2023-02-07:
For batman-adv-based networks, `prefix6` is the only IPv6 prefix relevant for Gluon - both the nodes' own addresses and the client addresses are assigned from this prefix (so clients and nodes can communicate with each other).

For our Babel support, additional prefixes were deemed necessary (`node_prefix6` and `node_client_prefix6`), but `node_client_prefix6` is only used for communication between l3roamd instances inside the mesh (and configuring this manually was deprecated in #2569. I'm not sure if `node_prefix6` is actually necessary at all, or if node and client addresses could be assigned from the same prefix (thus making `prefix6` the only IPv6 prefix in site.conf again).