freifunk-gluon/gluon #2721: FritzBox 7530/7520 does only have eth0

maurerle, 2022-12-11:
### Bug report

**What is the problem?**

The FB 7520 does not support using the lan ports, all four ports work as WAN with the current 2022 release.

```
root@ffac-7520-test:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-wan state UP group default qlen 1000
    link/ether ee:a9:90:3f:f3:0d brd ff:ff:ff:ff:ff:ff
3: dummy0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2e:14:42:bd:26:64 brd ff:ff:ff:ff:ff:ff
4: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
7: br-wan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 02:9b:7f:ac:9e:a8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.87/24 brd 192.168.178.255 scope global br-wan
       valid_lft forever preferred_lft forever
    inet6 fe80::9b:7fff:feac:9ea8/64 scope link 
       valid_lft forever preferred_lft forever
8: local-port@local-node: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP group default qlen 1000
    link/ether 3c:37:12:82:cb:ab brd ff:ff:ff:ff:ff:ff
9: local-node@local-port: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether ac:41:95:40:f7:dc brd ff:ff:ff:ff:ff:ff
    inet6 fdac::ac/128 scope global deprecated 
       valid_lft forever preferred_lft 0sec
    inet6 fe80::ae41:95ff:fe40:f7dc/64 scope link 
       valid_lft forever preferred_lft forever
10: br-client: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 3c:37:12:82:cb:ab brd ff:ff:ff:ff:ff:ff
    inet6 2a03:2260:3006:120:3e37:12ff:fe82:cbab/64 scope global dynamic noprefixroute 
       valid_lft 572sec preferred_lft 2sec
    inet6 fdac::3e37:12ff:fe82:cbab/64 scope global dynamic noprefixroute 
       valid_lft 86328sec preferred_lft 14328sec
    inet6 fe80::3e37:12ff:fe82:cbab/64 scope link 
       valid_lft forever preferred_lft forever
11: wg_mesh: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN group default qlen 1000
    link/none 
    inet6 fe80::251:d9ff:fe42:b35a/128 scope link 
       valid_lft forever preferred_lft forever
12: bat0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UNKNOWN group default qlen 1000
    link/ether 3c:37:12:82:cb:ab brd ff:ff:ff:ff:ff:ff
    inet6 fe80::3e37:12ff:fe82:cbab/64 scope link 
       valid_lft forever preferred_lft forever
13: primary0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UNKNOWN group default qlen 1000
    link/ether 02:9b:7f:ac:9e:ab brd ff:ff:ff:ff:ff:ff
    inet6 fe80::9b:7fff:feac:9eab/64 scope link 
       valid_lft forever preferred_lft forever
14: mesh-vpn: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1350 qdisc noqueue master bat0 state UNKNOWN group default qlen 1000
    link/ether 2a:03:7d:de:fe:f7 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2803:7dff:fede:fef7/64 scope link 
       valid_lft forever preferred_lft forever
15: mesh0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UP group default qlen 1000
    link/ether 02:9b:7f:ac:9e:a9 brd ff:ff:ff:ff:ff:ff permaddr 3c:37:12:82:cb:ab
    inet6 fe80::9b:7fff:feac:9ea9/64 scope link 
       valid_lft forever preferred_lft forever
16: mesh1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1532 qdisc noqueue master bat0 state UP group default qlen 1000
    link/ether 02:9b:7f:ac:9e:ad brd ff:ff:ff:ff:ff:ff permaddr 3c:37:12:82:cb:ac
    inet6 fe80::9b:7fff:feac:9ead/64 scope link 
       valid_lft forever preferred_lft forever
17: client0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP group default qlen 1000
    link/ether 02:9b:7f:ac:9e:a8 brd ff:ff:ff:ff:ff:ff permaddr 3c:37:12:82:cb:ab
    inet6 fe80::9b:7fff:feac:9ea8/64 scope link 
       valid_lft forever preferred_lft forever
18: client1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-client state UP group default qlen 1000
    link/ether 02:9b:7f:ac:9e:ac brd ff:ff:ff:ff:ff:ff permaddr 3c:37:12:82:cb:ac
    inet6 fe80::9b:7fff:feac:9eac/64 scope link 
       valid_lft forever preferred_lft forever
```

**What is the expected behaviour?**

One should be able to set LAN1 as WAN and use the other ports as LAN.
I did not yet check the behavior on 19.07 but will do so in the upcoming days.

**Gluon Version:**

v2022.1.1

**Site Configuration:**

FFAC - https://github.com/maurerle/site

**Custom patches:**

blocktrron, 2022-12-11:
> The FB 7520 does not support using the lan ports, all four ports work as WAN with the current 2022 release.

This is intended behavior. If there's only on type of ports, they are treated as a single port and thus bridged.

maurerle, 2022-12-11:
So uci show network shows:
```
network.globals=globals
network.globals.ula_prefix='auto'
network.@switch[0]=switch
network.@switch[0].name='switch0'
network.@switch[0].reset='1'
network.@switch[0].enable_vlan='1'
network.@switch_vlan[0]=switch_vlan
network.@switch_vlan[0].device='switch0'
network.@switch_vlan[0].vlan='1'
network.@switch_vlan[0].ports='1 2 3 4 0'
```

I could not fix this by setting:
`lan_ifname, wan_ifname = 'eth0.2 eth0.3 eth0.4', 'eth0.1'`  for this device in 
`package/gluon-core/luasrc/lib/gluon/upgrade/020-interfaces`

So there is no possibility to fix this?

maurerle, 2022-12-11:
We should at least add mark5 to the docs here then:
https://gluon.readthedocs.io/en/v2021.1.x/user/supported_devices.html#id24

right?

rotanid, 2022-12-11:
> > The FB 7520 does not support using the lan ports, all four ports work as WAN with the current 2022 release.
> 
> This is intended behavior. If there's only on type of ports, they are treated as a single port and thus bridged.

no, in recent device additions we treated port 1 as WAN.

blocktrron, 2022-12-11:
@rotanid This was done on boards which were not added prior introducing DSA support. The AVM 7520 and 7530 did not support it because of DSA.

We can not migrate these later down the road, as users might operate them with WAN attached not to port 1.

closed 2022-12-11 by blocktrron

maurerle, 2022-12-11:
@rotanid can you maybe describe how this is done in cases like this?

The issue with updating is of course relevant, so we can not change this in gluon.
But in our case I would like to have a custom patch as we don't have these devices in the community yet and would like to allow mesh_on_lan with them.

rotanid, 2022-12-12:
@maurerle look at Netgear WAC104 for example

maurerle, 2022-12-12:
Hi @rotanid ,
if I understand that correctly, this is done if `ip a` shows lan1, lan2 as seperated dev.
I implemented this successfully for the o2 box 6431 in my PR #2595 

But this is not the case here right?
So I assumed that one would have to add the device here?

https://github.com/freifunk-gluon/gluon/blob/4d70f1d95077c0ffe8731986bc104fee477a5986/package/gluon-core/luasrc/lib/gluon/upgrade/115-swconfig#L6-L10

But i am unsure how this would look like for 4 ports..
Might be that I am on the wrong track here

maurerle, 2023-04-13:
Sooo starting with openwrt-23 this is not the case anymore:

```
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP qlen 1000
    link/ether 7e:d2:0f:2c:e9:f3 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::7cd2:fff:fe2c:e9f3/64 scope link 
       valid_lft forever preferred_lft forever
3: lan1@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-wan state LOWERLAYERDOWN qlen 1000
    link/ether 7e:d2:0f:2c:e9:f3 brd ff:ff:ff:ff:ff:ff
4: lan2@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP qlen 1000
    link/ether 7e:d2:0f:2c:e9:f3 brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-wan state UP qlen 1000
    link/ether 7e:d2:0f:2c:e9:f3 brd ff:ff:ff:ff:ff:ff
6: lan4@eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue master br-wan state LOWERLAYERDOWN qlen 1000
    link/ether 7e:d2:0f:2c:e9:f3 brd ff:ff:ff:ff:ff:ff
```

So I would expect that we can set lan2-4 as lan port. Yet this is not the case..

```
root@ffsg-lu-salentu-1:~# uci show gluon
gluon.core=core
gluon.iface_single=interface
gluon.iface_single.name='/single'
gluon.iface_single.role='uplink'
gluon.wireless=wireless
gluon.wireless.outdoor='0'
gluon.wireless.preserve_channels='0'
gluon.mesh_vpn=mesh_vpn
gluon.mesh_vpn.enabled='1'
gluon.mesh_vpn.limit_enabled='0'
```
the device recognizes only iface_single, even on a fresh install.
While it should be split into iface_wan and iface_lan

On the other hand, the exact same configuration works using the o2-box 6431 which has nearly the same board.json

I also wonder, why the wan/dsl interface is not listed at the 7530 while it is listed in the o2-box 6431

[6431-board.json.txt](https://github.com/freifunk-gluon/gluon/files/11224857/6431-board.json.txt)

[7520-board.json.txt](https://github.com/freifunk-gluon/gluon/files/11224856/7520-board.json.txt)

Is this something with the .dts files which hinders the 7520/30 from getting proper support?

I added this code to split lan and wan interfaces correctly but it did not work: https://github.com/ffac/site/commit/26b89c2bed953fb2a50e243ecbb695bb3390d312

What am I missing here?

PR #2722 (2022-12-11): docs: add LAN-AS-WAN label to FB 7520/7530
These devices use all LAN ports as a WAN bridge.

Closes #2721
commit 4d70f1d9 (2022-12-11): docs: add LAN-AS-WAN label to FB 7520/7530 (#2722)

These devices use all LAN ports as a WAN bridge.

Closes #2721