freifunk-gluon/gluon #2724: Abbility to turn off VXLan on individual interfaces

mkg20001, 2022-12-13:
Sometimes it may be necessary to disable VXLAN on an interface, for example when a VLAN is already being used or meshing to a node without VXLAN support is desired.

There should be a way to specify this in the webinterface aswell, perhaps this could be paired with https://github.com/freifunk-gluon/gluon/pull/2688

mkg20001, 2023-01-06:
I'm going to go with mesh-without-vxlan role and then they will simply be tossed into one bridge, as currently done with just the mesh interfaces. Does that sound sane?

neocturne, 2025-03-30:
Generally, a `mesh-raw` or similar interface role would make sense.

To note: An additional bridge + mesh interface would also mean we have to reserve one more MAC address for wired meshing. This can be worked into our current scheme (https://gluon.readthedocs.io/en/latest/dev/mac_addresses.html) - `owe0` and `owe1` could be reused - but it does pose a problem for my plans to reduce the base allocation to 4 addresses on single-radio devices (by getting rid of the distinct primary address).

With the new bridge, the `mesh-raw` role would be an "exclusive" role like `client` that can't be combined with other roles. This is only a limitation for batman-adv though, where it is possible to use non-VXLAN Mesh-on-WAN due to batman-adv's custom Ethertype; for L3 routing protocols, combining the roles wouldn't make sense anyways.