freifunk-gluon/gluon #2729: .min.js are not always using their appropriate suffix

AiyionPrime, 2022-12-18:
minified js files should end in `.min.js`.
We mostly adhere to this, but overlooked three occurrences:

- [ ] [package/gluon-status-page/files/lib/gluon/status-page/www/static/status-page.js](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-status-page/files/lib/gluon/status-page/www/static/status-page.js)
- [ ] [package/gluon-web-model/files/lib/gluon/web/www/static/gluon-web-model.js](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-model/files/lib/gluon/web/www/static/gluon-web-model.js)
- [ ] [package/gluon-web-osm/files/lib/gluon/web/www/static/gluon-web-osm.js](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-osm/files/lib/gluon/web/www/static/gluon-web-osm.js)

Renaming them and their usages to end in `min.js` would help IDEs to not reformat them.

mweinelt, 2022-12-18:
Please submit a PR if possible to rename. I'm in favor.

AiyionPrime, 2022-12-18:
Same goes for `.min.css` while we're at it.

> Please submit a PR if possible to rename. I'm in favor.

will do, thanks

PR #2730 (2023-01-04): Minimized js naming
~This won't compile. I've made at least one mistake in the statuspage I think.~

Compiles just fine, testing is needed though.

The goal is to keep the `js`-suffix for deployed files, regardless of whether they are minimized or not.

Eventually, this resolves #2729