freifunk-gluon/gluon #2733: respondd reloads: are they working?

mkg20001, 2022-12-20:
### Bug report

**What is the problem?**
reloading respondd should work and respondd should re-listen on interfaces

the init.d trigger uses reload

```
	procd_open_trigger
	procd_add_raw_trigger "interface.*" 0 "/etc/init.d/$name" reload
	procd_close_trigger

```

there's no reload method so it's propably sighup

https://github.com/freifunk-gluon/packages/blob/master/net/respondd/src/respondd.c doesn't show any signs of checking for SIGHUP or other reload mechanisms

note: just patching it so it is restarted instead is working and might be preferred because otherwise an interface similar to mmfd's socket would be needed (to remove/add interfaces) but respondd doesn't really carry that much state or cause trouble when it's down for a sec that I think restart would just do the job

**What is the expected behaviour?**
it doesn't

**Gluon Version:**
85f3e40a14c5f346fb3ad320b2efe6d3031459b4

**Site Configuration:**
https://github.com/ffgraz/funkfeuer-graz-gluon-site

**Custom patches:**
whatever graz has

closed 2023-02-22 by mkg20001

rotanid, 2023-02-22:
@mkg20001 what was the reasoning behind closing this issue?

mkg20001, 2023-02-23:
It appears that the service is restarted via the reload call in the init script which appears to work, so this issue is resolved, unless there's another reason to keep it open.