freifunk-gluon/gluon #2742: Gluon on Aerohive AP330

mersl, 2023-01-02:
### Bug report

Most recent versions of Gluon do not boot anymore on Aerohive AP330 and/or system upgrades are failing.

It seems, this is related to the changes on OpenWRT to boot Linux 5.x kernels on those devices. The full story can be read at 
[https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=cfe79f2eb8b3c874f3d1a4441b8a60af32169974](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=cfe79f2eb8b3c874f3d1a4441b8a60af32169974)

I was able to freshly install OpenWRT/Gluon/FFMUC v2022.10.5 on my AP330s using the comments on the link above and it's working fine as far as I can see. 

Nevertheless, an unattended upgrade from an previous version will fail as there are changes on the bootloader variables required. 

**What is the expected behaviour?**

an upgrade from an older version of gluon is denied with some meaningful message OR some magic is added to adjust the settings required to boot new versions of OpenWRT/Gluon as described in above link.

**Gluon Version:**

```# gluon-info 
Hostname:               	XXX
MAC address:            	XX:XX:XX:XX:XX:XX
Hardware model:         	Aerohive HiveAP-330
Gluon version / Site version:	v2022.1.1+ / v2022.10.4-10-ge97d1d9
Firmware release:       	v2022.10.5
Site:                   	Freifunk München
Public VPN key:         	n/a
```

mersl, 2023-01-02:
this replaces [https://github.com/freifunkMUC/site-ffm/issues/224](https://github.com/freifunkMUC/site-ffm/issues/224) since it's an upstream issue ;-)

herbetom, 2023-01-02:
Communities with those devices: https://stats.darmstadt.freifunk.net/d/CA5PRFmMz/gluon-census?orgId=1&var-version_major=All&var-version=All&var-models=Aerohive%20HiveAP-330&viewPanel=22

rotanid, 2023-01-03:
during my experiments, the device did *not* brick and did not do any sysupgrade without using the -F (force) parameter.
therefore, the requested behaviour
```
an upgrade from an older version of gluon is denied with some meaningful message
```
is already the case or what did i miss?

rotanid, 2023-01-03:
if you have access to the device, there's an easy solution:

1. try "sysupgrade /tmp/sysupgrade.img" and note the message
2. follow the instructions given by the message (see below, i pasted them)
3. ONLY AFTER SUCCESS OF STEP 2, do `sysupgrade -n -F /tmp/sysupgrade.img`
4. BE PATIENT! the flashing might take up to 15 minutes, this device takes the longest to flash from the ones i know.
5. connect laptop/PC to router and configure the gluon node via config mode

after the above procedure, further updates work normally (at least i tried once)

```
Device aerohive,hiveap-330 not supported by this image
Supported devices: aerohive,hiveap-330 - Image version mismatch: image 2.0, device 1.0. Please wipe config during upgrade (force required) or reinstall. Reason: 
!The partitioning of the HiveAP 330 has changed! 
To upgrade, please take a look at the install instructions over at the device's wiki: <https://openwrt.org/toh/aerohive/hiveap-330> 
An abridged version for the console is provided here for comfort. 
Run the following script into a shell on the device and retry this sysupgrade again: 

cat <<- "EOF" > /tmp/uboot-fix.sh; sh /tmp/uboot-fix.sh 
. /lib/functions.sh 
. /lib/functions/system.sh 
opkg update && opkg install uboot-envtools kmod-mtd-rw || exit 2 
insmod mtd-rw i_want_a_brick=y || exit 3 
echo "/dev/mtd$(find_mtd_index u-boot-env) 0x0 0x20000 0x10000" > "/etc/fw_env.config" 
fw_setenv owrt_boot 'setenv bootargs console=ttyS0,9600;bootm 0xEC040000 - 0xEC000000' 
cp "/dev/mtd$(find_mtd_index 'u-boot')" /tmp/uboot 
cp /tmp/uboot /tmp/uboot_patched 
strings -td < /tmp/uboot | grep '^ *[0-9]* *\(run owrt_boot\|setenv bootargs\).*cp\.l' | 
awk '{print $1}' | 
while read offset; do 
echo -n "run owrt_boot;            " | dd of=/tmp/uboot_patched bs=1 seek=${offset} conv=notrunc 
done
mtd write /tmp/uboot_patched u-boot;
uci set system.@system[0].compat_version=2.0; uci commit; 
EOF

Note that if this fails, you will need to use the serial console to re-install OpenWrt. 
Note that after this sysupgrade, the AP will be unavailable for 7 minutes to reformat flash."
Image check failed.

```

kpanic23, 2023-10-26:
This is great.
Why is this not mentioned in the release notes? It's only in this closed issue.
I just found out about this after all of our HiveAP-330 went offline after the upgrade from v2021.1.2 to 2023.1.1.
It doesn't help either that the autoupdater doesn't re-enable networking after the image check failed, leaving the nodes without connectivity until you power-cycle them.

herbetom, 2023-10-26:
@kpanic23 The Issue had been closed and nobody noticed that it probably should have stayed open to either develop a solution/workarround or note it as a known Issue.

Closed issues (without special tags, or as part of a milestone) are normally not considered for the creation of the release notes, but only open issues and code changes (PRs).

Just a series of unfortunate events.

Please feel free to open an Issue for adapting the autoupdater.

mersl, 2023-10-26:
the problem with that is, there was a major change in the upstream/openwrt (check the link in the initial post). 
Gluon and all Freifunk downstreams just depend on that and since there wasn't any special release note on OpenWRT, I assume, this was not visible for Glueon and other projects.
I doubt, this can be managed with autoupdater since the flash needs to be completely reinitialized and uboot reconfiguration to run the (much larger) 5.x kernels as explained in the OpenWRT repo for Aerohive. Once Aerohive is on the new flash schema, further autoupdates just work as usual. Ie FFMUC did release in the meantime newer versions and AP330s just upgraded as normal.
Unlucky us, Aerohives are less common as other devices and things don't show up to a broughter audience...

BTW: AP330s runs rock solid on the new firmware ;-)

mersl, 2023-10-26:
@herbetom : A comment in 'known issues' would be beneficial I guess ;-)