freifunk-gluon/gluon #2750: cellular: authentication dropdown missing

AiyionPrime, 2023-01-04:
> thxs 4 the nice work.
> for some provider, the auth feature is required to get a connection.
> could you be so kind and add an dropdown menu for the auth with:
> none, pap, chap or both
> txs!!!

_Originally posted by @heini66 in https://github.com/freifunk-gluon/gluon/issues/2631#issuecomment-1355086451_

Djfe, 2023-06-16:
a setting for `roaming` might be useful as well

T0biii, 2024-07-10:
> a setting for `roaming` might be useful as well

@Djfe should this be a separate issue? the #3307 only adds the auth dropdown

Djfe, 2024-07-10:
I expected roaming to be a uci setting. It might be a setting on the sim card that needs to be executed with uqmi.
yes, a separate issue makes sense. but first I should test whether roaming currently works by default, if the sim was set to allow it. feel free to close this issue with your PR

PR #3307 (2024-07-11): gluon-web-cellular: add auth option to GUI
this should fixes #2750 by adding a dropdown menu for the auth methods.

atm it is untested, but we believe that this can be tested soon by someone from [ffmuc](https://github.com/freifunkMUC/site-ffm/pull/456).

More Infos:
https://openwrt.org/docs/guide-user/network/wan/wwan/ltedongle#qmi_protocol_configuration_parameters
closed 2024-07-11 by blocktrron