freifunk-gluon/gluon #2756: software-controlled LEDs spanning multiple LAN ports (DSA)

AiyionPrime, 2023-01-10:
I think there is no need to stall #2716 much longer.

> I think we have to take a look at software-controlled LEDs spanning multiple LAN ports in conjunction with DSA, as this is currently not handled nicely upstream. I assume some other devices have lost sane handling with migrating to DSA also.

_Originally posted by @blocktrron in https://github.com/freifunk-gluon/gluon/issues/2716#issuecomment-1345694190_

Djfe, 2023-02-05:
Also: LED handles for MT7621 devices (MT7530 switch) were removed by https://github.com/openwrt/openwrt/commit/fc33c41c21362b7186aa051a2140623943fa3143

So there is only a hacky way right now to disable them if you want to by using mdio manually.
https://forum.openwrt.org/t/openwrt-support-for-wax206/127523/294?u=djfe

Why where they removed? Because registering them as mdio and firing them with netdev doesn't show the offloaded traffic between lan ports on a switch and because they cause many interrupts to be fired on mdio which reduces performance massively.
Discussion was here (it spans over several posts):
https://github.com/openwrt/openwrt/pull/10829#issuecomment-1281930122 

This is how I got aware of it:
https://github.com/openwrt/openwrt/pull/11601#issuecomment-1364692035
https://github.com/openwrt/openwrt/pull/11601#issuecomment-1365076527

OpenWRT maintainers are waiting for this to be resolved upstream (Linux), so they don't have to maintain a solution downstream that belongs upstream. The solution would be to allow setting registers on the switch to enable and disable it from there (or something like that). Right now the switch handles blinking on packets itself. (in hardware/offloaded)
This was a first try to send in a patch upstream, I think:
https://lwn.net/ml/linux-kernel/20220503151633.18760-1-ansuelsmth@gmail.com/

Issue in OpenWRT:
https://github.com/openwrt/openwrt/issues/11010

namiltd, 2025-08-24:
I created a module to support LEDs for all LAN interfaces (ledtrig-netdev-lan): https://forum.openwrt.org/t/multiple-devices-ports-status-single-led/239622/3
The patch also includes a similar module for WiFi (ledtrig-netdev-phy).
Tests and any comments are welcome.

Kistelini, 2026-11-13:
@namiltd thanks for your work! I opened a pull request for the affected devices in Gluon: https://github.com/openwrt/openwrt/pull/26689