freifunk-gluon/gluon #2762: V2022.1.x – command to view "default_domain"

cptechnik, 2023-01-18:
default_domain

please add function to check "default_domain"

- in /local/bin/gluon-info

and 

- /local/bin/gluon-switch-domain with new parameter "status" , so if called "gluon-switch-domain status" the output is eg... freifunk_hood_berlin

cptechnik, 2023-01-18:
... and the option to view possible domains....
like 
` "gluon-switch-domain view" `
= 
`freifunk_hood_berlin, freifunk_hood_brandenburg`

maybe 
ls /lib/gluon/domains/|grep json  
...but w/o .json

cptechnik, 2023-01-18:
> in /local/bin/gluon-info

something like ..."uci get gluon.core.domain"

AiyionPrime, 2025-07-19:
Hey @cptechnik, sorry it took me so long to get back to this.

Extensions to `gluon-switch-domain` like `--list` or `--status` are indeed feasible.

The motivation for the default domain does not hit me yet.
Could you elaborate, when you'd need it?

Thanks for the ideas and have a nice weekend, Aiyion