freifunk-gluon/gluon #2784: 2022.1.2: Ubiquiti AC-Mesh/ubnt_unifiac-mesh hangs on boot

ecsv, 2023-02-06:
It was noticed that the AC mesh is no longer booting when it is upgraded to gluon 2022.1.2. Even on a fresh install, the output is only:

```
U-Boot unifi-v2.0.9.307-gd60f380a (Mar 25 2021 - 15:00:50)

DRAM:
sri
ath_ddr_initial_config(278): (ddr2 init)
ath_sys_frequency: cpu 775 ddr 650 ahb 258
Tap values = (0xf, 0xf, 0xf, 0xf)
128 MB
Base:0x80000000, Top:0x88000000, Res logbuf:0xa7ff3800, log_magic:0x55555555 kseg: 0xa0000000
Top of RAM usable for U-Boot at: 87ff3000
Reserving 252k for U-Boot at: 87fb0000
Reserving 192k for malloc() at: 87f80000
Reserving 44 Bytes for Board Info at: 87f7ffd4
Reserving 36 Bytes for Global Data at: 87f7ffb0
Reserving 128k for boot params() at: 87f5ffb0
Stack Pointer at: 87f5ff98
Now running in RAM - U-Boot at: 87fb0000
Flash: 16 MB
PCIe WLAN Module found (tries: 1).
In:    serial
Out:   serial
Err:   serial
Net:   ath_gmac_enet_initialize...
Fetching MAC Address from 0x9fff0000
ath_gmac_enet_initialize: reset mask:c02200
athr_mgmt_init ::done
Dragonfly  ----> S17 PHY *
 ath_gmac_enet_initialize: is_s17()=0, is_ar8033()=1, phy id1=4d phy_id2=d074
WAN AR8033 PHY init
athrs_ar8033_reg_init: Done 111
Max resets limit reached exiting...
athr_gmac_sgmii_setup SGMII done
: cfg1 0x80000000 cfg2 0x7114
eth0: 74:ac:b9:66:ae:49
eth0 up
eth0
Setting 0x181162c0 to 0x20402100
Board: Copyright Ubiquiti Networks Inc. 2014
Hit any key to stop autoboot:  0 
## Starting application at 0x80200020 ...
Board: Ubiquiti Networks AR956X board (e557-12.1150.0030.0040)
 0. Name = u-boot, offset = 0, start_addr=9f000000, size=393216,start_sector=0, end_sector=5 
 1. Name = u-boot-env, offset = 60000, start_addr=9f060000, size=65536,start_sector=6, end_sector=6 
 2. Name = kernel0, offset = 70000, start_addr=9f070000, size=7929856,start_sector=7, end_sector=127 
 3. Name = kernel1, offset = 800000, start_addr=9f800000, size=7929856,start_sector=128, end_sector=248 
 4. Name = bs, offset = f90000, start_addr=9ff90000, size=131072,start_sector=249, end_sector=250 
 5. Name = cfg, offset = fb0000, start_addr=9ffb0000, size=262144,start_sector=251, end_sector=254 
 6. Name = EEPROM, offset = ff0000, start_addr=9fff0000, size=65536,start_sector=255, end_sector=255
get_mtd_params: name=bs
ubnt_flash_read: addr=8023c170, sa=9ff90000, sz=131072 
ubnt_bootsel_init: bootsel magic=a34de82b, bootsel = 0 
UBNT application initialized 
## Application terminated, rc = 0x0
## Starting application at 0x80200020 ...
keep cfg partition. 
## Application terminated, rc = 0x0
## Starting application at 0x80200020 ...
ubnt_uwrite: Nothing to flash, exiting 
## Application terminated, rc = 0x0
## Starting application at 0x80200020 ...
Number of boot partitions = 2 
get_mtd_params: name=bs
ubnt_flash_read: addr=8023c170, sa=9ff90000, sz=131072 
ubnt_get_bootsel: Boot partition selected = 0 
Loading Kernel Image @ 81000000, size = 7929856 
Verifying 'kernel0' parition:OK
## Application terminated, rc = 0x0
## Booting image at 9f070000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.161
   Created:      2022-03-22  20:06:42 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2212954 Bytes =  2.1 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum at 0x9f070040 ...OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80060000) ...
## Giving linux memsize in bytes, 134217728

Starting kernel ...

[    0.000000] Linux version 5.10.161 (gluonbuilder@atlas) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20044+13-ddeeb35007) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Tue Mar 22 20:06:42 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Ubiquiti UniFi AC Mesh
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
```

I've also tried to boot the original OpenWrt 22.03.3 image: [ubnt_unifiac-mesh-initramfs-kernel.bin](https://downloads.openwrt.org/releases/22.03.3/targets/ath79/generic/openwrt-22.03.3-ath79-generic-ubnt_unifiac-mesh-initramfs-kernel.bin)

```
       setenv serverip 192.168.112.232
       setenv ipaddr 192.168.112.123
       tftpboot  0x81000000 acmesh.bin && bootm $fileaddr
```

But this is booting fine.

Simply loading the kernel from gluon is definitely not fine:

```
bootm    9f070000            
## Booting image at 9f070000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.161
   Created:      2022-03-22  20:06:42 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    2212954 Bytes =  2.1 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum at 0x9f070040 ...OK
   Uncompressing Kernel Image ... OK
No initrd
## Transferring control to Linux (at address 80060000) ...
## Giving linux memsize in bytes, 134217728

Starting kernel ...

[    0.000000] Linux version 5.10.161 (gluonbuilder@atlas) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20044+13-ddeeb35007) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Tue Mar 22 20:06:42 2022
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is Ubiquiti UniFi AC Mesh
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
```

It shows the same output when trying via ram boot (yes, I know that it isn't supposed to work... but it should work a little bit better than what is shown earlier):

```
       setenv serverip 192.168.112.232
       setenv ipaddr 192.168.112.123
       tftpboot  0x81000000 ubnt_unifiac-mesh-squashfs-sysupgrade.bin && bootm $fileaddr
```

The "sysupgrade ramboot" hack works better (tm) with the 22.03.3 image from OpenWrt.

Next steps: build with the actual OpenWrt 22.03.3 in the gluon modules. If this works then it is bisect time.

ecsv, 2023-02-06:
Just for full transparency: The gluon bisect resulted in following commit

```
b85a9c54d5db6e66df8f1564461689c8c2865dfb is the first bad commit
commit b85a9c54d5db6e66df8f1564461689c8c2865dfb
Author: Andreas Ziegler <dev@andreas-ziegler.de>
Date:   Fri Dec 30 00:45:48 2022 +0100

    modules: update openwrt
    
    6c3e16055a kernel: backport ksmbd security fix ZDI-22-1690
    275c51bf40 Revert "image-commands.mk: Be consistent in command invocation"
    fec4fb3a56 sunxi: remove frequency for NanoPi R1
    51a763ab0f arm-trusted-firmware-sunxi: drop CPE ID
    377d805887 trusted-firmware-a.mk: use correct CPE ID
    f30414c56d kernel: bump 5.10 to 5.10.160
    47ac52a7c4 ath79: image: don't depend on other COMPILE targets
    da79da3ac1 image: fix device profile specific COMPILE targets
    9aa95bbebb rules: fix broken commitcount on alpine system
    fcff234fd8 image-commands.mk: Be consistent in command invocation
    be7b97d873 sunxi: fix typo in device packages for MarsBoard A10
    709351bf32 kernel: Make KERNEL_MAKEOPTS recursively expanded
    3b066a6581 lantiq: vr9: include usb driver for fritz 7430
    39dcfcd987 wolfssl: fix build with /dev/crypto
    e51f3a22aa toolchain: Update glibc 2.34 to recent HEAD
    298d2d0412 Revert "mbedtls: import patch to fix illegal instruction on mpc85xx"
    3de4572ed3 tools/ccache: fix build with musl and gcc 12
    8e43a44f08 rpcd: update to latest Git HEAD
    f43ceb60d5 iwinfo: update to latest Git HEAD
    31baecb70e iwinfo: update to the latest version
    f34c4a950e iwinfo: update to the latest version
    9b7a3e1a5e iwinfo: update to the latest version
    ce881873c0 kernel: add missing dependency to KERNEL_RPI_AXIPERF
    7ccb4c6651 kernel: add missing symbol for bcm27xx
    b18a0d0b92 generic: add support for EON EN25QX128A spi nor flash
    d909fb1d3b kernel: bump 5.10 to 5.10.159
    e5cec69484 kernel: bump 5.10 to 5.10.158
    3a58bda06b kernel: bump 5.10 to 5.10.157
    ee2341cac6 kernel: add symbol in generic config for 5.10.157
    bc64ca15d4 kernel: kmod-net-rtl8192su: Remove package
    b1204ce28c umbim: Allow roaming and partner connections
    15bc6b23c7 kernel: further cleanup of xfrm[4|6]_mode*
    6e396eff1a kernel: remove handling of xfrm[4|6]_mode_* modules
    78fc12ddd0 kernel: fix typo for tegra crypto-sha1 module
    4408f7d564 kernel: add kmod-btsdio package
    6b2c445b88 kernel: remove CONFIG_MMC_BLOCK_BOUNCE
    aef33ab13a ath79: fix Teltonika RUT230 v1 MAC assignment
    6cd1035d8d ramips: add support for Wavlink WS-WN572HP3 4G
    3f7ce4bd40 comgt-ncm: add support for quectel modem EC200T-EU
    03ab5a60da realtek: update GPIO bindings for DGS-1210-10P
    63fde4607c ustream-ssl: update to Git version 2022-12-07
    1c174fe0ba CI: kernel: don't checkout and install feeds
    385f78780c CI: build: skip sdk adapt to external toolchain on cache hit
    8be50c369b CI: build: fix use of sdk as toolchain
    1af58a2d39 ramips: mt7621: enable lzma-loader for ZyXEL WAP6805
    8fbf03690e ramips: mt7621: enable lzma-loader for netis WF2881
    282dd271ad kernel: ca8210: Fix crash by zero initializing data
    1a0f2c3a3e e2fsprogs: Fix CVE-2022-1304
    1285938d2c bcm4908: Refresh kernel patches
    eb639e54fc ucode: update to latest Git HEAD
    9b6dcf322b mvebu: disable also wrt32x due to broken switch
    69ed96fba8 kernel: netconsole: add network console logging support
    1f9bf1ee82 kernel: kmod-isdn4linux: Remove package
    a3dee42206 kernel: kmod-w1-slave-ds2760: Remove package
    2a1bf860f3 kernel: kmod-rtc-pt7c4338: Remove package
    a0bae2fef8 mvebu: cortexa9: disable devices using broken mv88e6176 switch
    309a6f22ba CI: trigger check also on build and check-kernel-patches workflow change
    ec9ca325c0 CI: build: fix matching for openwrt release branch for toolchain parsing
    63a2ea5470 CI: fix matching for openwrt release branch for container selection
    9db78a7e86 CI: build: add support to fallback to sdk for external toolchain
[... 203 lines cut ...]
 modules                                                |  2 +-
 ...ps-mt7621-make-DSA-images-swconfig-upgradable.patch | 18 +++++++++---------
 ...iq-xrx200-make-DSA-images-swconfig-upgradable.patch |  2 +-
 ...-Querier-wake-up-calls-Android-bug-workaround.patch |  4 ++--
 4 files changed, 13 insertions(+), 13 deletions(-)
```

This is the bisect log:

```
# bad: [2751daa655c8adae66ffd2c004a4fca060ae1d43] Merge pull request #2773 from blocktrron/v2022.1.2-release-notes
# good: [ab1fb054f6be33fdba74c3f3344d9ef746287c68] Merge pull request #2670 from blocktrron/v2022.1.1-rn
git bisect start 'v2022.1.2' 'v2022.1.1'
# bad: [9b840d1d8755b61aa590d4c9816fb466bdef776c] Merge pull request #2740 from freifunk-gluon/v2022.1.x-modules-bump
git bisect bad 9b840d1d8755b61aa590d4c9816fb466bdef776c
# good: [940874972b207b162292ea31763560791e5db913] Merge pull request #2715 from freifunk-gluon/backport-2705-to-v2022.1.x
git bisect good 940874972b207b162292ea31763560791e5db913
# good: [8c1f5e9c9cd521953493a093af2a6324ecb4f1b8] Merge pull request #2685 from T-X/v2022.1.x-pr-backport-bridge-mcast-wakeupcall
git bisect good 8c1f5e9c9cd521953493a093af2a6324ecb4f1b8
# bad: [c82fc9616bfccf43cb466ed3d8a894b0db199273] modules: update packages
git bisect bad c82fc9616bfccf43cb466ed3d8a894b0db199273
# bad: [b85a9c54d5db6e66df8f1564461689c8c2865dfb] modules: update openwrt
git bisect bad b85a9c54d5db6e66df8f1564461689c8c2865dfb
# first bad commit: [b85a9c54d5db6e66df8f1564461689c8c2865dfb] modules: update openwrt
```

ecsv, 2023-02-07:
In OpenWrt, the first bad commit is:

```
f30414c56db372acbd32d4cf77a2d0164c3e2b71 is the first bad commit
commit f30414c56db372acbd32d4cf77a2d0164c3e2b71
Author: John Audia <therealgraysky@proton.me>
Date:   Mon Dec 19 09:18:02 2022 -0500

    kernel: bump 5.10 to 5.10.160
    
    No patches affected by this update.
    
    Build system: x86_64
    Build-tested: ramips/tplink_archer-a6-v3
    Run-tested: ramips/tplink_archer-a6-v3
    
    Signed-off-by: John Audia <therealgraysky@proton.me>
    (cherry picked from commit 1e375c8fbd6a3259ffbbbde13eb0c106bedbcd1c)

 include/kernel-5.10 | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)
```

Bisect log:

```
# bad: [6c3e16055ae5d79205812e8436110fd862266c94] kernel: backport ksmbd security fix ZDI-22-1690
# good: [00d7702796d922e4258b7acb6e6b88a93071eebe] mac80211: Update to version 5.15.74-1
git bisect start '6c3e16055ae5d79205812e8436110fd862266c94' '00d7702796d922e4258b7acb6e6b88a93071eebe'
# good: [119c6573c49ec1ca31b0fac56aa41120315fe229] CI: usability improvements for tools
git bisect good 119c6573c49ec1ca31b0fac56aa41120315fe229
# good: [1a0f2c3a3ecb19f65bdc269818ed373958d032e6] e2fsprogs: Fix CVE-2022-1304
git bisect good 1a0f2c3a3ecb19f65bdc269818ed373958d032e6
# good: [b18a0d0b92963ad184ca309b93678946c09d7e99] generic: add support for EON EN25QX128A spi nor flash
git bisect good b18a0d0b92963ad184ca309b93678946c09d7e99
# good: [3b066a6581cc4291d261a672663630fce7f4e66f] lantiq: vr9: include usb driver for fritz 7430
git bisect good 3b066a6581cc4291d261a672663630fce7f4e66f
# good: [47ac52a7c443ef065c9a35db34712e382d364986] ath79: image: don't depend on other COMPILE targets
git bisect good 47ac52a7c443ef065c9a35db34712e382d364986
# bad: [51a763ab0fd0b2d5952d3c7f5c9943c81eb6528a] arm-trusted-firmware-sunxi: drop CPE ID
git bisect bad 51a763ab0fd0b2d5952d3c7f5c9943c81eb6528a
# bad: [377d8058879524d07231a11da0e5ccfeed0424e2] trusted-firmware-a.mk: use correct CPE ID
git bisect bad 377d8058879524d07231a11da0e5ccfeed0424e2
# bad: [f30414c56db372acbd32d4cf77a2d0164c3e2b71] kernel: bump 5.10 to 5.10.160
git bisect bad f30414c56db372acbd32d4cf77a2d0164c3e2b71
# first bad commit: [f30414c56db372acbd32d4cf77a2d0164c3e2b71] kernel: bump 5.10 to 5.10.160
```

ecsv, 2023-02-07:
First bad commit in the kernel:

```
965d93fb39b99348d6c327853afd4708b610e132 is the first bad commit
commit 965d93fb39b99348d6c327853afd4708b610e132
Author: Amir Goldstein <amir73il@gmail.com>
Date:   Tue Dec 13 15:13:41 2022 +0200

    vfs: fix copy_file_range() averts filesystem freeze protection
    
    commit 10bc8e4af65946b727728d7479c028742321b60a upstream.
    
    [backport comments for pre v5.15:
    - ksmbd mentions are irrelevant - ksmbd hunks were dropped
    - sb_write_started() is missing - assert was dropped
    ]
    
    Commit 868f9f2f8e00 ("vfs: fix copy_file_range() regression in cross-fs
    copies") removed fallback to generic_copy_file_range() for cross-fs
    cases inside vfs_copy_file_range().
    
    To preserve behavior of nfsd and ksmbd server-side-copy, the fallback to
    generic_copy_file_range() was added in nfsd and ksmbd code, but that
    call is missing sb_start_write(), fsnotify hooks and more.
    
    Ideally, nfsd and ksmbd would pass a flag to vfs_copy_file_range() that
    will take care of the fallback, but that code would be subtle and we got
    vfs_copy_file_range() logic wrong too many times already.
    
    Instead, add a flag to explicitly request vfs_copy_file_range() to
    perform only generic_copy_file_range() and let nfsd and ksmbd use this
    flag only in the fallback path.
    
    This choise keeps the logic changes to minimum in the non-nfsd/ksmbd code
    paths to reduce the risk of further regressions.
    
    Fixes: 868f9f2f8e00 ("vfs: fix copy_file_range() regression in cross-fs copies")
    Tested-by: Namjae Jeon <linkinjeon@kernel.org>
    Tested-by: Luis Henriques <lhenriques@suse.de>
    Signed-off-by: Amir Goldstein <amir73il@gmail.com>
    Signed-off-by: Al Viro <viro@zeniv.linux.org.uk>
    Signed-off-by: Amir Goldstein <amir73il@gmail.com>
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>

 fs/nfsd/vfs.c      |  4 ++--
 fs/read_write.c    | 17 +++++++++++++----
 include/linux/fs.h |  8 ++++++++
 3 files changed, 23 insertions(+), 6 deletions(-)
```

Bisect log:

```
# bad: [a2428a8dcb4f3eb80e7d38dba0bf71e4ff20cecd] Linux 5.10.160
# good: [931578be69875087a62524da69964d575426d287] Linux 5.10.159
git bisect start 'v5.10.160' 'v5.10.159'
# bad: [344739dc56f1b3e33e6a3170b89731d450455df6] ASoC: fsl_micfil: explicitly clear CHnF flags
git bisect bad 344739dc56f1b3e33e6a3170b89731d450455df6
# bad: [eb6313c12955c58c3d3d40f086c22e44ca1c9a1b] nfp: fix use-after-free in area_cache_get()
git bisect bad eb6313c12955c58c3d3d40f086c22e44ca1c9a1b
# good: [ed9673394979b7a5dff10ba878178054625beda9] vfs: fix copy_file_range() regression in cross-fs copies
git bisect good ed9673394979b7a5dff10ba878178054625beda9
# bad: [965d93fb39b99348d6c327853afd4708b610e132] vfs: fix copy_file_range() averts filesystem freeze protection
git bisect bad 965d93fb39b99348d6c327853afd4708b610e132
# first bad commit: [965d93fb39b99348d6c327853afd4708b610e132] vfs: fix copy_file_range() averts filesystem freeze protection
```

ecsv, 2023-02-07:
This is the kernel config diff for the OpenWrt (working) and gluon (not-working) for the same OpenWrt version (f30414c56db372acbd32d4cf77a2d0164c3e2b71)

Not sure how to start - most likely somewhere near the filesystems (like CONFIG_FS_IOMAP)

<details> 

```
--- <unnamed>
+++ <unnamed>
@@ -105,8 +105,7 @@
 CONFIG_TASKS_TRACE_RCU=y
 # end of RCU Subsystem
 
-CONFIG_IKCONFIG=m
-CONFIG_IKCONFIG_PROC=y
+# CONFIG_IKCONFIG is not set
 # CONFIG_IKHEADERS is not set
 CONFIG_LOG_BUF_SHIFT=17
 CONFIG_PRINTK_SAFE_LOG_BUF_SHIFT=13
@@ -117,7 +116,25 @@
 #
 # end of Scheduler features
 
-# CONFIG_CGROUPS is not set
+CONFIG_CGROUPS=y
+CONFIG_PAGE_COUNTER=y
+CONFIG_MEMCG=y
+CONFIG_MEMCG_SWAP=y
+CONFIG_MEMCG_KMEM=y
+CONFIG_BLK_CGROUP=y
+CONFIG_CGROUP_WRITEBACK=y
+CONFIG_CGROUP_SCHED=y
+CONFIG_FAIR_GROUP_SCHED=y
+CONFIG_CFS_BANDWIDTH=y
+CONFIG_RT_GROUP_SCHED=y
+CONFIG_CGROUP_PIDS=y
+CONFIG_CGROUP_RDMA=y
+# CONFIG_CGROUP_FREEZER is not set
+# CONFIG_CGROUP_DEVICE is not set
+CONFIG_CGROUP_CPUACCT=y
+CONFIG_CGROUP_BPF=y
+# CONFIG_CGROUP_DEBUG is not set
+CONFIG_SOCK_CGROUP_DATA=y
 CONFIG_NAMESPACES=y
 CONFIG_UTS_NS=y
 CONFIG_IPC_NS=y
@@ -127,9 +144,11 @@
 # CONFIG_CHECKPOINT_RESTORE is not set
 # CONFIG_SCHED_AUTOGROUP is not set
 # CONFIG_SYSFS_DEPRECATED is not set
-CONFIG_RELAY=y
+# CONFIG_RELAY is not set
 CONFIG_BLK_DEV_INITRD=y
-CONFIG_INITRAMFS_SOURCE=""
+CONFIG_INITRAMFS_SOURCE="/home/sven/tmp/openwrt/build_dir/target-mips_24kc_musl/root-ath79 /home/sven/tmp/openwrt/target/linux/generic/image/initramfs-base-files.txt"
+CONFIG_INITRAMFS_ROOT_UID=1000
+CONFIG_INITRAMFS_ROOT_GID=1000
 # CONFIG_RD_GZIP is not set
 # CONFIG_RD_BZIP2 is not set
 # CONFIG_RD_LZMA is not set
@@ -137,6 +156,7 @@
 # CONFIG_RD_LZO is not set
 # CONFIG_RD_LZ4 is not set
 # CONFIG_RD_ZSTD is not set
+CONFIG_INITRAMFS_COMPRESSION_NONE=y
 # CONFIG_BOOT_CONFIG is not set
 CONFIG_CC_OPTIMIZE_FOR_PERFORMANCE=y
 # CONFIG_CC_OPTIMIZE_FOR_SIZE is not set
@@ -150,7 +170,7 @@
 CONFIG_MULTIUSER=y
[... 5245 lines cut ...]
 CONFIG_GENERIC_GETTIMEOFDAY=y
-CONFIG_SG_POOL=y
 CONFIG_SBITMAP=y
 # CONFIG_STRING_SELFTEST is not set
 # end of Library routines
@@ -6574,7 +3290,6 @@
 # CONFIG_INTERVAL_TREE_TEST is not set
 # CONFIG_PERCPU_TEST is not set
 # CONFIG_ATOMIC64_SELFTEST is not set
-# CONFIG_ASYNC_RAID6_TEST is not set
 # CONFIG_TEST_HEXDUMP is not set
 # CONFIG_TEST_STRING_HELPERS is not set
 # CONFIG_TEST_STRSCPY is not set
@@ -6591,7 +3306,7 @@
 # CONFIG_TEST_BITOPS is not set
 # CONFIG_TEST_VMALLOC is not set
 # CONFIG_TEST_USER_COPY is not set
-CONFIG_TEST_BPF=m
+# CONFIG_TEST_BPF is not set
 # CONFIG_TEST_BLACKHOLE_DEV is not set
 # CONFIG_FIND_BIT_BENCHMARK is not set
 # CONFIG_TEST_FIRMWARE is not set

```
</details>

ecsv, 2023-02-07:
With following openwrt default config, it is also possible to reproduce it:

```
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_ubnt_unifiac-mesh=y
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
CONFIG_COLLECT_KERNEL_DEBUG=y
CONFIG_FASTD_ENABLE_CIPHER_NULL=y
CONFIG_FASTD_ENABLE_CIPHER_SALSA2012=y
CONFIG_FASTD_ENABLE_MAC_UHASH=y
CONFIG_FASTD_ENABLE_METHOD_COMPOSED_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_GENERIC_UMAC=y
CONFIG_FASTD_ENABLE_METHOD_NULL=y
CONFIG_FASTD_ENABLE_METHOD_NULL_L2TP=y
CONFIG_FASTD_WITH_OFFLOAD_L2TP=y
CONFIG_FASTD_WITH_STATUS_SOCKET=y
# CONFIG_FEED_ffffm_packages is not set
# CONFIG_FEED_gluon is not set
# CONFIG_FEED_gluon_base is not set
# CONFIG_FEED_gluon_ssid_changer is not set
CONFIG_GLUON_AUTOUPDATER_BRANCH="experimental"
CONFIG_GLUON_AUTOUPDATER_ENABLED=y
CONFIG_GLUON_MULTIDOMAIN=y
CONFIG_GLUON_RELEASE="b20230207-exp"
CONFIG_GLUON_SITEDIR="/home/sven/tmp/gluon/site"
CONFIG_GLUON_SITE_VERSION="b20230130-1-g69cb433"
CONFIG_GLUON_VERSION="v2022.1.1-22-g16c9e6c"
CONFIG_GLUON_WEB_LANG_de=y
# CONFIG_KERNEL_AIO is not set
# CONFIG_KERNEL_CGROUPS is not set
# CONFIG_KERNEL_FANOTIFY is not set
# CONFIG_KERNEL_FHANDLE is not set
# CONFIG_KERNEL_IO_URING is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
# CONFIG_KERNEL_IPV6_SEG6_LWTUNNEL is not set
# CONFIG_KERNEL_IP_MROUTE is not set
CONFIG_KERNEL_PROC_STRIPPED=y
# CONFIG_KERNEL_SECCOMP is not set
CONFIG_PACKAGE_ATH_DEBUG=y
CONFIG_PACKAGE_ath10k-firmware-qca988x=m
CONFIG_PACKAGE_autoupdater=m
CONFIG_PACKAGE_ca-bundle=m
CONFIG_PACKAGE_ebtables-tiny=m
CONFIG_PACKAGE_fastd=m
CONFIG_PACKAGE_ffffm-ath9k-broken-wifi-workaround=m
CONFIG_PACKAGE_firewall=y
CONFIG_PACKAGE_firewall4=m
CONFIG_PACKAGE_gluon-authorized-keys=m
CONFIG_PACKAGE_gluon-autoupdater=m
CONFIG_PACKAGE_gluon-client-bridge=m
CONFIG_PACKAGE_gluon-config-mode-autoupdater=m
CONFIG_PACKAGE_gluon-config-mode-contact-info=m
CONFIG_PACKAGE_gluon-config-mode-core=m
CONFIG_PACKAGE_gluon-config-mode-domain-select=m
CONFIG_PACKAGE_gluon-config-mode-geo-location=m
CONFIG_PACKAGE_gluon-config-mode-geo-location-osm=m
CONFIG_PACKAGE_gluon-config-mode-hostname=m
CONFIG_PACKAGE_gluon-config-mode-mesh-vpn=m
CONFIG_PACKAGE_gluon-config-mode-outdoor=m
CONFIG_PACKAGE_gluon-config-mode-theme=m
CONFIG_PACKAGE_gluon-core=y
CONFIG_PACKAGE_gluon-ebtables=m
CONFIG_PACKAGE_gluon-ebtables-filter-multicast=m
CONFIG_PACKAGE_gluon-ebtables-filter-ra-dhcp=m
CONFIG_PACKAGE_gluon-ebtables-limit-arp=m
CONFIG_PACKAGE_gluon-lock-password=m
CONFIG_PACKAGE_gluon-mesh-batman-adv-15=m
CONFIG_PACKAGE_gluon-mesh-vpn-core=m
CONFIG_PACKAGE_gluon-mesh-vpn-fastd=m
CONFIG_PACKAGE_gluon-mesh-vpn-fastd-l2tp=m
CONFIG_PACKAGE_gluon-neighbour-info=m
CONFIG_PACKAGE_gluon-node-info=m
CONFIG_PACKAGE_gluon-private-wifi=m
CONFIG_PACKAGE_gluon-radv-filterd=m
CONFIG_PACKAGE_gluon-radvd=m
CONFIG_PACKAGE_gluon-respondd=m
CONFIG_PACKAGE_gluon-setup-mode=m
CONFIG_PACKAGE_gluon-site=y
CONFIG_PACKAGE_gluon-ssid-changer=m
CONFIG_PACKAGE_gluon-status-page=m
CONFIG_PACKAGE_gluon-status-page-mesh-batman-adv=m
CONFIG_PACKAGE_gluon-wan-dnsmasq=m
CONFIG_PACKAGE_gluon-web=m
CONFIG_PACKAGE_gluon-web-admin=m
CONFIG_PACKAGE_gluon-web-autoupdater=m
CONFIG_PACKAGE_gluon-web-model=m
CONFIG_PACKAGE_gluon-web-network=m
CONFIG_PACKAGE_gluon-web-osm=m
CONFIG_PACKAGE_gluon-web-private-wifi=m
CONFIG_PACKAGE_gluon-web-wifi-config=m
CONFIG_PACKAGE_hostapd-mini=m
CONFIG_PACKAGE_ip6tables-zz-legacy=y
CONFIG_PACKAGE_jansson=m
CONFIG_PACKAGE_kmod-crypto-crc32c=m
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-nat=y
# CONFIG_PACKAGE_kmod-ipt-offload is not set
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-lib-crc-ccitt=m
CONFIG_PACKAGE_kmod-lib-crc32c=m
CONFIG_PACKAGE_kmod-nf-flow=m
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nfnetlink=m
CONFIG_PACKAGE_kmod-nft-core=m
CONFIG_PACKAGE_kmod-nft-fib=m
CONFIG_PACKAGE_kmod-nft-nat=m
CONFIG_PACKAGE_kmod-nft-offload=m
CONFIG_PACKAGE_kmod-ppp=m
CONFIG_PACKAGE_kmod-pppoe=m
CONFIG_PACKAGE_kmod-pppox=m
CONFIG_PACKAGE_kmod-slhc=m
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-vxlan=y
CONFIG_PACKAGE_libbatadv=m
CONFIG_PACKAGE_libecdsautil=m
CONFIG_PACKAGE_libgluonutil=y
CONFIG_PACKAGE_libip4tc=y
CONFIG_PACKAGE_libip6tc=y
CONFIG_PACKAGE_libiptext=y
CONFIG_PACKAGE_libiptext6=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_libmnl=m
CONFIG_PACKAGE_libnftnl=m
CONFIG_PACKAGE_libpacketmark=m
CONFIG_PACKAGE_libplatforminfo=y
CONFIG_PACKAGE_libubus-lua=m
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libucode=m
CONFIG_PACKAGE_libuecc=m
CONFIG_PACKAGE_libxtables=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_lua-bit32=y
CONFIG_PACKAGE_lua-hash=y
CONFIG_PACKAGE_lua-jsonc=y
CONFIG_PACKAGE_lua-platform-info=y
CONFIG_PACKAGE_lua-simple-uci=y
CONFIG_PACKAGE_luabitop=y
CONFIG_PACKAGE_luaposix=y
CONFIG_PACKAGE_micrond=m
CONFIG_PACKAGE_nftables-json=m
CONFIG_PACKAGE_odhcpd-ipv6only=m
CONFIG_PACKAGE_ppp=m
CONFIG_PACKAGE_ppp-mod-pppoe=m
CONFIG_PACKAGE_pretty-hostname=y
CONFIG_PACKAGE_respondd=m
CONFIG_PACKAGE_respondd-module-airtime=m
CONFIG_PACKAGE_simple-tc=m
CONFIG_PACKAGE_sse-multiplex=m
CONFIG_PACKAGE_ucode=m
CONFIG_PACKAGE_ucode-mod-fs=m
CONFIG_PACKAGE_ucode-mod-ubus=m
CONFIG_PACKAGE_ucode-mod-uci=m
CONFIG_PACKAGE_uhttpd=m
CONFIG_PACKAGE_uradvd=m
CONFIG_PACKAGE_vxlan=y
CONFIG_PACKAGE_wpad-basic-wolfssl=m
CONFIG_PACKAGE_xtables-legacy=y
# CONFIG_SECCOMP is not set
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
```

AiyionPrime, 2023-02-13:
fml.
I encountered this issue three? weeks ago and was under the impression it was a problem with the Librerouter I'm testing in #2654 with @T-X.

rotanid, 2023-02-13:
@ecsv are you sure about the other AC-* devices?
i tested AC LR and at least this device does *not* have the issue.
i may be able to test AC Pro later this week, AC Lite would be more difficult but not impossible.

@AiyionPrime what makes you think your issues are related and what exactly is your issue? maybe describe it in #2654, currently very vague over there.

AiyionPrime, 2023-02-13:
Wrote @rotanid earlier; I pasted my boot-issue in hackint#gluon a few days ago towards @T-X, I think.

The LibreRouter v1 does not boot reliably, but from time to time.
And if it doesn't, it hangs at the very same line @ecsv logged above in the first post.

```[    0.000000] Inode-cache hash table entries: [...]```

I'll describe my findings on Wednesday evening on the other PR as @rotanid asked.

For this issue I'll recommend cold and warm booting the devices in question though, in order to see, whether your boot behavior might be as unstable as the one I currently find.

Just want to make sure we do not rule out other affected devices as working to early.

Djfe, 2023-02-14:
I ran across the same kernel freeze issue on the tp-link archer c7 in January and still had trouble reproducing it reliably with openwrt. It happened on two different Archer C7 v2.
warm boot mostly worked, while cold boot didn't, atleast when wan/lan was plugged in.
If I unplugged wan, then the boot went through successfully past
`[    0.000000] Inode-cache hash table entries: [...]`

Tried to build master with kernel debugging to find the cause but it always booted successfully.
Likely because [my config was different 
](https://github.com/freifunk-gluon/gluon/issues/2784#issuecomment-1420495594) 

I'm wondering whether it breaks on kernel 5.15 with @ecsv 's config. with my master build and a different config it didn' t break. (same for default openwrt build, which is why it probably never surfaced upstream)

I guess I could try to confirm his bisect for my device.

AiyionPrime, 2023-02-14:
@Djfe Your described problem resulted in a hang directly after said line as well?

> If I unplugged wan, then the boot went through successfully past
> `[    0.000000] Inode-cache hash table entries: [...]`

Djfe, 2023-02-14:
exactly, every time it froze (once I had connected it via ttl), it froze right on that line. never earlier never later

Djfe, 2023-02-14:
The following logs were not recorded consecutively.
I forgot enabling logging for the run with the 3 consecutive attempts that failed. 🙈

I also have successful logs of a full boot and of booting to config mode, in-case they are relevant.
```
U-Boot 1.1.4 (Jun 16 2016 - 12:02:34)

ap135 - Scorpion 1.0DRAM:  
sri
Scorpion 1.0
ath_ddr_initial_config(178): (32bit) ddr2 init
tap = 0x00000003
Tap (low, high) = (0x4, 0x1e)
Tap values = (0x11, 0x11, 0x11, 0x11)
128 MB
Flash Manuf Id 0xef, DeviceId0 0x40, DeviceId1 0x18
flash size 16MB, sector count = 256
Flash: 16 MB
Using default environment

*** Warning *** : PCIe WLAN Module not found !!!
In:    serial
Out:   serial
Err:   serial
Net:   ath_gmac_enet_initialize...
athrs_sgmii_res_cal: cal value = 0xe
No valid address in Flash. Using fixed address
No valid address in Flash. Using fixed address
ath_gmac_enet_initialize: reset mask:c02200 
Scorpion  ----> S17 PHY *
Vlan config...
TEST: FINAL REG VAL after TX Calibration - 0x46000000
TEST: FINAL XMII VAL after RX Calibration - 0x56000000
TEST: FINAL ETH_CFG VAL after RX Calibration - 0x00028001
athrs17_reg_init: complete
: cfg1 0x80000000 cfg2 0x7335
eth0: ba:be:fa:ce:08:41
eth0 up
athrs17_reg_init_wan done
SGMII in forced mode
athr_gmac_sgmii_setup SGMII done
: cfg1 0x800c0000 cfg2 0x7214
eth1: ba:be:fa:ce:08:41
eth1 up
eth0, eth1
Setting 0x18116290 to 0x58b1214f
Autobooting in 1 seconds
## Booting image at 9f020000 ...
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.160 (florian@ffac-community-build) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20014+9-6c3e16055a) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Tue Dec 22 03:35:17 2020
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
```

```
U-Boot 1.1.4 (Jun 16 2016 - 12:02:34)

ap135 - Scorpion 1.0DRAM:  
sri
Scorpion 1.0
ath_ddr_initial_config(178): (32bit) ddr2 init
tap = 0x00000003
Tap (low, high) = (0x4, 0x1f)
Tap values = (0x11, 0x11, 0x11, 0x11)
128 MB
Flash Manuf Id 0xef, DeviceId0 0x40, DeviceId1 0x18
flash size 16MB, sector count = 256
Flash: 16 MB
Using default environment

*** Warning *** : PCIe WLAN Module not found !!!
In:    serial
Out:   serial
Err:   serial
Net:   ath_gmac_enet_initialize...
athrs_sgmii_res_cal: cal value = 0xe
No valid address in Flash. Using fixed address
No valid address in Flash. Using fixed address
ath_gmac_enet_initialize: reset mask:c02200 
Scorpion  ----> S17 PHY *
Vlan config...
TEST: FINAL REG VAL after TX Calibration - 0x46000000
TEST: FINAL XMII VAL after RX Calibration - 0x56000000
TEST: FINAL ETH_CFG VAL after RX Calibration - 0x00028001
athrs17_reg_init: complete
: cfg1 0x80000000 cfg2 0x7335
eth0: ba:be:fa:ce:08:41
eth0 up
athrs17_reg_init_wan done
SGMII in forced mode
athr_gmac_sgmii_setup SGMII done
: cfg1 0x800c0000 cfg2 0x7214
eth1: ba:be:fa:ce:08:41
eth1 up
eth0, eth1
Setting 0x18116290 to 0x58b1214f
Autobooting in 1 seconds
## Booting image at 9f020000 ...
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 5.10.160 (florian@ffac-community-build) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 11.2.0 r20014+9-6c3e16055a) 11.2.0, GNU ld (GNU Binutils) 2.37) #0 Tue Dec 22 03:35:17 2020
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v2
[    0.000000] SoC: Qualcomm Atheros QCA9558 ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
```

AiyionPrime, 2023-02-15:
On the LibreRouter (unstable) the next line in the bootlog would be:

`[    0.000000] Writing ErrCtl register=00000000`

rotanid, 2023-02-20:
at the Gluon meeting last week we decided to go with the same workaround @ecsv applied for FreifunkVogtland ( https://github.com/FreifunkVogtland/gluon/commit/40d54258fc550092e3df6fec1fb012cc398decf9 ) until we find a better solution

blocktrron, 2023-02-21:
@ecsv As I've just came back (and to be sure i don't waste time looking at the wrong version) - You built gluon from the tag commit of v2022.1.2 (2751daa655c8adae66ffd2c004a4fca060ae1d43)

rotanid, 2023-02-21:
at least i did, yes - without and with workaround.

ecsv, 2023-02-21:
@blocktrron Yes. you can also have a look at the bisect logs for the various versions I've tested (and the results I had)

blocktrron, 2023-02-23:
I've did some tests - From my testing it indicates there's a difference depending on the SoC type used, while it worked on no QCA955x / QCA956x board, older devices (Siemens WS-AP3610 / TL-WR1043v1) work just fine.

The difference here is the CPU (24Kc / 74Kc) as well as the system memory (DDR1/DDR2).

As none of the failing boards I've had at hand offered a jtag connector, I've tried around with the kernel start procedure and did make some observations.

1. Adding a message just before the call to `vfs_caches_init_early` fixes the freeze for me
2. Adding a message before `trap_init` leads to the kernel freezing after the message.

I'll see if we still have a Archer C7 v2 in storage, as it is the only QCA955x / 74Kc board i know of that has a debug port.

blocktrron, 2023-02-23:
dropping this patch from OpenWrt also fixes the boot-process:

https://github.com/openwrt/openwrt/blob/master/target/linux/generic/pending-5.10/308-mips32r2_tune.patch

rotanid, 2023-02-24:
removing the issue from the milestones as workarounds have been implemented.

blocktrron, 2023-02-28:
For the record - The same kernel boots on the Archer C7 v2 without anomalies, so it is probably not an issue related with the CPU architecture (24kc - 74kc).

Djfe, 2023-02-28:
Which Kernel? (before the workaround?)

blocktrron, 2023-02-28:
The same one which is consistently failing on the AC Mesh.

Can you provide the image you see sporadously failing on your C7 v2? I was only able to reproduce it on a OCEDO Koala (which again does not offer a debugging interface).

Djfe, 2023-02-28:
will do, until then:
Pls try a cold boot while wan and lan1 are connected.
The issue never manifested for me with no wan connected, I think.
Once I unplugged wan, it booted successfully.

blocktrron, 2023-03-01:
I did experiment with this further.

Where my ac lite consistently hangs is with the call to `tlbmiss_handler_setup_pgd`. This function is generated by the kernel itself in `build_setup_pgd` where the icache is cleared in `flush_tlb_handlers` subsequently.

However, the function memory itself is pre-compiled with an infinite loop (compare https://github.com/torvalds/linux/blob/master/arch/mips/mm/tlb-funcs.S#L20), sure enough, removing this infinite loop without altering the code leads to the AP booting 100% successfully (up from 0%).

One thing i did notice was the alteration of the functions length, which also reduced from 72 bytes to 64. I'm not familiar with MIPS caching, but is it possible that the icache can only be cleared with a multiple of the cache-line size? In that case, i would assume the sizing is a mistake, given the infinite loop was added later without altering the size of the function:

https://github.com/torvalds/linux/blob/master/arch/mips/mm/tlb-funcs.S#L20t

NeoRaider, 2023-03-01:
@blocktrron The function should flush all cache lines that overlap with the passed range (and I don't think the function start is aligned to 64 bytes anyways).

Djfe, 2023-03-01:
> Can you provide the image you see sporadously failing on your C7 v2? I was only able to reproduce it on a OCEDO Koala (which again does not offer a debugging interface).

Sorry for being a bit late :/ ^^
regardless, here is the image:
[gluon-ffac-v2022.1.11-tp-link-archer-c7-v2-sysupgrade.bin.zip](https://github.com/freifunk-gluon/gluon/files/10866167/gluon-ffac-v2022.1.11-tp-link-archer-c7-v2-sysupgrade.bin.zip)
I'll do some testing myself now.

blocktrron, 2023-03-02:
Maybe this patch is related? https://lore.kernel.org/lkml/20160301022738.GB12741@NP-P-BURTON/T/

blocktrron, 2023-03-02:
The explanation of the patch makes sense in our case. I did apply it to Gluon and in fact, i yet have to encounter a failed boot on my AC Lite.

This would also explain why the issue was not observed on a 24kc board, as only 74kc supports out of order execution. This still has a taste to it, as 24kc boards are arguably not as common nowadays as 74kc.

For reference, the patch to Gluon i did apply was this: https://github.com/blocktrron/gluon/commit/e6c23d6d22d1772acdb650f049323a14ae1676ca

@Djfe can you have a look?

Djfe, 2023-03-02:
wait the patch fixed the issue not introduced it? interesting
Sure will do. But first I have to reproduce the issue again in v2022.1.2 🙈
No clue, why it's booting flawlessly now or if something is different.
I have trouble getting my device into a kernel freeze at all now.

closed 2023-03-06 by blocktrron

action-server, 2023-12-27:
I have a spare [Arcadyan ARV7506PW11](https://openwrt.org/toh/arcadyan/arv7506pw11) that doesn't work with openwrt version 23 and up.
Serial console with u-boot shows following message:

> *** Warning - bad CRC, using default environment
> ...
> ...
> ...
> Starting kernel ...
> 
> [    0.000000] Linux version 5.15.145 (builder@buildhost) (mips-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24698-b799dd3c70) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Wed Dec 27 15:01:30 2023
> [    0.000000] SoC: Danube rev 1.5
> [    0.000000] printk: bootconsole [early0] enabled
> [    0.000000] CPU0 revision is: xxxxxxx (MIPS 24KEc)
> [    0.000000] MIPS: machine is Alice/O2 IAD 4421
> [    0.000000] Initrd not found or empty - disabling initrd

Will be using lower versions in the meantime

PR #2810 (2023-03-06): modules: update to latest OpenWrt HEAD
Fixes #2784