freifunk-gluon/gluon #2801: gluon-mesh-oslrd is missing respondd entry nodeinfo.software.oslrd

lemoer, 2023-02-26:
In order to identify oslr-firmwares as such, gluon-mesh-oslrd should implement a respondd provider for nodeinfo.software:

Batman-adv does that here:

https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-mesh-batman-adv/src/respondd-nodeinfo.c#L176-L182

lemoer, 2023-02-26:
@mkg20001, can you take this over?

mkg20001, 2023-02-26:
We have it downstream in https://github.com/ffgraz/gluon

I'm not going to do more than one upstream pr and it's on my todo list to
pr it upstream after clientap

The current solution downstream relies on openwrts uclient for fetching
data from httpinfo, which does not want to work properly without
forking/popen in respondd and then there's other issues