freifunk-gluon/gluon #2805: 7360v2: no configmode on Port 3,4

AiyionPrime, 2023-03-01:
### Bug report

**What is the problem?**

- What is not working as expected?

The Config Mode is not reachable on the ports three and four, but does on one and two.

- How is it misbehaving?

The Ports do work in normal operation, but not in Config Mode.

- When did the problem first start showing up?

The problem appears in current master and existed in [v2021.1.2](https://github.com/freifunk-gluon/gluon/tree/v2021.1.2) as well.

Possible the faulty behavior was just not spotted in its introduction to gluon in #1889.

- What were you doing when you first noticed the problem?

@bschelm was entering the Config Mode.
@grische and me verified the problem on a different device.

- On which devices (vendor, model and revision) is it misbehaving?

AVM FritzBox 7360 v2

- Does the issue appear on multiple devices or targets?

Haven't tested it on v1/SL, but would be worth a shot; the devices are not expected to differ much, excluding the memory.

**What is the expected behaviour?**
The configmode should work regardless of the driver the ports use.
Unlikely it ever worked on both ports.

**Gluon Version:**

master, gluon-v2022.1-268-g4484426

**Site Configuration:**

https://github.com/freifunkh/site/commit/58842a3ad2750b58d84d77978f17da507c48f6fb

**Custom patches:**
Three, but none related to lantiq, or even the device:

[0001-fix-when-vlan-crc-is-00000000.patch](https://github.com/freifunkh/site/blob/master/patches/0001-fix-when-vlan-crc-is-00000000.patch)
[0002-patches-uradvd-change-ip-lifetimes-to-150-300-seco](https://github.com/freifunkh/site/blob/master/patches/0002-patches-uradvd-change-ip-lifetimes-to-150-300-second.patch)
[0003-patches-openwrt-add-TL-WR841ND-N-8M-and-16M-variants.patch](https://github.com/freifunkh/site/blob/master/patches/0003-patches-openwrt-add-TL-WR841ND-N-8M-and-16M-variants.patch)

**Note**

as @grische found, ports one and two have a different driver:

```console
[    1.491010] gswip 1e108000.switch lan3 (uninitialized): PHY [1e108000.switch-mii:00] driver [Qualcomm Atheros AR8030] (irq=POLL)
[    1.504519] gswip 1e108000.switch lan4 (uninitialized): PHY [1e108000.switch-mii:01] driver [Qualcomm Atheros AR8030] (irq=POLL)
[    1.520087] gswip 1e108000.switch lan2 (uninitialized): PHY [1e108000.switch-mii:11] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
[    1.536986] gswip 1e108000.switch lan1 (uninitialized): PHY [1e108000.switch-mii:13] driver [Intel XWAY PHY11G (xRX v1.2 integrated)] (irq=POLL)
```

Djfe, 2023-05-17:
I tested 7360SL and 7360 V2 on v2022.1.4 and on master
Config mode is working on all ports for me

neocturne, 2025-03-25:
@AiyionPrime Do you know if this is still an issue with current Gluon main?

grische, 2025-08-12:
I was able to reproduce the issue with the same box and Gluon v2023.2.5. 

@AiyionPrime can we reopen the issue?

AiyionPrime, 2025-09-01:
@fsffunk appears to be unavailable?

I can ask @bschelm to lend me the v2 again and verify the problems existence on `v2023.2.5` and `main` on Thursday, if that would help.

@Djfe would you mind verifying `v2023.2.5` is broken on your device as well?

fsffunk, 2026-11-15:
Issue fixed on Gluon v2026.1-rc1