freifunk-gluon/gluon #2816: 2022.1.2 does not boot/blink on Archer C7v5, Ubiquiti AC Pro/Lite

smoe, 2023-03-17:
### Bug report

2022.1.2 works find with the Ubiquiti ERX EdgeRouter, Netgear r7800, TP-Link WDR3600 and 4300. However, the boot fails on TP-Link Archer C7v5, Ubiquit AC Pro and Ubiquiti AC Lite. No problems are observed with gluon-v2022.1-11-gcf939ad.

Our automated builds are flashing just fine. After the reboot however no LED blinks and nothing happens.

    $ git describe --always
    v2022.1.2

We have on patch for the sake of compatibility with out map:
```
--- a/package/gluon-respondd/files/etc/init.d/gluon-respondd
+++ b/package/gluon-respondd/files/etc/init.d/gluon-respondd
@@ -13,7 +13,7 @@ start_service() {
        local clientdevs=$(for dev in $(echo "$ifdump" | jsonfilter -e "@.interface[@.interface='$(cat /lib/gluon/respondd/client.dev 2>/dev/null)' && @.up=true].device"); do echo " -i $dev -t $MAXDELAY";done;)
 
        procd_open_instance
-       procd_set_param command $DAEMON -d /usr/lib/respondd -p 1001 -g ff02::2:1001 $meshdevs -g ff05::2:1001 $clientdevs
+       procd_set_param command $DAEMON -d /usr/lib/respondd -p 1001 -g ff02::2:1001 $meshdevs $clientdevs -g ff05::2:1001 $clientdevs
        procd_set_param respawn ${respawn_threshold:-3600} ${respawn_timeout:-5} ${respawn_retry:-5}
        procd_set_param stderr 1
        procd_close_instance
```

Many thanks and regards!

AiyionPrime, 2023-03-17:
@bschelm just noted he can't confirm the behavior on an Archer C7v5 using v2022.1.3

bschelm, 2023-03-17:
c7v5 seemt to reboot when i press the wifi button fpr longer than 5 seconds.
However, there is also a reset button and that seems to work as expected

blocktrron, 2023-03-17:
@smoe can you check if the boot works with a firmware compiled from https://github.com/freifunk-gluon/gluon/commit/ec57f6c6669af9052faf5e231b03c23763455870 ?

smoe, 2023-03-18:
Firmware is building. The installation is from 2015 or so, which sounds like old, but I'll ask the individual currently in possession of those lites about it.

smoe, 2023-03-19:
> Is the AC Lite your issue occured with a old one (glossy UBNT Logo, hard plastic, no 802.3af) or a new one (rubber case, 802.3af, printed logo)?

Glossy golden logo, hard plastic - but it has Power over Ethernet - the lite only has no pass-through, which the pro has, but you likely mean something else. Report on https://github.com/freifunk-gluon/gluon/commit/ec57f6c6669af9052faf5e231b03c23763455870 still pending.

rotanid, 2023-03-19:
> Glossy golden logo, hard plastic - but it has Power over Ethernet - the lite only has no pass-through, which the pro has, but you likely mean something else.

the difference is ubiquiti-vendor-"passive" PoE versus 802.3af standardized PoE

smoe, 2023-03-19:
@blocktrron  The referenced version https://github.com/freifunk-gluon/gluon/commit/ec57f6c6669af9052faf5e231b03c23763455870 of Gluon works fine. Many thanks!

smoe, 2023-03-23:
I want to close this issue but could someone please give some concluding remarks, like a sensible direction on what the latest reliable firmware for users of those devices shall be? We'll now continue using ec5f76c, but I cannot claim to have fully grasped what we are doing.

AiyionPrime, 2023-03-23:
@smoe You have verified ec57f6c works. That means gluons current master appears to be fine.

Version 2022.1.3 applies a workaround, which appears to be working for us in hannover, but feel free to verify that with your firmware before closing this issue.

I'd generally recommend using `2022.1.3` instead of `2022.1.2`.

closed 2023-04-11 by rotanid