freifunk-gluon/gluon #2824: Web status page no longer shows info for wireless mesh with newer OpenWrt

stweil, 2023-03-21:
### Bug report

Firmware which is based on Gluon next or other branches with newer OpenWrt no longer
shows wireless mesh information in the web status page of a node.

**What is the expected behaviour?**

That's a regression, and the previous behaviour is expected.

**Gluon Version:**

Tested with firmware based on branch `next`.

**Site Configuration:**

FFRN

**Custom patches:**

None.

stweil, 2023-03-21:
See also discussion for issue #2822.

The regression might be caused by OpenWrt commit https://github.com/openwrt/openwrt/commit/6eeb5d45649ffa591b06d2ccd6a43e47939b6b18.

blocktrron, 2023-03-22:
We do not track bugs with Gluons next branch. `next` is an unsupported tech-demo.

closed 2023-03-22 by blocktrron