freifunk-gluon/gluon #2839: 2021.1.2 Joy-IT OR750i can't sysupgrade

AiyionPrime, 2023-04-06:
### Bug report

**What is the problem?**

Sysupgrades immediately reboot the device, and silently fail.
The autoupdater therefore does not work either.

We encountered the problem, when we were looking for devices, that have not yet updated to `2022.1.3`, yet.

**What is the expected behaviour?**

Sysupgrades actually upgrade the device.

**Gluon Version:**
[v2021.1.2](https://github.com/freifunk-gluon/gluon/tree/v2021.1.2)

The problem is only apparent on `2021.1.x`, not on `master` or newer branches.

**Site Configuration:**
[vH25](https://github.com/freifunkh/site/commit/2538a3f56f2ba1fd6a9c93dfab5c171b68d4ab56)

**Custom patches:**

- [0001-fix-when-vlan-crc-is-00000000.patch](https://github.com/freifunkh/site/blob/2538a3f56f2ba1fd6a9c93dfab5c171b68d4ab56/patches/0001-fix-when-vlan-crc-is-00000000.patch)
- [0002-patches-uradvd-change-ip-lifetimes-to-150-300-secon](https://github.com/freifunkh/site/blob/2538a3f56f2ba1fd6a9c93dfab5c171b68d4ab56/patches/0002-patches-uradvd-change-ip-lifetimes-to-150-300-second.patch)
- [0003-added-TL-WR841ND-N-8M-and-16M-Variants-migration-to-ath79.patch](https://github.com/freifunkh/site/blob/2538a3f56f2ba1fd6a9c93dfab5c171b68d4ab56/patches/0003-added-TL-WR841ND-N-8M-and-16M-Variants-migration-to-ath79.patch)

AiyionPrime, 2023-04-06:
While the patch does resolve the issue, it does not help any device on 2021.1.x release until the next one.

All affected devices running such a version are not capable of leaving that version without loosing their config and manual intervention.
I think we should mark that as known issue for the affected releases.

AiyionPrime, 2023-04-06:
Done. Once both are merged, this can stay closed.
(I'll reopen the issue after the first PR has been merged)

Djfe, 2024-12-15:
does this affect updates from v2021.1 to v2021.1 as well? Or just updates towards newer releases like v2022.1?

blocktrron, 2024-12-16:
Behavior is dependent on Autoupdate source version, not the target version.

closed 2024-12-16 by blocktrron