freifunk-gluon/gluon #2845: sysupgrade from LANasWAN to separated WAN-Port (DSA-Config rule in 020-interfaces)

Djfe, 2023-04-11:
### Bug report

**What is the problem?**
I needed to reset the config to get config mode on lan ports instead of on the one remaining wan port.

**What is the expected behaviour?**
The new DSA config should be applied. Config Mode shouldn't interpret the device anymore as LAN-Only (with LAN-bridge) after the sysupgrade.

**Gluon Version:**
2022.1.3 sysupgrade to master

**Site Configuration:**
[FFAC Site](https://github.com/ffac/site/tree/v2022.1.x)
Our 2022.1.3-based has now a patch as well
But upgrading to this causes the inconsistency in config mode between resets on which ports config mode is available.
https://github.com/ffac/site/blob/add7ded6e226e03ff2c46391725b0bb33ad20638/patches/020-interfaces.patch

**Custom patches:**
https://github.com/ffac/site/tree/v2022.1.x/patches

NeoRaider, 2023-04-12:
If you can reproduce the inconsistent state (by downgrading with config reset and upgrading again), the contents of the `lan_ifname`, `wan_ifname` and `setup_ifname` files in `/lib/gluon/core/sysconfig` would be helpful.

Djfe, 2023-04-12:
will do :)

Djfe, 2023-05-17:
The original device was a TP-Link TD-W8970.
I just tested this on an AVM Fritz!Box 7360SL and wasn't able to reproduce it.
I'll close this for now and retest on the TP-Link eventually.

closed 2023-05-17 by Djfe