freifunk-gluon/gluon #2854: Node does not reply to Neighbour Discovery for v6 Nextnode

nrbffs, 2023-04-14:
### Bug report

**What is the problem?**

Sometimes, the GL-inet XE300 (maybe also other models) does not reply to neighbour discovery requests for its v6 nextnode address. Therefore, ping, status page and ssh do not work over the nextnode address.

From a client, a ping to the node's v6 address on `br-client` works fine. However, pinging the nextnode does not work. One can see from a packet capture that the node doesn't reply to neighbour discovery requests for the nextnode address.

tcpdump on `-i any` in promiscious mode on the node shows that the neighbour discoveries actually reach the node. Listening on `local-node` also shows the neighbour discoveries.

Without promiscious mode (i.e. with `tcpdump -npi local-node`), the neighbour discovery is not visible on the node.

**What is the expected behaviour?**

The node responds to neighbour discoveries and replies to pings.

**Gluon Version:** 157a74c5ab1355ce2d62b53683262fe8e0d11c70

**Site Configuration:** https://gitlab.freifunk-stuttgart.de/firmware/site-ffs/-/tree/6429e1491dde9a9482d0b067f4b1d2993beacfc4

**Custom patches:**

*  <del>https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/beab5a498fc230922290098264102acf6e19d9bf</del> we normally have this in our firmware, removing it didn't change anything
* https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/b6da00d79dbf508e9b470124e735c71f35480104
* https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/d6cf79c6c5589529e1e289de709c14a2d5fcf99b
* https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/61695dc20d45e9f3bdf93fae4d90b2cb64390ef5

maurerle, 2024-01-13:
I have not seen this issue for any device yet.
In your scenario, the link-local address exists on `local-node@local-port` interface but the node does not reply to it? Or is the local-node address not set at all?

If this only affects `GL-inet XE300` you might want to prefix the title with the model?
Or close this issue if this has not been seen recently anymore

nrbffs, 2024-01-13:
> I have not seen this issue for any device yet. In your scenario, the link-local address exists on `local-node@local-port` interface but the node does not reply to it? Or is the local-node address not set at all?

The address was configured on `local-node@local-port` and was reachable from the node itself. Clients were unable to reach it, though.

> If this only affects `GL-inet XE300` you might want to prefix the title with the model? Or close this issue if this has not been seen recently anymore

I remember having observed this on other models, too. Please give me a bit of time to try and reproduce this. Any ideas which debug information I can provide to help further track this down?

nrbffs, 2024-01-13:
Unable to reproduce on XE300 and OR750i with our firmware build 3.0+2024-01-09-g.62dbf788-s.3621e0e (v2023.2). Closing for now.

closed 2024-01-13 by nrbffs