freifunk-gluon/gluon #2855: Pending device backports

AiyionPrime, 2023-04-14:
We talked about device backports on the last meetup, which we neglected a little in the past.
Someone argued we should retain the order of commits if possible.

I'll create a list of device PRs which could be backported, if we found a tester for them.

These commits are likely part of the list, but not the first ones.

These commits are then even as likely not the first ones to be backported, if we'd try to maintain order as far as possible.

- #2851
- #2852
- #2853

rotanid, 2023-04-16:
> I'll create a list of device PRs which could be backported, if we found a tester for them.

why would we need to re-test if the stable branch (currently v2022.1.x) is using the same openwrt branch for which the device support was already tested in master branch?

AiyionPrime, 2023-04-17:
I wrote it down, because we talked about it in the last meetup.
The idea was that the devices we are going to backport have been in master for a while now and may have received additional fixes we might miss else.

I have no strong opinion on this.

maurerle, 2023-04-17:
Then I will add those two to the list here, so that they can be merged in the right order:
#2681
 
#2863

AiyionPrime, 2023-04-18:
#2861 is 'only' adding an openwrt patch and not changing target files.
It does not need to follow this order.

The other one is reflected in the list above. I'd mark your comment as resolved, to keep this as clean as possible.

AiyionPrime, 2023-04-21:
That was it.

closed 2023-04-21 by AiyionPrime