freifunk-gluon/gluon #2858: Compiler warnings in gluon builds

stweil, 2023-04-16:
### Bug report

By default a gluon build is not verbose and compiler output is suppressed, but a verbose build (with `V=s`) shows lots of compiler warnings some of which are real bugs in host tools, kernel or package code.

Using newer code (like in gluon's next branch) sometimes fixes such bugs, but even with the current next branch I get more than 1800 warnings in a build for a single device. Here is a list of the different kinds of warning with their frequency (made with `grep warning:.*-W build.log | sort | uniq | sed s/.*-W/-W/ | sed 's/.$//' | sort | uniq -c`:

```
     63 -Waddress
*    11 -Waddress-of-packed-member
*     6 -Warray-bounds
      1 -Warray-parameter=
      3 -Wattributes
     91 -Wbuiltin-declaration-mismatch
      3 -Wcast-align
     32 -Wcast-function-type
    240 -Wc++-compat
      1 -Wclobbered
     15 -Wcpp
      2 -Wdangling-pointer=
      5 -Wdelete-non-virtual-dtor
      4 -Wdeprecated
    242 -Wdeprecated-declarations
    134 -Wdiscarded-qualifiers
*    23 -Wformat=
     84 -Wformat-extra-args
      6 -Wformat-nonliteral
*     5 -Wformat-overflow=
      2 -Wformat-truncation=
      6 -Wframe-larger-than=
     40 -Wimplicit-fallthrough=
      4 -Wimplicit-function-declaration
      2 -Winfinite-recursion
      4 -Winline
    293 -Wint-conversion
*     3 -Wlogical-not-parentheses
*    12 -Wmaybe-uninitialized
*    25 -Wmisleading-indentation
      2 -Wmissing-prototypes
      1 -Wmultistatement-macros
*     1 -Wnonnull
      1 -Wnull-dereference
      2 -Wother
      6 -Wparentheses
     44 -Wpointer-sign
*    10 -Wpointer-to-int-cast
      3 -Wreorder
      3 -Wreorder
*     4 -Wreturn-local-addr
      6 -Wsequence-point
      1 -Wshift-count-negative
      5 -Wshift-count-overflow
      4 -Wsign-compare
*     1 -Wsizeof-pointer-div
*     2 -Wsizeof-pointer-memaccess
      5 -Wstrict-prototypes
*    18 -Wstringop-overflow=
     49 -Wstringop-truncation
      1 -Wswitch
      1 -Wtautological-compare
     11 -Wunused-but-set-variable
      5 -Wunused-const-variable=
     67 -Wunused-function
      1 -Wunused-label
     14 -Wunused-parameter
      4 -Wunused-result
     89 -Wunused-variable
*     2 -Wuse-after-free
     10 -Wvla
*    25 -Wwrite-strings
     93 -Wyacc
      1 -Wzero-length-bounds
```

I have marked the warnings which I consider critical ones with an asterix `*`. My experience is that they are often real bugs.

blocktrron, 2023-04-16:
I don't see this as a Gluon issue but rather the issue of OpenWrt or the respective upstream project. In case build-warnings occur on Gluon tools, feel free to create a PR fixing those. The upstream ones are not applicable to Gluon

closed 2023-04-16 by blocktrron

AiyionPrime, 2023-04-16:
Sidenote: Gluons `next`-branch is not supported. Never was, never will be.

stweil, 2023-04-16:
@blocktrron, @AiyionPrime, maybe I was not clear enough, but the core message of this issue was that the supported tagged release raises even more compiler warnings (some of them are bugs) because it uses old software releases. As long as the current released gluon uses an older OpenWrt, it gets packages with known bugs which are fixed in newer software.

AiyionPrime, 2023-04-16:
We do not argue, that this would not be problematic. But what you are trying to inspect here is an OpenWrt issue. You simply opened it in the wrong repo.
We are using one of the latest commits on OpenWrt currently supported branch.
If you find that that branch is lacking updates, we can not help you in this repo.