freifunk-gluon/gluon #2883: gluon-reconfigure "./500-mesh-vpn:14: attempt to index local 'o' (a nil value)"

edeso, 2023-05-02:
### Bug report

**What is the problem?**

running `gluon-reconfigure` seems to not completely execute `500-mesh-vpn`. error is

```
Configuring: 500-mesh-vpn
/usr/bin/lua: ./500-mesh-vpn:14: attempt to index local 'o' (a nil value)
stack traceback:
        ./500-mesh-vpn:14: in main chunk
        [C]: ?
```

we use this  custom gluon package https://github.com/ff-kbu/gluon-mesh-vpn-wireguard to do wireguard meshing.

**What is the expected behaviour?**
no error

**Gluon Version:**
2022.1.3

**Site Configuration:**
https://github.com/ff-kbu/site-ffkbu-multidomain/tree/wireguard

**Custom patches:**
none

blocktrron, 2023-05-02:
Your provider specification is missing within the package (See https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-mesh-vpn-fastd/luasrc/usr/lib/lua/gluon/mesh-vpn/provider/fastd.lua for out fastd example).

As this does not look like a Gluon issue to me, I will close this issue in the next 7 days if the source has not been identified to be Gluon itself in this timeframe.

edeso, 2023-05-03:
yeah, i'm just trying to figure out what does what just now. 

is there documentation on how to utilize the gluon wireguard package? 

currently we run an "elaborate" script 
https://github.com/ff-kbu/gluon-mesh-vpn-wireguard/blob/master/gluon-mesh-vpn-wireguard-kbu/files/lib/gluon/gluon-mesh-wireguard/checkuplink
that fetches keys from the endpoints and (re)connects if needed. surely this is not implemented by default.

edeso, 2023-05-03:
@blocktrron could you please hint me on how to have the Makefile 
https://github.com/ff-kbu/gluon-mesh-vpn-wireguard/blob/master/gluon-mesh-vpn-wireguard-kbu/Makefile
install the lua script?

blocktrron, 2023-05-03:
You need to add it to the `Package/gluon-mesh-vpn-wireguard-kbu/install` section like you did with the non-lua files.

closed 2023-05-03 by blocktrron

edeso, 2023-05-04:
is there a way to reuse the install routine 
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon.mk#L91-L108
from a site package?

blocktrron, 2023-05-04:
OpenWrt defines a variable for It's topdir, you can use that relatively to reference this Makefile.

One sidenote - The code for bootstrapping the connection appears to have multiple issues in regards to device-integrity. I strongly advise you to revise this before deployment.

edeso, 2023-05-04:
> OpenWrt defines a variable for It's topdir, you can use that relatively to reference this Makefile.

i found this example, which seems to do what you suggest
https://github.com/maurerle/gluon-mesh-vpn-wireguard/blob/main/ff-mesh-vpn-wireguard-openwrt19/Makefile

> One sidenote - The code for bootstrapping the connection appears to have multiple issues in regards to device-integrity. I strongly advise you to revise this before deployment.

this is code that is currently running on nodes since maybe a year or so. what issues do you see with it? we can as well deal with it in an issue in it's repo of course.

just to reiterate. i just took it on as nobody is doing firmware releases just now. the previous maintainers left. just got some basic openwrt/gluon knowledge, that's it. feel free to treat me as a novice therefor.