freifunk-gluon/gluon #2893: br-client does not come up occasionally v2021.1.x

2tata, 2023-05-07:
### Bug report

**What is the problem?**

On domain switch the `br-client bright` occasionally doesn't come up again.

**What is the expected behaviour?**

on faliuor:
```
root@LegacyMonitor:/# brctl show
bridge name	bridge id		STP enabled	interfaces
br-wan		7fff.ee0000000000	no		eth0
br-client		7fff.ee0000000000	no
```
when working:
```
root@LegacyMonitor:~# brctl show
bridge name	bridge id		STP enabled	interfaces
br-wan		7fff.ee0000000000	no		eth0
br-client		7fff.ee0000000000	no		local-port
							bat0
```

On faliour:
```
root@LegacyMonitor:/# ifconfig br-client 
br-client Link encap:Ethernet  HWaddr EE:00:00:00:00:00  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:314 errors:0 dropped:0 overruns:0 frame:0
          TX packets:56 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:9510 (9.2 KiB)  TX bytes:6265 (6.1 KiB)
```
when working:
```
root@LegacyMonitor:~# ifconfig br-client
br-client Link encap:Ethernet  HWaddr EE:00:00:00:00:00  
          inet6 addr: fe80::ec00:ff:fe00:0/64 Scope:Link
          inet6 addr: 2a0f:b506:ff05:0:ec00:ff:fe00:0/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:35 errors:0 dropped:0 overruns:0 frame:0
          TX packets:33 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1556 (1.5 KiB)  TX bytes:3394 (3.3 KiB)
```

Here are the full logs during `gluon-reload` call
I add a logger TEST flag in the gluon-reload script to better separate post stages.
https://git.ffnw.de/tata/legacy-monitor/-/issues/2#note_4519 (github does not allow me to post the full log here)...

**Gluon Version:**
v2021.1.x 
Commit ID: 595abcf8cb2dc794801c6ed5f1641783fccf5806

I will check if is still in the newes release as well.

rotanid, 2023-05-09:
@2tata as  v2021.1.x is not supported anymore, can you please test with a supported gluon branch and report back?

https://github.com/freifunk-gluon/gluon/wiki/Release-life-cycle

2tata, 2023-05-11:
@rotanid I tried several times on the newer one but it seems not to trigger anymore. Ill close it and open it again if its occur in future.

closed 2023-05-11 by 2tata