freifunk-gluon/gluon #2903: backport "ath79-generic: enable ath10k-smallbuffers", unbreak tp-link-archer-c25-v1

edeso, 2023-05-10:
in case there are more 2022.1.x maintenance releases planned would it be possible to backport
 https://github.com/JayBraker/gluon/commit/a723cbfd4f170c4ca6c45391b08490013fbfb345
?

also i'd like to unbreak tp-link-archer-c25-v1 which is working nicely with smallbuffers since 2 years for me.

edeso, 2023-05-10:
does this go for both points? memory patch and **tp-link-archer-c25-v1**?

do you need me to rebase the https://github.com/freifunk-gluon/gluon/commit/53e9997a4a09b50e572f875e1213e5b1809c9488 ? visiting the commit i see no possiblity to create a PR against brach `2022.1.x` gainst it. can you please give me a hint?

blocktrron, 2023-05-10:
Which memory patch is required? I only see the change switching the kmod packages.

edeso, 2023-05-10:
yes that one. https://github.com/freifunk-gluon/gluon/commit/53e9997a4a09b50e572f875e1213e5b1809c9488 
switching  the kmods saves memory on low memory devices.

blocktrron, 2023-05-10:
Ah okay.

1) Switching to smallbuffers kmod can be backported w/o bigger issues on v2022.1.x.

2) Unbreaking the specific model requires doing the checklist on master first. It can be backported to v2022.1.x afterwards. For the backport 1) is required.

I hope this makes it clear :laughing:

edeso, 2023-05-10:
> Ah okay.
> 
> 1. Switching to smallbuffers kmod can be backported w/o bigger issues on v2022.1.x.

so you don't need a PR from me for that?

> 2. Unbreaking the specific model requires doing the checklist on master first. It can be backported to v2022.1.x afterwards. For the backport 1) is required.

sorry. which checklist? can you please point me to it?

thanks!

blocktrron, 2023-05-10:
Please create the PR is you want to see the backport in the release.

Please see https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

edeso, 2023-05-10:
> Please create the PR is you want to see the backport in the release.

is there a way for me to create a PR directly from the commit to `master`? sorry i looked but didn't see one. 

> Please see https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

that's quite a list to test, which i currently have the time to.  the comment in target file pretty clearly states that it is marked broken because of "OOM with 5GHz enabled in most environments" which is fixed by the above commit

https://github.com/freifunk-gluon/gluon/blob/979dd354faaf4e47f04d9609615dc9db8fa61cb1/targets/ath79-generic#L392-L396

edeso, 2023-07-26:
hey @blocktrron or anybody else.

1.
what do you need me to do to get https://github.com/freifunk-gluon/gluon/commit/53e9997a4a09b50e572f875e1213e5b1809c9488 into the 2022.1.x branch?

2.
ideally i'd like to unbreak, remove  `broken = true, -- OOM with 5GHz enabled in most environments ` the tp-link-archer-c25-v1. i will not be able to do the requested list unfortunately. so if this indeed is a prerequisite i can abstain from it.

thanks!

rotanid, 2023-09-06:
@edeso if it's only about the checklist, i could do that, if someone else does everything else and points me to the repo/branch i need to build with our site

edeso, 2023-10-08:
@rotanid the patch really is just removing the broken flag. as the smallmemory kmods are now in 2023.x that's all that needs to be done.

edeso, 2023-10-08:
additionally we could prpbably unbreak other 5GHz OOM devices although i have no possibility to test those :)

rotanid, 2023-10-08:
how many nodes are in the mesh in which your test device is? can you link the map?
how many nodes does the segment/domain of your test device have?

edeso, 2023-10-09:
just two nodes and a handful of users. see
https://map.kbu.freifunk.net/#!/en/map/50c7bf3a5a0f

the device was pretty much unusable without smallbuffers oom'ing like minutes after wifi5 went up. now it is purring since the smallbuffers exist. i patched and built gluon manually since then.

according to reports on the openwrt ml and in the forum the smallbuffers kmods solved the 5Ghz oom issues. i'd suggest unbreaking the affected devices and waiting for reports. after all these are old and small and probably sparingly used if at all anymore. still, no reason to trash them.

rotanid, 2023-10-09:
i think i'd merge such a Pull request after a few days waiting for someone's veto

edeso, 2023-10-09:
sounds reasonable. MR against master or 2023.x or both or ...?

rotanid, 2023-10-09:
have you tested both? they are based off of different OpenWrt branches so there's another hurdle as we ususally prefer "master branch first"

edeso, 2023-10-10:
as i am using the "ath*-smallbuffers" kmods since they were introduced, first "ath10k-ct-smallbuffers" now "ath10k-smallbuffers" i am not expecting any weirdness. after all they are the ordinary drivers just patched to use smaller buffers.

even if we commit to master first, i wouldn't expect testers showing up as we are talking older smaller devices that seem to be used by so little people that nobody even showed up in this ticket :)

again. i feel confident that without unbreaking them nobody will even realise that they are still valuable devices or even comeback with test results.

rotanid, 2023-10-11:
i mostly agree, though i really prefer we keep to our usual process, please test it with master branch and after that we can backport to stable.

edeso, 2023-10-11:
is master site.conf 2023.x compatible?

maurerle, 2024-01-14:
I think it is not very valuable anymore to backport to v2022.x
The requested commit which adds smallbuffers is already part of gluon v2023.1 and up and is used for the requested devices.

So one could still try to do the checklist for these devices to remove the broken state, if they work properly with the smallbuffers driver (on master and eventually backport it to v2023.2)
Otherwise this issue can be closed.

maurerle, 2024-01-14:
And @edeso - yes the site.conf is compatible between branches, but you do need to switch your package configuration from site.mk to image-customization.lua when going from v2023.1.x to v2023.2.x and higher

edeso, 2024-01-14:
> The requested commit which adds smallbuffers is already part of gluon v2023.1 and up and is used for the requested devices (which are built if the broken flag is set)

as said. stably running the **tp-link-archer-c25-v1** since years with the smallbuffers kmod versions. first patched manually, now with v2023.1 incl. the patch. my experience goes along with other openwrt users in forum or on ml.

but, sure i agree that the device and the other tiny ones are outdated, i can live without it staying **broken** albeit it in reality clearly isn't. it'd be just a minor contribution against unneeded scrapping.

maurerle, 2024-01-14:
The broken flag for this device was reintroduced here: https://github.com/freifunk-gluon/gluon/commit/61bd88309e8b90406f95e31b9e2168a328d2ddee long time before smallbuffers

From my point of view we could remove the broken flag for these devices if we find someone who confirms that they are working stable on current master branch:

- tp-link-archer-c2-v3
- tp-link-archer-c25-v1
- tp-link-tl-wr902ac-v1

Though I do not know in which scenarios the driver went OOM without the smallbuffers or how long one needs to wait for it to happen..

There are a lot of these devices in this community:
https://services.freifunk-suedwest.de/maps/hg
They are currently running v2023.1.1 - so maybe one can find testers there to test Firmware based on gluon master?

blocktrron, 2024-01-14:
@maurerle memory usage of ath10k significantly increased with the latest OpenWrt version bump. I'm rather thinking about enabling smallbuffers for the regular ath10k 128M devices.

I'm against unbreaking these devices. Stating they are supported just to discontinue them in 1-2 years makes no sense, especially since these devices are not manufactured new.

edeso, 2024-01-15:
@blocktrron really don't see advantage to keep em broken here. generally they won't serve that many clients, but still run reliably. at worst some ppl might complain about the performance but at least there will be some feedback. and 1-2 years is still some long time. 

anyway. your prerogative i can only advocate :)

maurerle, 2024-01-15:
Thanks @blocktrron good point.
I think that communities who have a need for these devices, can just build firmware with these devices by setting BROKEN=1
So devices should not be scrapped unneccessarily but can be used that way.

I think the effect of having them listed as unbroken is not that huge here, so I am in favor of closing this issue.

blocktrron, 2024-01-15:
You can now even selectively unbreak them with the image-customization framework.

I'm for closing this issue.

edeso, 2024-01-16:
> You can now even selectively unbreak them with the image-customization framework.

good pointer. albeit this just started with current 2023.2 gluon release.

> I'm for closing this issue.

go ahead. it's not like i can't live with the status quo :) at least the smallbuffers making the devices purr again are in, so that's the most important thing.

maurerle, 2024-07-15:
I'd close this issue, as it is now quite dated.

closed 2024-07-16 by blocktrron