freifunk-gluon/gluon #2904: TP-Link WDR4300 hangs after autoupgrade

grische, 2023-05-13:
### Bug report

**What is the problem?**
Occasionally (>10% of all devices), hang after an autoupdate and need a manual powercycle to reboot.

I managed to reproduce this while a serial cable was attached:
```
root@ffmuc-e894f6d457e8:/# autoupdater -f
Retrieving manifest from http://[2001:678:ed0:f000::]/experimental/sysupgrade/experimental.manifest ...
Stopping cron...
Stopping urngd...
Stopping micrond...
Stopping sysntpd...
Stopping gluon-radvd...
Stopping uhttpd...
Stopping sse-multiplexd...
Stopping gluon-respondd...
[  350.261277] sysctl (6540): drop_caches: 3
vm.drop_caches = 3
Downloading image:  6144 / 6144 KiB
[  372.830625] device client1 left promiscuous mode
[  372.835595] br-client: port 4(client1) entered disabled state
[  373.170806] device client0 left promiscuous mode
[  373.175746] br-client: port 5(client0) entered disabled state
Stopping network...
[  374.316737] batman_adv: bat0: Interface deactivated: mesh1
[  374.564283] batman_adv: bat0: Removing interface: mesh1
[  374.587411] batman_adv: bat0: Interface deactivated: mesh0
[  374.782898] batman_adv: bat0: Removing interface: mesh0
[  376.235548] br-client: port 3(bat0) entered disabled state
[  376.241674] br-client: port 2(local-port) entered disabled state
[  376.247889] br-client: port 1(eth0.1) entered disabled state
[  376.336981] device bat0 left promiscuous mode
[  376.341546] br-client: port 3(bat0) entered disabled state
[  376.393905] device eth0.1 left promiscuous mode
[  376.398752] br-client: port 1(eth0.1) entered disabled state
[  376.498386] device local-port left promiscuous mode
[  376.503522] br-client: port 2(local-port) entered disabled state
[  376.807475] batman_adv: bat0: Interface deactivated: primary0
[  376.813326] batman_adv: bat0: Removing interface: primary0
[  376.819582] batman_adv: bat0: Interface deactivated: mesh-vpn
[  376.825424] batman_adv: bat0: Removing interface: mesh-vpn
Cannot find device "bat0"
autoupdater: warning: execution of /usr/lib/autoupdater/upgrade.d/10stop-network exited with status code 1
[  379.953731] sysctl (6840): drop_caches: 3
vm.drop_caches = 3
Sat May 13 12:33:01 CEST 2023 upgrade: Saving config files...
Sat May 13 12:33:02 CEST 2023 upgrade: Commencing upgrade. Closing all shell sessions.
Watchdog handover: fd=3
- watchdog -
Watchdog did not previously reset the system
Sat May 13 12:33:04 CEST 2023 upgrade: Sending TERM to remaining processes ...
Sat May 13 12:33:04 CEST 2023 upgrade: Sending signal TERM to udhcpc (2029)
Sat May 13 12:33:04 CEST 2023 upgrade: Sending signal TERM to odhcp6c (2034)
Sat May 13 12:33:04 CEST 2023 upgrade: Sending signal TERM to dnsmasq (2883)
Sat May 13 12:33:08 CEST 2023 upgrade: Sending KILL to remaining processes ...
[  400.548946] stage2 (7004): drop_caches: 3
Sat May 13 12:33:15 CEST 2023 upgrade: Switching to ramdisk...
mount: mounting /dev/mtdblock4 on /overlay failed: Resource busy
[  404.592005] VFS: Busy inodes after unmount of jffs2. Self-destruct in 5 seconds.  Have a nice day...
Sat May 13 10:33:18 UTC 2023 upgrade: Performing system upgrade...
[  404.686640] do_stage2 (7004): drop_caches: 3
Unlocking firmware ...

Writing from <stdin> to firmware ...
Appending jffs2 data from /tmp/sysupgrade.tgz to firmware..                                                                                                  
Sat May 13 10:33:49 UTC 2023 upgrade: Upgrade completed
Sat May 13 10:33:50 UTC 2023 upgrade: Rebooting system...
umount: can't unmount /dev: Resource busy
umount: can't unmount /tmp: Resource [  436.218456] reboot: Restarting system
▒
```

I am not sure if this is related to #185, but we were not able to reproduce it (yet) with a reboot.

**What is the expected behaviour?**
That the WDR4300 comes back up after an update.

**Gluon Version:**
v2022.1.2 and v2022.1.3
Probably also earlier v2022.x 

We experienced similar behaviour during the initial v2022.1 deployment, but discarded it as "random".
It was more severe with the v2022.1.3 deployment (probably just because of chance) and I was able to reproduce it with a serial cable attached when upgrading from v2022.1.3 to v2022.1.4.

**Site Configuration:**
https://github.com/freifunkMUC/site-ffm/blob/833829e68f97e4781f175bdd688d7f498a7efe53/site.conf

**Custom patches:**
https://github.com/freifunkMUC/site-ffm/tree/833829e68f97e4781f175bdd688d7f498a7efe53/patches

rotanid, 2023-05-14:
does this also happen with the very similar WDR3600 ?

grische, 2023-07-07:
> does this also happen with the very similar WDR3600 ?

Probably. We had a few isolated cases where a WDR3600 needed a power cycle after an upgrade but it is not clear if this is at all related to the problem described here. We don't have enough (failing) devices to have a confident answer.

grische, 2023-07-07:
It might be worth mentioning that the special symbol at the end of the log is printed during a boot as well. I'm not sure if this is printed before or after the bootloader loaded though.

smoe, 2023-08-13:
We observed this when transitioning from 2022.1.2 to 2022.1.4 on WDR4300 and more frequently on Ubiquiti AC lite. In our observation, the update was fine when the machine was rebooted just prior to the update, which may be suggesting an out-of-memory issue.

grische, 2023-11-21:
Just to clarify, we were able to reproduce the issue on a freshly booted device as well.
I assume the WDRs and the AC Lite are different issues here.

blocktrron, 2023-12-11:
@grische 

One thing that comes to my mind is the usage of the newer ar934x SPI controller driver, at least no device reported in this issue uses the older ar71xx driver.

This driver was first shipped with OpenWrt 21.02, matching the observation it does not break with older releases based on OpenWrt 19.07 and older.

https://github.com/openwrt/openwrt/commit/ebf0d8dadeca443121f4f597c51bf6591e341caf

If you are still able to reproduce this issue, you can modify the ar934x DTSI to use the compatible for the ar71xx SPI controller. Ping me in case i should provide you with a patch. If this fixes the reboot issue, we have a better path where to look next.

blocktrron, 2023-12-12:
@grische Almost. Just revert this commit in the file:

https://github.com/openwrt/openwrt/commit/ebf0d8dadeca443121f4f597c51bf6591e341caf#diff-45ad725f9ec8cc2da88738047b1d5c4d1e69df19194bd22394d3736e03093613

grische, 2024-01-07:
@blocktrron I was able to reproduce a hang after reboot even with the above commit reverted using Gluon v2023.1:
https://gist.github.com/grische/27e4e780530f9a0795d96afaf749a4ed

Here is the respective branch: https://github.com/grische/site-ffm/commits/test/revert-ath79-add-new-ar934x-spi-driver/

blocktrron, 2024-01-07:
@grische Are these hangs only reproducible after writing a upgrade image or does a regular reboot invocation also trigger a spurious hang?

grische, 2024-01-07:
I have a test WDR4300 device where I can reproduce the hangs during a reboot every other time. Surprisingly often actually.
This device has a manually installed serial port and a serial cable attached to the port. 

Do you think I should test another device that has not been modified to be sure?

grische, 2024-01-08:
On the exact same setup, I tested it with 
- gluon-v2021.1.2: no hang was observed for ~150 reboots
- gluon-v2022.1.4: the hangs were observed after as early as 2 reboots or as long as 5 reboots.
- gluon-v2023.2: the first hang was observed after 3 reboots.

grische, 2024-01-11:
The bug was fixed upstream in 
- main: https://github.com/openwrt/openwrt/commit/2fe8ecd880396b5ae25fe9583aaa1d71be0b8468
- 23.05: https://github.com/openwrt/openwrt/commit/cdc907abedf1ad0d17cc88fd33d35a3ef3f8fc75
- 22.03: pending

nrbffs, 2024-12-15:
We have lost several WDR3600 on a recent upgrade to 2023.2.4. 

I attached a WDR3600 to a serial console and used this script to reboot it in a loop: https://gitlab.freifunk-stuttgart.de/-/snippets/8

I was able to observe failing reboots after 5, 20 and 250 tries.

With a patch like this, I have >1500 successful reboots now:
```
--- a/drivers/reset/reset-ath79.c
+++ b/drivers/reset/reset-ath79.c
@@ -79,8 +79,12 @@ static int ath79_reset_restart_handler(s
 {
 	struct ath79_reset *ath79_reset =
 		container_of(nb, struct ath79_reset, restart_nb);
+	unsigned int i = 0;
 
-	ath79_reset_assert(&ath79_reset->rcdev, FULL_CHIP_RESET);
+	while (1) {
+		ath79_reset_assert(&ath79_reset->rcdev, FULL_CHIP_RESET);
+		printk("reset: still alive after %u tries\n", i);
+	}
 
 	return NOTIFY_DONE;
 }
```

The printk can never be seen, but I suppose that's because there is never a chance to flush out the buffer to the console.

It's not clear to me why this works, but neither is the solution of reading back the register (ioremap should already disable cache).

Djfe, 2024-12-15:
I'm pretty certain we have seen this on a few wdr4300 in Aachen, too.
Kind of sad this can only be fixed by doing another reboot so after loosing a couple more devices, temporarily (for future updates)

thanks for reporting a fix ❤️

Djfe, 2024-12-15:
oh wait, I'm wrong. This code is run on boot so it fixes it immediately, right? (not before rebooting)

blocktrron, 2024-12-16:
@nrbffs can you provide the exact gluon commit of the upgrade failures start version (before upgrading) and commit hashes of the automatic reboots you have seen failing?

smoe, 2024-12-16:
> oh wait, I'm wrong. This code is run on boot so it fixes it immediately, right? (not before rebooting)

At least the "i" should be incremented :-)

nrbffs, 2024-12-17:
> can you provide the exact gluon commit of the upgrade failures start version (before upgrading)

ec72498a44c66a512ceeb45d89cf9ea8f4e5eda8 (v2023.2.3)
plus these two patches by our community:

- https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/baad32dc264ae3e92284fe500706e89d1733cd00
- https://gitlab.freifunk-stuttgart.de/firmware/gluon/-/commit/1a770db51ace5ee47e363cff6a9d72c2d0ec31a9

> and commit hashes of the automatic reboots you have seen failing?

I have seen the failures also on the commit above as well as v2023.2.4

My proposed fix is in #3397 (tested on main, 1487 successful reboots)

grische, 2025-01-03:
I can confirmt hat the new patch seem to properly fix the issue at last!

- latest Gluon v2023.1.x  and got 2 hangs out of 8 reboots
- latest Gluon v2023.2.x and got 2 hangs out of 6 reboots
- latest Gluon v2023.2.x + [backported #3397](https://github.com/freifunkMUC/site-ffm/pull/545) to Gluon v2023.2.x and got 0 hangs out of 800+ reboots

blocktrron, 2025-01-05:
I did talk with @nrbffs at 38c3 about this topic and we agreed on working out a upstream compatible version of this patch.

Please test https://github.com/freifunk-gluon/gluon/tree/ar9344-reset and post the dmesg output when running this firmware.

I don't have compatible hardware at hand for now.

nrbffs, 2025-01-05:
Thanks, dmesg at https://paste.debian.net/1342540/

grische, 2025-01-06:
I also ran the reboot tests on main and on the ar9344-reset branch:

- current main ( c31c7d7534817b7f5edb92296c25c4418e77b7ff ): 1 hang out of 62 reboots
- ar9344-reset ( d3f23425f763baf2e27ddc99d090bc4f83f80d8b ): 0 hangs out of 700 reboots

Serial console output can be found here: https://gist.github.com/grische/be44b330d7f8c7fff88939f979be9d32

grische, 2025-01-07:
I might have bad news: the ar9344-reset branch got stuck during after a reboot on my WDR4300 at the 705th reboot.

I added the last two reboots into the above gist: https://gist.githubusercontent.com/grische/be44b330d7f8c7fff88939f979be9d32/raw/634a35543445ee220a2c77e8ee6a3b7e285e72d3/ttyUSB_2025-01-05T16%25EF%2580%25BA39%25EF%2580%25BA52+01%25EF%2580%25BA00.0.tail

smoe, 2025-01-07:
The hanging reboot is 10 seconds faster and these bits here are not happening:
```
[   58.165707] batman_adv: bat0: IGMP Querier appeared
[   58.170686] batman_adv: bat0: MLD Querier appeared
[   68.025161] br-client: port 4(client1) entered blocking state
[   68.031072] br-client: port 4(client1) entered disabled state
[   68.037222] ath9k 0000:00:00.0 client1: entered allmulticast mode
[   68.043808] ath9k 0000:00:00.0 client1: entered promiscuous mode
[   68.157288] br-client: port 4(client1) entered blocking state
[   68.163153] br-client: port 4(client1) entered forwarding state
[   69.998095] br-client: port 5(client0) entered blocking state
[   70.004107] br-client: port 5(client0) entered disabled state
[   70.010052] ath9k 18100000.wmac client0: entered allmulticast mode
[   70.016831] ath9k 18100000.wmac client0: entered promiscuous mode
[   70.118377] br-client: port 5(client0) entered blocking state
[   70.124295] br-client: port 5(client0) entered forwarding state
[   71.289033] br-client: port 5(client0) entered disabled state
[   71.321065] br-client: port 5(client0) entered disabled state
[   71.343286] ath9k 18100000.wmac client0: left allmulticast mode
[   71.349408] ath9k 18100000.wmac client0: left promiscuous mode
[   71.355560] br-client: port 5(client0) entered disabled state
[   71.552137] br-client: port 4(client1) entered disabled state
[   71.648189] ath9k 0000:00:00.0 client1 (unregistering): left allmulticast mode
[   71.655621] ath9k 0000:00:00.0 client1 (unregistering): left promiscuous mode
[   71.662889] br-client: port 4(client1) entered disabled state
```

grische, 2025-01-07:
The timings across all reboots are not very reliable as I attempt a reboot "every 30 seconds", so the 10s shift could have been coincidental. 

When I grep the log of almost 700 reboots, I can find the following parts
- `ath9k 0000:00:00.0 client1 (unregistering): left allmulticast mode` happened 116 times 
- `ath9k 18100000.wmac client0: entered allmulticast mode`  happened 276 times
- `ath9k 0000:00:00.0 client1: entered allmulticast mode` happened 300 times

rotanid, 2025-01-08:
@grische should be fixed in Gluon main due to the merge of #3411

grische, 2025-01-08:
@rotanid the tested patch https://github.com/freifunk-gluon/gluon/commit/d3f23425f763baf2e27ddc99d090bc4f83f80d8b and the upstreamed patch https://github.com/openwrt/openwrt/commit/0c52c9d6fc53ebe0834b3945c5d58277752d9648 are identical. 

And yes, this would need a backport to Gluon v2023.2.x. At least the first version of the patch was easily backported to kernel 5.15.

grische, 2025-01-11:
It seems the backport used in ab1c311845ba72484c88294fe2735fbc028b9ba5 does not have any effect. I am back at around 1 hang per 5 reboots on average using this commit. 

The build artifacts can be found here: https://github.com/freifunkMUC/site-ffm/actions/runs/12723841862?pr=558

The patches that are being applied on top of the above commit should have no impact:
https://github.com/freifunkMUC/site-ffm/tree/7a495f0ce8153f55b80871ec0a90bfb908805f2b/patches

Any suggestion what could have gone wrong?

Adorfer, 2025-01-12:
question: does those "hang" are on (about) 10% of "simple" reboot, or does it need something like an autoupdater performing a job?

Djfe, 2025-01-12:
yes, it's on simple reboots. so if your community does regular reboots then every now and then a device will stay offline until powercycled by the user.
https://github.com/freifunk-gluon/gluon/issues/2904#issuecomment-2569144224

though this issue has been present in earlier versions

Adorfer, 2025-01-15:
3600 on gluon 2023.1.x did more than 500 reboots now (every 4-5 minutes) without any issue. 
But this node is wifimesh-only. so perhaps the issue has more conditions than "just reboot". 
https://map.eulenfunk.de/stats/d/000000004/node-byid?orgId=1&var-nodeid=6466b3ce1b8c
[image]

rotanid, 2025-01-20:
new update has been pushed by @blocktrron for this issue
https://github.com/freifunk-gluon/gluon/pull/3428
upstream commit:
https://github.com/openwrt/openwrt/commit/1e079d790ab3eb2d82990c4d547d6ce2635a25f7

grische, 2025-01-22:
I was able to confirm that there were no hangs with the modified patch from nrb with the 1/1/10ms had no hungs after 2500+ reboots (within 72+ hours):
https://github.com/freifunkMUC/gluon/commit/1515a09c45e088827991d9b58e4e79125be2b352

I assume the recently upstreamed patch is identical to the tested one?

rotanid, 2025-01-23:
i don't know any C but to me it looks different to the upstream patch, just look at them in two windows side-by-side

neocturne, 2025-03-30:
Can someone confirm whether this issue is resolved in current Gluon main?

Djfe, 2025-03-30:
it isn't, I was planning on backporting it into 24.10
It's about time. I'll prepare a backport pr today

neocturne, 2025-04-08:
@Djfe Huh, is https://github.com/openwrt/openwrt/commit/1e079d790ab3eb2d82990c4d547d6ce2635a25f7 not sufficient? That is in the 24.10 branch

grische, 2025-06-14:
I did a reboot loop test with a WDR4300 based on Gluon main 480fdf574e1818b30809ef4d03ff9213fdcdb508 and I was able to run 2500+ successful reboots without a hang.

It seems the fix works as expected on main. 
Should I also test Gluon v2023.2.x?

neocturne, 2025-06-16:
Ah, no, testing with v2023.2.x is not useful yet - the OpenWrt 23.05 backport is missing after all.

rotanid, 2025-08-26:
the backport was merged upstream 12 days ago:
539228933caab51a21c239ec3b7e8f7a366e96b9
we might think of updating modules

PR #3159 (2024-01-13): [v2023.1.x] modules: update openwrt
This fixes #2904  on the v2023.1.x branch too

```
a08553b3b3 ath79: read back reset register
debf4b56cd kernel: bump 5.10 to 5.10.206
6121581765 kernel: bump 5.10 to 5.10.203
17ee3e0b20 raimps: mtk_eth_soc: drop rst_esw from ESW driver
5ef01117b7 ramips: dts: mt7628an: reset FE and ESW cores together
b80c17b093 ramips: dts: rt5350: reset FE and ESW cores together
0c84a15288 ramips: dts: rt3050: reset FE and ESW cores together
37ed4c0ec2 ramips: dts: rt3352: reset FE and ESW cores together
8b4b924b85 ramips: mtk_eth_soc: wait longer after FE core reset to se [... cut]