freifunk-gluon/gluon #2929: factory images missing for D-Link DIR-505 after migration to ath79-generic

edeso, 2023-07-25:
flag is set to false, it wasn't before.

https://github.com/freifunk-gluon/gluon/blob/v2021.1.2/targets/ar71xx-generic#L95-L97
vs.
https://github.com/freifunk-gluon/gluon/blob/master/targets/ath79-generic#L114-L119

edeso, 2023-07-25:
ok,

will revoke this issue. just read that the ath79 target cannot produce factory images and older ar71xx factory images are uosed to be used
https://openwrt.org/toh/d-link/dir-505#installation

will close the issue. sorry for the noise.

closed 2023-07-25 by edeso

NeoRaider, 2023-07-29:
Hmm, no idea what's up with that, the `mkcameofw` tool used to build the old factory images still exists in firmware-utils, but is now unused in OpenWrt. It shouldn't be too hard to reintroduce the factory images on the OpenWrt side if someone has a device for testing.

edeso, 2023-07-30:
the wiki page says https://openwrt.org/toh/d-link/dir-505#installation

> The latest openwrt releases are based on the ath79 architecture using device trees. Therefore an old installation image based on an ar71xx architecture has to be used for the initial installation.

so it reads as if there is more to it than just reenabling the firmware tool.

rotanid, 2023-08-02:
i don't think this would be worth any effort, given these aren't sold as new anymore and you can always use any older factory for flashing...