freifunk-gluon/gluon #2930: MT7615 DBDC - 2.4GHz phy announces 5GHz support (switch case fall through)

Djfe, 2023-07-27:
Upstream just fixed an old-standing issue with MT7615 DBDC WiFi
https://github.com/openwrt/mt76/commit/152608a40aa715409acd210b82e383771733d5bf
Affected devices in Gluon could be:
TP-Link RE550 and RE650
GL-iNet MT1300

@ambassador86 If you got the time, then it would be nice if you could run a few commands for me on your GL.iNet MT1300 and then send us the output. (with Gluon installed, v2022.1.4 or master)

```
iw list
uci show wireless
```

blocktrron, 2023-07-27:
Can you elaborate how this affects Gluon? Our PHY distribution logic does ensure we distribute multiband PHYs evenly and in this case we assign the band-variable PHY to the 2.4 GHz radio.

See https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/luasrc/lib/gluon/upgrade/200-wireless#L15

Djfe, 2023-07-27:
Now that's great, that likely means, that Gluon isn't affected negatively by the bug, awesome.
Since I don't have access to an affected device: it might still be useful to get the output of the commands above in order to confirm this :)

Djfe, 2023-08-10:
I'll close this for now. If anyone actually reports issues we may still take action

closed 2023-08-10 by Djfe