freifunk-gluon/gluon #2935: 2023.1 on has dnsmasq clogging the CPU

smoe, 2023-08-13:
The machine (https://coco.ffoh.de/meshviewer/#!/en/map/10feede60b26)[runs fine] and is performant but the CPU is kept busy with dnsmasq:
```
 1254  1209 dnsmasq  R     3164   3%  97% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411c.pid
16375     2 root     IW       0   0%   1% [kworker/u2:0-ba]
```
This does not go away, also not after a reboot.  The site.conf shows
```
        dns = {
                cacheentries = 5000,
                servers = { 'fd73:111:e824::1', },
        },

        next_node = {
                name = { 'nextnode.ffoh.de', 'nextnode', 'nn' },
                ip6 = 'fd73:111:e824::1',
                ip4 = '10.135.0.1',
                mac = '16:41:95:40:f7:dc',
        },
```
Have not yet tried any other hardware - will do :)

smoe, 2023-08-13:
This issue is reproducible on a second 4300, meshing with the first, no contact with gateway, no users.

skorpy2009, 2023-08-25:
Hey @smoe 

You have configured a loop. The dnsmasq querys it self the way you configured it.
```
        dns = {
                cacheentries = 5000,
                servers = { 'fd73:111:e824::1', },
        },
```
        
dns servers needs to be an upstream server, maybe you run a unbound on your gateway or centrally in your backbone.
https://gluon.readthedocs.io/en/latest/features/dns-cache.html has a wrong config example but the text describes it correctly.

Best

skorpy

smoe, 2023-08-25:
That was indeed helpful. Many thanks.

closed 2023-09-06 by rotanid