freifunk-gluon/gluon #2938: Deactivate mesh_vpn if no interface has role "uplink"

maurerle, 2023-08-21:
For now, we leave mesh_vpn activated, even if no role has uplink configured.

We should activate mesh_vpn if there is at least one "uplink" role on gluon-reconfigure and deactivate it, if there is non anymore.

To activate mesh_vpn, current instructions include:
`uci set gluon.mesh_vpn.enabled=1; uci set gluon.core.reconfigure=1; uci commit gluon; reboot` (neoraider, IRC)

wouldn't it be better to completely switch to roles for these things, as there already is a way to manage if we want to act as uplink, mesh or client?

Djfe, 2023-10-03:
This should be part of the config mode overhaul.
Right now we have general settings:
-  deactivate/activate vpn

and we have advanced settings:
- set the uplink port for vpn (or set none)

regular users might never visit advanced settings.
We therefore can't just kill the vpn checkbox from the general page.

Also there is the issue that general settings are only saved on rebooting. so advanced settings cannot really depend on them unless the config mode decides this on `POST` and overwrites your decision in the general settings.

It's a dilemma we can only clear up with a redesign, I think.
I'm also wondering whether uplink is the best name, atleast in German we could do with a translation or some explanation on the config mode page, what the new settings are meant for.
Even the old config mode had this flaw:
Enabling mesh-on-wan didn't disable vpn, right?

Djfe, 2023-10-04:
I just had an extra thought: what happens if you want to bridge wan to another port without doing vpn?

overall it could be interesting to have maybe more than one wan port for bridging.
or the option to bridge wwan (cellular) to one lan port (not lan mesh or client) and actual wan to another port

neocturne, 2023-10-04:
I don't think VPN and interface role configuration should interact.

- VPN enable/disable is a simple flag that can also be set in the wizard. There is no clear way how this flag could be mapped to interface roles. Disabling VPN must not make a node forget which interfaces are assigned as uplink (as the same interface should be used as soon as VPN is enabled)
- Interface roles are an advanced setting. We can expect advanced users to disable VPN when they don't have any uplink ports set (or just keep it enabled, because it doesn't really matter)

@Djfe
- Mesh-on-WAN was never supposed to disable VPN; I'm not sure where you see a flaw. Using both on the same port is a supported configuration
- We currently bridge uplink ports if there are multiple configured, but whether this is a bug or a feature is debatable
- WWAN can't be bridged, it is not an Ethernet-like interface
- Please try to keep your comments on-topic

maurerle, 2023-10-04:
Thanks, I agree. I close this issue therefore.

closed 2023-10-04 by maurerle