freifunk-gluon/gluon #2939: Read-Only filesystem on Ubiquiti airmax devices

blocktrron, 2023-08-22:
### Bug report

**What is the problem?**

Several Ubiquiti airmax devices report a Read-only filesystem with OpenWrt 23.05

Upstream ticket: https://github.com/openwrt/openwrt/issues/12882

**What is the expected behaviour?**

The rootfs_data partition is not read-only

**Gluon Version:**

Gluon based on OpenWrt 23.05

**Site Configuration:**
n/a

**Custom patches:**
n/a

blocktrron, 2023-12-10:
Removing this ticket for the upcoming v2023.2 milestone, as the devices were removed from Gluon. 

Keeping this ticket open for visibility.

rotanid, 2023-12-26:
@blocktrron this seems to be solved upstream?

blocktrron, 2023-12-27:
DO you have a commit as reference?

rotanid, 2023-12-27:
@blocktrron f024f4b1b0380b3b2e18115bd8e4f35393fccc70 was mentioned here https://github.com/openwrt/openwrt/pull/14237

herbetom, 2024-01-02:
I could validate te fix on the NanoBeam 5AC 19 (XC). Hoewer that device is currently in Darmstadt and it's currently raining which doesn't help with my motivation to do so right now.

Hoewer, in theory we could validate it and reintroduce support before the next release.

blocktrron, 2024-01-02:
@rotanid I've seen this commit, I'd like to have this verified before reverting though.

nrbffs, 2024-01-13:
We have M5 Loco XM (5x) and M5 Loco XW (3x). Is the fix already in master? What needs to be done to verify this?

rotanid, 2024-04-21:
https://github.com/openwrt/openwrt/commit/c55aaa7c9a98e7c0a5e1ea8293a534dc5b395cf3

@blocktrron shouldn't this be part of gluon already which would render this issue fixed?

nrbffs, 2025-03-01:
As far as I understand, the fix from OpenWRT commit c55aaa7c9a ("ath79: generic: disable SPI-NOR write protect unconditionally") is now in Gluon main. 

To drive this issue forward, I would do the following:

- make a build of latest gluon main, with reverted a0f5b4e9468 ("ath79-generic: remove Ubiquiti AirMax targets (#3067)")
- flash it onto a Ubiquiti Nanostation Loco M (XW) or Ubiquiti Nanostation M (XW)  - does not matter if M2 or M5
- using `cat /proc/mtd`, find the mtd device which corresponds to the firmware partition
-  `cat /sys/block/mtdblock*/ro` and verify that it shows `0`
- click through configmode and verify the device reboots fine and persists its config (i.e. doesn't come up in config mode again)
- sysupgrade (loosing settings) to an older version to verify upgrade works

If the above test was successful and nobody objects until next week, I will prepare a PR for revert of a0f5b4e9468.

herbetom, 2025-03-01:
@nrbffs Yes, let's get this hopefully over with. I've rebased #3141 which is a revert of a0f5b4e9468e9c2fa28459143a7df6ced348f452. Locally it seems to build. So i think the CI should succeed as well.

The testing procedure sounds fine to me.

I'll test it on the NanoBeam 5AC 19 (XC) today or tommorrow and if you can also provide some testing results (ideally from another affected Model) then we should have something mergeable with a reasonable high degree of confidence.

herbetom, 2025-03-03:
These appear to be the current Upstream Issues:
- https://github.com/openwrt/openwrt/issues/17234
- https://github.com/openwrt/openwrt/issues/17285

(But there are more.)

And then there is a PR which claims to fix it:
- https://github.com/openwrt/openwrt/pull/17287