freifunk-gluon/gluon #2947: Feature: allow configuration of wlan radio bandwidth

skorpy2009, 2023-08-26:
While I can configure it on the device itself, currently only 20 MHz of channel bandwidth is used by Gluon and is not configurable per-band in the site configuration.

To achieve this, the site should be able to indicate the desired maximum bandwith (20/40/80/160) while Gluon should try to pair the widest supported channel bandwith with the most efficient HT-mode.

We mainly have to extand https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/luasrc/lib/gluon/upgrade/200-wireless#L64 and update https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/check_site.lua#L34

neocturne, 2023-09-07:
I believe one of the issues here was that hostapd will automatically reduce the channel width to 20MHz whenever another BSSID on one of the extension channels is detected (which will always be the case for mesh nodes), unless the noscan option is set.

Maybe this can be changed to be handled separately for AP and mesh VIFs (or maybe something about this has already changed in the last years since I last looked at channel width handling).

blocktrron, 2023-09-07:
@neocturne The behavior you are referring to was limited to 2.4 GHz IIRC and does not apply to 5GHz.

closed 2026-01-15 by rotanid