freifunk-gluon/gluon #2967: Gluon loses config when updating x86 devices from v2021.1.2 to v2023.1

lephisto, 2023-09-05:
### Bug report

**What is the problem?**

We are updating from v2021.1.2 to v2023.1. Upgrade works 95% fine on Hardware routers/APs. On VMs and Futros the upgrade procedure reproduceably kills the Config. after reboot the VM/Offloader is in Config mode.

**What is the expected behaviour?**

I would expect the previous config to persist

**Gluon Version:**

v2021.1.2

**Site Configuration:**

https://github.com/ffrgb/site-ffrgb

**Custom patches:**

Nothing fancy.. 

**Screenshot - if that help:**

[image]

blocktrron, 2023-09-05:
@lephisto Do you know the disk size of said systems?

lephisto, 2023-09-05:
In this case I just did a `qm importdisk` so the disk size went to 121MB. 

It happened already on a variety of disk sizes.

maurerle, 2023-09-05:
I can confirm this issue with a vm upgraded from v2021.1.2 to v2023.1
Error message is the same, including the:

> Partition layout has changed. Full image will be written

Would be quite good to have a working upgrade path directly from the last openwrt-19 gluon release for us

rotanid, 2023-09-05:
as there were no similar reports from v2021 to v2022, the issue has to be somewhat related to Gluon v2023.1 and not OpenWrt 22.03 ...

neocturne, 2023-09-06:
The bug is in 2021.1.x, it fails to rescan the partition table when the layout has changed for some reason:
```
Commencing upgrade. Closing all shell sessions.
killall: telnetd: no process killed
Sending TERM to remaining processes ... ubusd askfirst urngd logd dnsmasq gluon-arp-limit netifd uradvd micrond sse-multiplexd uhttpd dnsmasq ntpd alfred batadv-vis respondd 
Sending KILL to remaining processes ... ntpd 
Switching to ramdisk...
[   37.438782] EXT4-fs (loop0): re-mounted. Opts: data=ordered
Performing system upgrade...
Reading partition table from bootdisk...
0+1 records in
0+1 records out
zcat: write error: Broken pipe
zcat: write: Broken pipe
Reading partition table from image...
30791+1 records in
30791+1 records out
partx: /dev/sda: error deleting partition 2
partx: /dev/sda: error adding partition 2
mount: mounting /dev/sda1 on /mnt failed: Invalid argument
umount: can't unmount /mnt: Invalid argument
Upgrade completed
Rebooting system...
```

partx should only fail like that when the partition is still busy, so maybe the umount of the rootfs is not working correctly...

neocturne, 2023-09-06:
Updating from 2021.1.2 to 2022.1.1 to 2023.1 is working as expected, it only breaks when 2022.1 is skipped. As a first step, we should add a warning to the 2023.1 release notes.

I see two possible ways to proceed:
- Revert EFI boot support in 2023.1.x for now to avoid partition table change (or keep EFI as separate images)
- Accept that 2022.1.x is required as an intermediate step (I don't think it makes sense to put much effort into a fix on 2021.1.x, as an intermediate update from 2021.1.2 to 2023.1.x would be required anyways...)

Djfe, 2023-09-06:
is it possible to remove efi boot support without breaking anything else? then I'm all for it.
move efi boot support to v2023.2.x

Djfe, 2023-09-07:
I just informed the forum about this as a side note (in the changelog thread opened by @skorpy2009 )
Just to prevent further accidents.
I'll update the post Immediately once this issue got fixed properly

neocturne, 2023-10-10:
Fixed with v2023.1.1.

PR #2972 (2023-09-09): Revert "x86: switch to EFI-compatible images"
This reverts commit 41b8ecd6177f3916ec2ad16bf0f1a1551e77bfd4.

Unbreak upgrades from OpenWrt 19.07 (Gluon 2021.1.x) by undoing the partition table change.

It should be noted that the EFI images generated by OpenWrt are slightly smaller than the non-EFI images for some reason (due to a smaller gap between the boot and root partitions), so upgrading a VM created using a Gluon 2023.1 (EFI) image to a newer non-EFI image will truncate the root partition.

Linux will warn about the truncation on boot ("sda: p2 size 212992 extends beyond EOD, truncated" [... cut]
commit 4e2bd01d (2023-09-09): Revert "x86: switch to EFI-compatible images"

This reverts commit 41b8ecd6177f3916ec2ad16bf0f1a1551e77bfd4.

Unbreak upgrades from OpenWrt 19.07 (Gluon 2021.1.x) by undoing the
partition table change.

It should be noted that the EFI images generated by OpenWrt are slightly
smaller than the non-EFI images for some reason (due to a smaller gap
between the boot and root partitions), so upgrading a VM created using a
Gluon 2023.1 (EFI) image to a newer non-EFI image will truncate the root
partition.

Linux will warn about the truncation on boot ("sda: p2 size 212992 extends
beyond EOD, truncated [... cut]
closed 2023-10-10 by neocturne