freifunk-gluon/gluon #2968: gluon device names: model name vs marketing name, cleaning up existing device names

Djfe, 2023-09-05:
This is a continuation of questions that arose from #2966
Let's create a new issue to discuss this under ^^

Gluon uses model names for it's devices. But it doesn't do it for all supported devices, yet.
I could fix this with a PR, if desired. What's the maintainers' opinion on that?

The Mikrotik hAP AC² should/could actually be renamed to `RBD52G-5HacD2HnD-TC`?
https://github.com/freifunk-gluon/gluon/blob/ced0ccaf2bbb26f94df30608b47184125e644862/targets/ipq40xx-mikrotik#L5

here it says hAP in brackets
Would additional names in the supported_devices.rst be allowed? I feel like they have a use.
https://github.com/freifunk-gluon/gluon/blob/ced0ccaf2bbb26f94df30608b47184125e644862/docs/user/supported_devices.rst?plain=1#L170

Zyxel WSM20 (Multy M1) <- would work for me, but your decision.

More examples where the marketing name was preferred:
https://github.com/freifunk-gluon/gluon/blob/master/docs/user/supported_devices.rst?plain=1#L239-L241

same goes for Aruba:

https://github.com/freifunk-gluon/gluon/blob/ced0ccaf2bbb26f94df30608b47184125e644862/docs/user/supported_devices.rst?plain=1#L192

model name appears to be actually: APIN0303

An example how the backside of Aruba APs look like:
Aruba InstantOn AP12 (RW)
[image]
InstantOn AP11(D)/AP17 are clearly marketing and not model names.

neocturne, 2023-09-07:
Naming in the target files is determined by OpenWrt, this must match what libplatforminfo reports for the autoupdater to work (adding aliases or manifest aliases when different devices use the same image or image names have changed). If the name of a device is wrong, that is a bug and must be fixed.

In the docs, listing alternative marketing names sense.

Djfe, 2023-09-11:
I just stumbled across devolo and their MT numbers.
Those are probably just type numbers and don't count as a model name, right?
https://www.devolo.de/support/ce

Example:
the devolo WiFi pro 1750e has MT: 2785

neocturne, 2025-04-04:
Closing, as the naming is defined in OpenWrt, not Gluon. Patches for adding additional information to the docs that help identify a device are always welcome.

closed 2025-04-04 by neocturne

Djfe, 2025-04-04:
Works for now, I had started with the PR back then, I'll pick it up at same point again.
Once it's complete one can think about improving naming upstream but this was primarily about improving our docs for supported devices.