freifunk-gluon/gluon #2977: tunneldigger tunnel drops suddenly since 2023.1.x

cmarxmeier, 2023-09-11:
### Bug report

**What is the problem?**

while in gluon 2022.x works properly, with 2023.1 mesh_vpn starts, switches Traffic Routing from wifi mesh to tunnel and suddenly tunnel ist dropped at TQ <80%. This behaviour loops over and over.

**What is the expected behaviour?**

in gluon 2022.x and before, tunneldigger tunnel bahaved  properly, Switching node back to earlier firmware release stops dropping of tunnel
**Gluon Version:**

Problem started with first released Version of v2023.1.x
**Site Configuration:**

example site:
https://github.com/Freifunk-Rhein-Sieg/ffrsk-site/tree/v2023.x_lo-lo_Lohmar
**Custom patches:**

openwrt patched to build iw-full  like this:  https://github.com/Freifunk-Rhein-Sieg/ffrsk-site/blob/tools/patches/0099-mac8011-iwfull.patch
respondd patched back to former config: https://github.com/Freifunk-Rhein-Sieg/ffrsk-site/blob/tools/patches/package/102-fix-respondd-rsk.patch

cmarxmeier, 2023-09-11:
syslog on node:
Mon Sep 11 10:25:00 2023 daemon.info micrond[2429]: SSID Freifunk is correct, noting to do
Mon Sep 11 10:25:00 2023 daemon.warn td-client: Got termination signal, shutting down tunnel...
Mon Sep 11 10:25:00 2023 daemon.notice netifd: Network device 'mesh-vpn' link is down
Mon Sep 11 10:25:00 2023 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity loss
Mon Sep 11 10:25:00 2023 daemon.err tunneldigger[7060]: td-client: Got termination signal, shutting down tunnel...
Mon Sep 11 10:25:00 2023 kern.info kernel: [ 1286.731735] batman_adv: bat0: Interface deactivated: mesh-vpn
Mon Sep 11 10:25:00 2023 kern.info kernel: [ 1286.759592] batman_adv: bat0: Removing interface: mesh-vpn
Mon Sep 11 10:25:01 2023 daemon.notice netifd: Interface 'mesh_vpn' is now down

skorpy2009, 2023-09-11:
FYI: fastd L2TP works like a charm

cmarxmeier, 2023-09-11:
@skorpy2009 : fastd L2TP is 'Off topic'.

cmarxmeier, 2023-09-12:
tunneldigger from gluon 2023.1.x does not create .pid file in /var/run/tunneldigger.mesh-vpn.pid any more - therefore gluon-tunneldiugger-watchdog assumes failure and restarts tunneldigger. when started by minicrond.

cmarxmeier, 2023-09-12:
quick and dirty workaround: 
in /gluon/package/gluon-mesh-vpn-tunneldigger/luasrc/usr/bin/tunneldigger-watchdog:

local function restart_tunneldigger()
        os.execute('logger -t tunneldigger-watchdog "Restarting Tunneldigger is disabled."')
        os.execute('logger -t tunneldigger-watchdog "gluon v2023.1.x misses .pid file."')
--      os.execute('/etc/init.d/tunneldigger restart')
end

neocturne, 2023-09-12:
Ah, tunneldigger has been migrated to procd. I don't think we can just remove the watchdog (unless there have been upstream improvements that I'm not aware of), so the watchdog script will need to be updated to query procd whether tunneldigger should be running, or something similar.

Does anyone know what state of tunneldigger the watchdog is usually supposed to trigger on? Does the `mesh-vpn` interface still exist and just not allow data to pass through, or is `mesh-vpn` missing altogether?

cmarxmeier, 2023-09-12:
quick and dirty patch to generate needed .pid from tunneldigger script sent to gluonv2023.1.x tree
https://github.com/freifunk-gluon/gluon/pull/2979

commit 8f5286d4 (2023-09-12): tunneldigger-watchdog: remove broken restart check

The tunneldigger init script has been migrated to procd, so there is no
PID file anymore. Remove the obsolete check altogether - it shouldn't
be needed anymore, as procd will take care of crashes of tunneldigger by
itself. This fixes the watchdog resetting tunneldigger every time it runs.

Closes #2977
commit 7a43dabc (2023-09-12): gluon-mesh-vpn-tunneldigger: tunneldigger-watchdog: remove broken restart check

The tunneldigger init script has been migrated to procd, so there is no
PID file anymore. Remove the obsolete check altogether - it shouldn't
be needed anymore, as procd will take care of crashes of tunneldigger by
itself. This fixes the watchdog resetting tunneldigger every time it runs.

Closes #2977
PR #2981 (2023-09-17): gluon-mesh-vpn-tunneldigger: tunneldigger-watchdog: remove broken restart check
The tunneldigger init script has been migrated to procd, so there is no PID file anymore. Remove the obsolete check altogether - it shouldn't be needed anymore, as procd will take care of crashes of tunneldigger by itself. This fixes the watchdog resetting tunneldigger every time it runs.

Closes #2977
commit d9f8387e (2023-09-17): gluon-mesh-vpn-tunneldigger: tunneldigger-watchdog: remove broken restart check

The tunneldigger init script has been migrated to procd, so there is no
PID file anymore. Remove the obsolete check altogether - it shouldn't
be needed anymore, as procd will take care of crashes of tunneldigger by
itself. This fixes the watchdog resetting tunneldigger every time it runs.

Closes #2977

(cherry picked from commit 7a43dabc8f5d6466c8bd7b82b93a8bdfb356ea55)
closed 2023-09-17 by neocturne