freifunk-gluon/gluon #2988: Github wiki: Commandline administration needs to be updated

Djfe, 2023-09-18:
just to keep a note about this, since this came up on irc some time back:
the wiki page about `Commandline administration` could use some love.
gluon-reconfigure and gluon-reload aren't explained, yet.
WAN/LAN was updated to the new gluon-reconfigure syntax. Atleast vpn still needs the same overhaul

sometimes it's
`uci commit gluon`
then
`uci commit`
then
`uci commit gluon.setting`
These could all be unified into `uci commit gluon`
Else some newbies might be left wondering what the difference is.

If I notice anything else that could be improved, I'll add some comments below

I could try myself at updating some of these but I haven't developed them so I might be missing lots of important bits.
I'm more of a user in that regard still, who reads these pages, and not one, who knows it already and writes them. 😅

https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration

rotanid, 2023-09-21:
> sometimes it's `uci commit gluon` then `uci commit` then `uci commit gluon.setting` These could all be unified into `uci commit gluon` Else some newbies might be left wondering what the difference is.

i think you should always only commit the part you have actually changed.
nonetheless, when would this issue ever be closed, can there be a final state?
i think: change it or don't change it, but there's not much gain from having an issue about it

Djfe, 2023-09-22:
This issue was meant as a way to collect a list of parts that need updates.
And as a way to raise attention that the page is out-of-date.

What is your opinion on creating a legacy page?
An alternative would be using spoilers for legacy commands.

Putting a goal on this is a good idea. Else this note will never be closed.
This issue shall be closed after 01.11.23
Until then my goals are:
- Find all out-of-date sections
- Ask developers what the current way of solving these is
- Write all changes into the Text
- Get a review for my changes
-

maurerle, 2023-10-08:
I updated the github wiki docs:
* update uci commit
* clarify what gluon-wan does
* add VLAN example with roles API
* update the config (maybe someone could look over the  reverse switch mode section?)
* clarify what gluon-reconfigure does

https://github.com/freifunk-gluon/gluon/wiki/Commandline-administration#network-reconfiguration

So if there are no other known problems, we can close this issue.

closed 2024-02-13 by Djfe