freifunk-gluon/gluon #2998: l2tp tunnel down after sysupgrade (v2021.1.1 to v2023.1)

Vogelbecker, 2023-09-26:
**What is the problem?**

- What is not working as expected?
I've build a new firmware v2023.1.x. yesterday so #2991 should be already included.
- How is it misbehaving?
On my test-devices the l2tp tunnels are not comming up. 
- When did the problem first start showing up?
Testdevices are on gluon-v2021.1.1-33-gf77f3829+ 
Newly build firmware has been maunally copied to the device. Sysupgrade has been proceeded manually.
Firmware-update has been successful, but the tunnels are not comming up.
- What were you doing when you first noticed the problem?
I'm able to connect to the device by ssh. 
Logread:
````
[...]
Sun Apr 17 07:35:56 2022 daemon.err td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:56 2022 daemon.err td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:56 2022 daemon.err tunneldigger[8180]: td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:56 2022 daemon.err tunneldigger[8180]: td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:57 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:35:57 2022 daemon.err tunneldigger[8180]: td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:35:57 2022 daemon.info td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:35:57 2022 daemon.err tunneldigger[8180]: td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:35:59 2022 daemon.err td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:59 2022 daemon.err td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:59 2022 daemon.err tunneldigger[8180]: td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:35:59 2022 daemon.err tunneldigger[8180]: td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:36:00 2022 daemon.info micrond[2290]: node is considered offline
Sun Apr 17 07:36:00 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:36:00 2022 daemon.err tunneldigger[8180]: td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:36:00 2022 daemon.info td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:36:00 2022 daemon.err tunneldigger[8180]: td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
[...]
````

gluon-wan:
````
{
        "up": true,
        "pending": false,
        "available": true,
        "autostart": true,
        "dynamic": false,
        "uptime": 1418,
        "l3_device": "br-wan",
        "proto": "dhcp",
        "device": "br-wan",
        "updated": [
                "addresses",
                "routes",
                "data"
        ],
        "metric": 0,
        "dns_metric": 0,
        "delegation": true,
        "ipv4-address": [
                {
                        "address": "172.16.1.156",
                        "mask": 24
                }
        ],
        "ipv6-address": [

        ],
        "ipv6-prefix": [

        ],
        "ipv6-prefix-assignment": [

        ],
        "route": [
                {
                        "target": "0.0.0.0",
                        "mask": 0,
                        "nexthop": "172.16.1.1",
                        "source": "172.16.1.156/32"
                }
        ],
        "dns-server": [

        ],
        "dns-search": [

        ],
        "neighbors": [

        ],
        "inactive": {
                "ipv4-address": [

                ],
                "ipv6-address": [

                ],
                "route": [

                ],
                "dns-server": [
                        "172.16.1.51",
                        "8.8.8.8"
                ],
                "dns-search": [
                        "home"
                ],
                "neighbors": [

                ]
        },
        "data": {
                "dhcpserver": "172.16.1.1",
                "leasetime": 7200
        }
}
````

/var/gluon/wan-dnsmasq/resolv.conf
````
nameserver 172.16.1.51
nameserver 8.8.8.8
````

gluon-wan nslookup heise.de
```
Server:         127.0.0.1
Address:        127.0.0.1:53

Non-authoritative answer:
Name:   heise.de
Address: 193.99.144.80

Non-authoritative answer:
Name:   heise.de
Address: 2a02:2e0:3fe:1001:302::
```

- On which devices (vendor, model and revision) is it misbehaving?
Ubiquiti UniFi-AC-MESH
TP-Link CPE210 v1.1
- Does the issue appear on multiple devices or targets?
yes, see above

**What is the expected behaviour?**
- How do you think it should work instead?
L2tp tunnels should work.
- Did it work like that before?
on v2021.1.1 the tunnels are ok.

**Gluon Version:**
v2023.1-16-g48582f24

**Site Configuration:**
https://github.com/Vogelbecker/ffein-site/blob/master/site.conf

neocturne, 2023-09-26:
Duplicate of #2977, will be fixed with v2023.1.1.

neocturne, 2023-09-26:
Ugh sorry, I should have fully read the log. Seems to be a different issue.

neocturne, 2023-09-26:
- Does `gluon-wan nslookup` work for the two hosts appearing in the log?
  - How long does the lookup take?
- Does the log change if you manually restart tunneldigger via SSH?

Vogelbecker, 2023-09-26:
Yes, I get the correct ips for both hosts. I get nearly instant the the answer. 
````
root@ffein-vogelbeck-feuerwehr:~# gluon-wan nslookup host1.freifunkeinbeck.de
Server:		127.0.0.1
Address:	127.0.0.1:53

Non-authoritative answer:
Name:	host1.freifunkeinbeck.de
Address: 51.254.208.25

Non-authoritative answer:

root@ffein-vogelbeck-feuerwehr:~# gluon-wan nslookup host2.freifunkeinbeck.de
Server:		127.0.0.1
Address:	127.0.0.1:53

Non-authoritative answer:

Non-authoritative answer:
Name:	host2.freifunkeinbeck.de
Address: 78.46.71.175
````

No, restarting with `/etc/init.d/tunneldigger restart` and its still the same
````
Sun Apr 17 15:26:57 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:26:57 2022 daemon.err tunneldigger[21469]: td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:26:58 2022 daemon.info td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:26:58 2022 daemon.err tunneldigger[21469]: td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:26:58 2022 daemon.warn td-client: Got termination signal, shutting down tunnel...
Sun Apr 17 15:26:58 2022 daemon.err tunneldigger[21469]: td-client: Got termination signal, shutting down tunnel...
Sun Apr 17 15:26:58 2022 daemon.info td-client: Performing broker selection...
Sun Apr 17 15:26:58 2022 daemon.err tunneldigger[21928]: td-client: Performing broker selection...
Sun Apr 17 15:27:01 2022 daemon.info micrond[2290]: node is considered offline
Sun Apr 17 15:27:02 2022 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Sun Apr 17 15:27:02 2022 daemon.warn hostapd: Failed to check if DFS is required; ret=-1
Sun Apr 17 15:27:02 2022 daemon.err td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 15:27:02 2022 daemon.err td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 15:27:02 2022 daemon.err tunneldigger[21928]: td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 15:27:02 2022 daemon.err tunneldigger[21928]: td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 15:27:03 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:27:03 2022 daemon.err tunneldigger[21928]: td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:27:03 2022 daemon.info td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:27:03 2022 daemon.err tunneldigger[21928]: td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 15:27:04 2022 daemon.err td-client: [host1.freifunkeinbeck.de:20001] Failed to resolve hostname.
Sun Apr 17 15:27:04 2022 daemon.err tunneldigger[21928]: td-client: [host1.freifunkeinbeck.de:20001] Failed to resolve hostname.
Sun Apr 17 15:27:05 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.

````

neocturne, 2023-09-26:
Ah, I see what is going on. The tunneldigger package from gluon/packages was replaced with the a new package in openwrt/packages, which doesn't support the `group` UCI setting required to use the WAN dnsmasq instance.

neocturne, 2023-09-26:
Can you check if #2999 fixes the issue for you?

Vogelbecker, 2023-09-26:
Hi,

thanks, on the first test device it is working perfect:

````
Sun Apr 17 07:16:06 2022 daemon.notice netifd: Interface 'wan' is now up
Sun Apr 17 07:16:07 2022 authpriv.info dropbear[4244]: Child connection from 172.16.1.43:51786
Sun Apr 17 07:16:07 2022 user.notice firewall: Reloading firewall due to ifup of wan (br-wan)
Sun Apr 17 07:16:08 2022 daemon.err td-client: [vpn2.freifunk-einbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 daemon.err td-client: [vpn3.freifunk-einbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 daemon.err td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 daemon.err tunneldigger[3306]: td-client: [vpn2.freifunk-einbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 daemon.err tunneldigger[3306]: td-client: [vpn3.freifunk-einbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 daemon.err tunneldigger[3306]: td-client: [host1.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:08 2022 authpriv.notice dropbear[4244]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:YfB5JaCmRC+2TN585aXzcYzxxMLwi6zmGBWdmSmSc1o from 172.16.1.43:51786
Sun Apr 17 07:16:09 2022 daemon.info td-client: [vpn2.freifunk-einbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.err tunneldigger[3306]: td-client: [vpn2.freifunk-einbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.info dnsmasq[3120]: reading /var/gluon/wan-dnsmasq/resolv.conf
Sun Apr 17 07:16:09 2022 daemon.info dnsmasq[3120]: using nameserver 172.16.1.51#53
Sun Apr 17 07:16:09 2022 daemon.info dnsmasq[3120]: using nameserver 8.8.8.8#53
Sun Apr 17 07:16:09 2022 daemon.info td-client: [vpn3.freifunk-einbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.err tunneldigger[3306]: td-client: [vpn3.freifunk-einbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.info td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.err tunneldigger[3306]: td-client: [host1.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:09 2022 daemon.err td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:09 2022 daemon.err tunneldigger[3306]: td-client: [host2.freifunkeinbeck.de:20001] Hostname resolution timed out.
Sun Apr 17 07:16:10 2022 daemon.info td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:10 2022 daemon.err tunneldigger[3306]: td-client: [host2.freifunkeinbeck.de:20001] Reinitializing tunnel context.
Sun Apr 17 07:16:11 2022 daemon.debug td-client: [vpn2.freifunk-einbeck.de:20001] Broker usage: 3246
Sun Apr 17 07:16:11 2022 daemon.info td-client: Selected vpn2.freifunk-einbeck.de:20001 as the best broker.
Sun Apr 17 07:16:11 2022 daemon.err tunneldigger[3306]: td-client: [vpn2.freifunk-einbeck.de:20001] Broker usage: 3246
Sun Apr 17 07:16:11 2022 daemon.err tunneldigger[3306]: td-client: Selected vpn2.freifunk-einbeck.de:20001 as the best broker.
Sun Apr 17 07:16:12 2022 daemon.notice netifd: Interface 'mesh_vpn' is enabled
Sun Apr 17 07:16:12 2022 daemon.info td-client: [vpn2.freifunk-einbeck.de:20001] Tunnel successfully established.
Sun Apr 17 07:16:12 2022 daemon.notice netifd: Network device 'mesh-vpn' link is up
Sun Apr 17 07:16:12 2022 daemon.info td-client: [vpn2.freifunk-einbeck.de:20001] Setting MTU to 1364
Sun Apr 17 07:16:12 2022 daemon.err tunneldigger[3306]: td-client: [vpn2.freifunk-einbeck.de:20001] Tunnel successfully established.
Sun Apr 17 07:16:12 2022 daemon.err tunneldigger[3306]: td-client: [vpn2.freifunk-einbeck.de:20001] Setting MTU to 1364
Sun Apr 17 07:16:12 2022 daemon.notice netifd: Interface 'mesh_vpn' has link connectivity
Sun Apr 17 07:16:12 2022 daemon.notice netifd: Interface 'mesh_vpn' is setting up now
Sun Apr 17 07:16:12 2022 daemon.notice netifd: Interface 'mesh_vpn' is now up
Sun Apr 17 07:16:12 2022 kern.info kernel: [  134.349399] batman_adv: bat0: Adding interface: mesh-vpn
Sun Apr 17 07:16:12 2022 kern.info kernel: [  134.355056] batman_adv: bat0: The MTU of interface mesh-vpn is too small (1364) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
Sun Apr 17 07:16:12 2022 kern.info kernel: [  134.380212] batman_adv: bat0: Interface activated: mesh-vpn
Sun Apr 17 07:16:13 2022 daemon.err gluon-radv-filterd[2619]: Switching to f2:be:ef:00:01:03 (TQ=0)
Sun Apr 17 07:16:13 2022 user.notice firewall: Reloading firewall due to ifup of mesh_vpn (mesh-vpn)
````

PR #3000 (2023-09-28): Update modules
Fixes openvswitch build with groff 1.23.0+ on build host.

Closes #2998
Supersedes #2999
closed 2023-09-28 by blocktrron