freifunk-gluon/gluon #3003: D-Link DAP-X1860 A1: Wifi crashes immediately after boot

citronalco, 2023-09-28:
### Bug report

**What is the problem?**
One of our users has reported a non working Wifi on two of his D-Link DAP-X1860 (A1) with Gluon-v2023.1-13-g473ff9d
He has several of those devices. All of them work properly, except two of them. Those two do work flawless with Gluon 2022.1-4.

Wifi crashes immediately after boot. 100% reproducible:

dmesg:
```
...
[   43.230583] batman_adv: bat0: IGMP Querier appeared
[   43.235477] batman_adv: bat0: MLD Querier appeared
[   58.140491] mt7915e 0000:02:00.0: Message 00004eed (seq 15) timeout
[   78.620544] mt7915e 0000:02:00.0: Message 000007ed (seq 1) timeout
[  110.620479] mt7915e 0000:02:00.0: Message 00005aed (seq 2) timeout
[  131.100459] mt7915e 0000:02:00.0: Message 00005aed (seq 3) timeout
[  151.580499] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
[  172.060360] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
[  192.540121] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
[  213.019902] mt7915e 0000:02:00.0: Message 00005aed (seq 7) timeout
[  233.499631] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
[  253.979366] mt7915e 0000:02:00.0: Message 00005aed (seq 9) timeout
[  274.459085] mt7915e 0000:02:00.0: Message 00005aed (seq 10) timeout
[  294.938764] mt7915e 0000:02:00.0: Message 00005aed (seq 11) timeout
[  315.418483] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
[  335.898169] mt7915e 0000:02:00.0: Message 00005aed (seq 13) timeout
[  356.377828] mt7915e 0000:02:00.0: Message 00005aed (seq 14) timeout
[  376.857480] mt7915e 0000:02:00.0: Message 00005aed (seq 15) timeout
[  397.337120] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
(continues endlessly)
```

`iwinfo` runs very slow, taking several seconds per line. It displays:
```
client0   ESSID: "Freifunk"
          Access Point: 72:BF:FA:84:EC:D8
          Mode: Master  Channel: 5 (2.432 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

client1   ESSID: "Freifunk"
          Access Point: 72:BF:FA:84:EC:DC
          Mode: Master  Channel: 44 (6.170 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

mesh0     ESSID: unknown
          Access Point: 72:BF:FA:84:EC:D9
          Mode: Mesh Point  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ax/b/g/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

mesh1     ESSID: unknown
          Access Point: 72:BF:FA:84:EC:DD
          Mode: Mesh Point  Channel: unknown (unknown)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: unknown
          Bit Rate: unknown
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11ac/ax/n
          Hardware: 14C3:7915 14C3:7915 [MediaTek MT7915E]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1
```
Needless to say neither 0 dBm TX power nor channel 1 is configured in uci/site.conf.

**Gluon Version:**
v2023.1-13-g473ff9d

**Site Configuration:**
https://git.bingo-ev.de/freifunk/ffin-site/

**Custom patches:**
None.

herbetom, 2023-10-01:
@citronalco Thanks for reporting it.

I can report that i was able to reproduce the described issue. :(

Since i had an [NWA50AX](https://github.com/openwrt/openwrt/commit/a0b7fef0ffe4cd9cca39a652a37e4f3ce8f0a681) on hand and it's (to my understanding) not completly different from the [D-Link DAP-X1860 A1](https://github.com/openwrt/openwrt/commit/3c31f6b521bb5fc49e222a8f2bcd454b73452a0e) i also tested that one and there don't seem to be immediate visible issues.

Djfe, 2023-10-02:
maybe it's the last openwrt commit we added?
nbd168 updated mt76 and actually updated firmware and stuff 
https://github.com/openwrt/openwrt/commit/76b1e564d202c09d0035315eb6e958a9b0dd4491

We should try building without this commit to see if it fixes it.
That would explain why v2023.1 (the tag) was not affected and why v2023.1.1/the current branch is affected.
I might do a git bisect of the wifi driver (branch 22.03) to figure out which commit introduced the issue exactly in order to get this fixed (before release of openwrt 22.03.06)

herbetom, 2023-10-02:
Okay. I've tested a bit more.

- [v2023.1](https://github.com/freifunk-gluon/gluon/tree/v2023.1) was fine.
- [v2023.1.x (6947125)](https://github.com/freifunk-gluon/gluon/commit/6947125c3c807b65ece44e62c12298bd35a8de86) has the problem
- and the mt76 bump seems to be indeed the problem since reverting (#3011) resolves it

rotanid, 2023-10-02:
FYI: until we have confirmed that master (based on 23.05) is not affected, we should leave this issue open even after #3011 is merged

herbetom, 2023-10-02:
I've just confirmed that master seems to be okay. So apparently really just a problem with openwrt-22.03.

Still, i'm not sure if we should close this issue since we've currently only got a workarround and not a real solution (and not even an upstream issue/ticket).
Having those mt76 commits would be nice. Even if i so far at least i haven't really checked what we're missing out of by not having them.

citronalco, 2023-10-04:
I also can confirm it is working with this change.

lukasz1992, 2023-10-26:
1) openwrt-22.03 is stuck on old version of mt76, consider it as forgotten

2) It is also possible that new firmware solves the problem too: https://git01.mediatek.com/plugins/gitiles/openwrt/feeds/mtk-openwrt-feeds/+/dc2bc0ce919472f83efa8c62aa6fe56711887c01

Djfe, 2023-10-27:
actually nbd168 updated mt76 and it's firmware in the 22 branch which caused this regression.
https://github.com/openwrt/openwrt/commits/openwrt-22.03?author=nbd168

I might be able to further bisect the issue this weekend.
hopefully this can be fixed before the next release of v2022.

closed 2024-05-11 by rotanid