freifunk-gluon/gluon #3020: UniFi 6 LR status page - internal server error 500

Djfe, 2023-10-14:
### Bug report

**What is the problem?**
the status page on a UniFi 6 LR errors like this (probably affects all version, this one is a v3, PR pending)
see http://[2a03:2260:3006:115:6222:32ff:fe6a:37d5]/cgi-bin/status

> 500 Interner Serverfehler
> 
> Entschuldigung, auf dem Server ist ein unerwarteter Fehler aufgetreten.
> 
> Failed to execute dispatcher target for entry '/'.
> The called action terminated with an exception:
> /usr/lib/lua/gluon/web/template.lua:43: Failed to execute template 'status-page'.
> A runtime error occurred: /usr/lib/lua/gluon/wireless.lua:26: attempt to concatenate field '?' (a nil value)

**What is the expected behaviour?**
Status page should render

**Gluon Version:**
https://github.com/freifunk-gluon/gluon/commit/1b6ddac3c608961399c4877198dd552c12c3f07b (master 2 weeks ago)

**Site Configuration:**
https://github.com/ffac/site/tree/v2023.2.0-5

**Custom patches:**
quite a few, but none affecting gluon-core
https://github.com/ffac/site/tree/v2023.2.0-5/patches

```
root@DN-Viktoria-Str-46:/sys/devices# find . | grep ieee80211
./platform/18000000.wmac/ieee80211
./platform/18000000.wmac/ieee80211/wl0
./platform/18000000.wmac/ieee80211/wl0/uevent
./platform/18000000.wmac/ieee80211/wl0/power
./platform/18000000.wmac/ieee80211/wl0/power/runtime_active_time
./platform/18000000.wmac/ieee80211/wl0/power/runtime_status
./platform/18000000.wmac/ieee80211/wl0/power/autosuspend_delay_ms
./platform/18000000.wmac/ieee80211/wl0/power/runtime_suspended_time
./platform/18000000.wmac/ieee80211/wl0/power/control
./platform/18000000.wmac/ieee80211/wl0/macaddress
./platform/18000000.wmac/ieee80211/wl0/device
./platform/18000000.wmac/ieee80211/wl0/index
./platform/18000000.wmac/ieee80211/wl0/hwmon0
./platform/18000000.wmac/ieee80211/wl0/hwmon0/uevent
./platform/18000000.wmac/ieee80211/wl0/hwmon0/of_node
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power/runtime_active_time
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power/runtime_status
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power/autosuspend_delay_ms
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power/runtime_suspended_time
./platform/18000000.wmac/ieee80211/wl0/hwmon0/power/control
./platform/18000000.wmac/ieee80211/wl0/hwmon0/device
./platform/18000000.wmac/ieee80211/wl0/hwmon0/subsystem
./platform/18000000.wmac/ieee80211/wl0/hwmon0/temp1_input
./platform/18000000.wmac/ieee80211/wl0/hwmon0/name
./platform/18000000.wmac/ieee80211/wl0/addresses
./platform/18000000.wmac/ieee80211/wl0/subsystem
./platform/18000000.wmac/ieee80211/wl0/address_mask
./platform/18000000.wmac/ieee80211/wl0/name
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/uevent
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power/runtime_active_time
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power/runtime_status
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power/autosuspend_delay_ms
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power/runtime_suspended_time
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/power/control
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/macaddress
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/device
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/index
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/cooling_device
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/addresses
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/subsystem
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/address_mask
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/uevent
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/of_node
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power/runtime_active_time
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power/runtime_status
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power/autosuspend_delay_ms
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power/runtime_suspended_time
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/power/control
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/temp1_crit
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/device
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/throttle1
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/subsystem
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/temp1_input
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/temp1_max
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/hwmon1/name
./platform/1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/ieee80211/wl1/name
```

Djfe, 2023-10-14:
Related commits from Oct. 14th 2022:
https://github.com/openwrt/openwrt/commits?author=nbd168&before=359a6e36e972c06845ffff750f200bca3a95c724+251
https://github.com/openwrt/openwrt/commit/4d323303e7e5743f541e3b41dfb2ac1627e8d96d

I guess, we can use `get_board_phy_name()` from [netifd](https://github.com/openwrt/openwrt/blob/main/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh) now

rotanid, 2023-10-14:
@neocturne what do you think, editing the match-regex to include wl* or using something like get_board_phy_name() method?

neocturne, 2023-10-14:
I would like to understand where this name change comes from first. Is hostapd or some related script doing this? Can we be sure that all PHYs are always consistently renamed? Can we have a `phyX` and `wlX` devices with the same *X*?

rotanid, 2023-10-16:
> I would like to understand where this name change comes from first. Is hostapd or some related script doing this? Can we be sure that all PHYs are always consistently renamed? Can we have a `phyX` and `wlX` devices with the same _X_?

hm, @Djfe linked the commits, did you see them?

maurerle, 2024-01-09:
Seems like the interface names are nowadays taken from the board.json:
https://github.com/openwrt/openwrt/commit/7f9d3a00d8d41f2bf9fe35b30cd689b9992e7ff9#diff-65a80e521d1a8019e3118a4cb3eda1c5400643f0acf7c16a555514fb5563e770R655

This is enabled for some devices mt7622
https://github.com/openwrt/openwrt/commit/34054e4c0635c5820cdab69f3d3979075b4e8795

```
~# cat /etc/board.json 
{
	"model": {
		"id": "ubnt,unifi-6-lr-v3",
		"name": "Ubiquiti UniFi 6 LR v3"
	},
	"network": {
		"lan": {
			"device": "eth0",
			"protocol": "static"
		}
	},
	"wlan": {
		"wl0": {
			"path": "platform/18000000.wmac"
		},
		"wl1": {
			"path": "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
		}
	}
}
```

I think this issue has also been seen on a NETGEAR WAX206.

maurerle, 2024-01-09:
Executing a script with:

```
. /lib/netifd/wireless/mac80211.sh
get_board_phy_name "platform/18000000.wmac"
get_board_phy_name "1a143000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0"
```

returns
```
wl0
wl1
```

So I think that moving to `get_board_phy_name` is the correct fix instead of the hardcoded "phy" in `package/gluon-core/luasrc/usr/lib/lua/gluon/wireless.lua`

kpanic23, 2024-01-29:
Just replaced my node at home with a Xiaomi AX3200, it seems to be affected as well.

rotanid, 2024-03-26:
@Djfe @maurerle @kpanic23 this should be fixed in master branch after #3223

PR #3223 (2024-03-26): gluon-core: lua: wireless: use libiwinfo-lua for find_phy()
find_phy() did not match OpenWrt's logic for PHY names anymore, and could not handle PHYs that don't match the 'phyX' naming scheme. Instead of trying to add more complex logic to our own code, simply do what OpenWrt does and use iwinfo to resolve UCI sections to PHY names.

Supersedes #3155
Closes #3020

Draft: This is entirely untested. Will test on an FB4040 tomorrow - would be great to get a test on a device with wifiX PHY names as well.

There is some subsequent cleanup possible by passing around only UCI section names instead of [... cut]
closed 2024-03-26 by rotanid