freifunk-gluon/gluon #3023: [v2023.1.x] Fritzbox 7520/7530 bricked after update to v2023.1.1

maurerle, 2023-10-15:
### Bug report

**What is the problem?**

The FB7520 and FB7530 do not boot properly since some commit between the tag v2023.1 and v2023.1.1

This bug exists only on this branch, a build from master is working properly.

I don not have serial logs for this bug - the broken firmware can be currently found here:
http://firmware.freifunk-aachen.de/firmware/gluon-ffac-v2023.1.1-1-avm-fritz-box-7530-sysupgrade.bin

The Power LED stays lit in green and the device does nothing.
No Wifi, no Gateway, no ip address..
Power cycle does not help either.

Someone reported that a Computer attached to the switch can talk to another computer on the switch, when the device is on - but the device itself does not boot properly.

We have 13 devices which have this problem.

**What is the expected behaviour?**
The device should work as normal - as bevor v2023.1.1 or on current master.

**Gluon Version:**
v2023.1 and v2023.1.1

**Site Configuration:**
https://github.com/ffac/site

## Recovery
To recover, one has to use the respective AVM tool:
For 7520:
https://download.avm.de/fritzbox/fritzbox-7520/deutschland/recover/
For 7530:
https://download.avm.de/fritzbox/fritzbox-7530/deutschland/recover/
(works with playonlinux/WINE)

And can then run the (improved) fritz flash tool from here:
https://github.com/maurerle/fritz-tools/blob/master/fritzflash.py

Download an initramfs and a working firmware
And run:
`sudo python3 fritzflash.py --dev eno2 --initramfs ./openwrt-22.03.2-ipq40xx-generic-avm_fritzbox-7530-initramfs-fit-uImage.itb --sysupgrade ./gluon-ffac-v2023.1.0-3-avm-fritz-box-7520-sysupgrade.bin`
or execute this command without sudo in an admin prompt on windows.

Djfe, 2023-10-15:
on windows there are definitely a few more differences, I'll update this post, once I have tested a working command.
One difference: you execute `py` instead of `python3`

Djfe, 2023-10-20:
Finally got around to doing the upgrade on a box while being connected via serial
The box hangs at `Starting kernel ...`
https://pastebin.com/0NiNpjqr

Next step:
I'm building branch openwrt-22.03 to see whether it's affected or not.
Then I'm going to do a git bisect.

If openwrt upstream isn't affected then I might need further assistance to build Gluon with some kernel debugging enabled.
I might find some help already in some past github issues for similar booting issues

maurerle, 2023-10-26:
Searching for the other two lines before it dies:

```
Read 1024 bytes from 02bd800
maca pointer invalid: a5472102
Using machid 0x8010001 from environment
 
Starting kernel ...
```

led me to this issue with an unbootable 7530: https://github.com/chunkeey/FritzBox-4040-UBOOT/issues/9
So this might be related.
I currently do not have time to test various commits on the box

grische, 2023-10-26:
@maurerle from what I understand in the ticket you linked, it's either this patch
https://github.com/chunkeey/FritzBox-4040-UBOOT/commit/9d89013f9cc963eca25856c61fa066091d35f8de

Or some bad blocks on the NAND that can be fixed by having a smaller ubifs. I'm surprised that this affected all of the 7530 that were used in their community. 

Or am I missing something here?

blocktrron, 2023-10-26:
The other issue is not related, as ubifs inconsistency would fail the boot process when reading from NAND. In addition, the kernel image checksum is validated prior jumping to the image entry.

Djfe, 2023-10-26:
Sorry, no progress from me since last week.
I wasn't at home last weekend and didn't get any tests on openwrt done this week. Next time I can do some tests (the bisect) is Sunday at the earliest or Wednesday at the latest.

maurerle, 2023-10-26:
current master and 23.05 works

v2023.1.x (e521759b0c4e1735eba5c3da3c9510a09da753d2) is broken
v2023.1.1 is broken
48582f2489c1279f0e5d2a540e56c2b713d24475 is broken
v2023.1 is broken
d651a2cb007f23e1d5091300082ec38f10193b3f is the last working commit

So one still has to bisect the openwrt difference of these changes:
https://github.com/freifunk-gluon/gluon/commit/df79e08d30d63ac2450918fe3b236765de9e5286

maurerle, 2023-10-28:
So the next steps are to check if these openwrt commits are working:

- [ ] 419218af13b4c036b3f27e450855727005b8d141
- [ ] de29f15af173e9434d11a00ffcf437bd6bc97727
- [ ] 188c49b32112aae146a3ac3f870ad486cf1db11f

maurerle, 2023-12-18:
This issue is fixed by the update to latest openwrt based on v22.03.6 in #3087
see https://github.com/openwrt/openwrt/issues/13822#issuecomment-1861806858 for more information

closed 2023-12-18 by maurerle