freifunk-gluon/gluon #3101: gluon-core: don't depend on outdoor_chanlist for enabling outdoor-mode

blocktrron, 2023-12-12:
We might be able to not depend on an outdoor-chanlist in case the detection of country3 is enough (which is already implemented in hostapd) https://w1.fi/cgit/hostap/commit/?id=7ed17eee3adaf96ff2104aeb21554110c9654f9d

blocktrron, 2023-12-12:
This was thought thru when checking https://github.com/freifunk-gluon/gluon/pull/2246 and determining what happens when we update from a firmware which has an outdoor_chanlist set to a firmware without one set (and outdoor mode being previously enabled).

neocturne, 2025-03-31:
@blocktrron As we're already setting country3 - how to determine if it is sufficient?

maurerle, 2026-01-04:
I am quite sure that we had issues with `outdoor_chanlist = '100-140',` in combination with `htmode=HE40`.

https://github.com/ffac/site/blob/v2025.1.x/site.conf#L91
```
outdoor_chanlist = '100-116 132-136', -- exclude/avoid weather radar frequencies
-- exclude 136-140 temporarily to fix 40MHz operation
-- https://homepage.univie.ac.at/albert.rafetseder/RADARs/radars.html
```

As far a I remember the issue was that the node did select [142 in HE40 mode](https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/n/ac/ax/be)) and clients could not see/use the wifi anymore.

Eventually, this would not happen when removing the `outdoor_chanlist` param, but I did not have issues with outdoor and 40mhz bandwidth since this fix.