freifunk-gluon/gluon #3130: only link-local ipv6 address available on gluon node

maurerle, 2023-12-30:
In FFMUC and FFAC we experienced the problem, that a node did only have its fe80: link-local ipv6 address and did not get a

`fdac::` address (prefix6 in FFAC) or `fd62::` address respectively (prefix6 in FFMUC)

normally, this ip is also set when not connected to a gateway.
The node did also not get a public ipv6 address (which is received through radvd).

This was an issue which only happened sometimes and does only affect the ipv6 address on the gluon node itself - clients are getting a proper ipv6 and can connect normally.
The broken node has working respondd at first which stops working at some point of resetting the node..

I did build a few gluon versions in between to track this issue down, as it started to appear somehwere:

works: 1b6ddac3c608961399c4877198dd552c12c3f07b
works: 12807014ebfed6d974722ad02bbd608008382c36
works: e52409f4492ef55ba1059404b496d66858b9c86f
broken: 7426c31487cf2b5f6183742c7fcde67517243a3a
broken: a4d5955c757de409241aa1a041a0fe1d4dcebc25
broken: c2dc338abfbebb34dcf62124dc09be85fa88f8ef
works: d517e0ff7c065fde70e098f2659a828c02f13ce2

I think at c2dc338abfbebb34dcf62124dc09be85fa88f8ef I did something like a reset and somehow got it working with this exact version, but a firstboot did not help in the device @grische had.

So it is unclear when this bug happens..

Eventually the root of the cause lays somewhere here:
https://github.com/freifunk-gluon/gluon/compare/e52409f4492ef55ba1059404b496d66858b9c86f...7426c31487cf2b5f6183742c7fcde67517243a3a
@blocktrron did suggest, that it might be related to the change in netifd, which is part of this openwrt update commit:
da10e581afa01ae3e495bec98f208433333ffdfc

That it currently works on my devices suspects that it is fixed by the latest version bumping openwrt at https://github.com/freifunk-gluon/gluon/compare/c2dc338abfbebb34dcf62124dc09be85fa88f8ef...d517e0ff7c065fde70e098f2659a828c02f13ce2
but there is nothing related to this somewhere here 065dedae1390aef0f0cffb0679cd75cd0bf80d38.

Though I currently do not have a device which has this issue.

Eventually this won't ever occur again, but as I did spend some time with this issue, I wanted to track what I know about it somewhere.

Really hard to know, as it is not easy to reproduce.

maurerle, 2023-12-30:
I do not think, that this issue should block the v2023.2 release though.

rotanid, 2023-12-31:
@maurerle was it somehow related if the node had mesh via wifi and/or mesh via vxlan and/or VPN ?

neocturne, 2023-12-31:
The address should be set from router advertisements, which are normally sent from the node itself via uradvd, in addition to the RAs from the mesh.

As the IPv6 address is required to connect to the autoupdater mirror, I believe that this issue *is* a blocker for the v2023.2 release if it is not fully solved in the latest Gluon master.

blocktrron, 2024-01-01:
I can't find any specimen of an affected device in this issue. Please provide those, otherwise further action is not possible.

neocturne, 2024-01-01:
@grische Do I understand correctly that you're seeing this issue on a recent Gluon master even after firstboot? Can you provide the images you're using?

maurerle, 2024-01-01:
I think grische saw this only on this device:
https://map.ffmuc.net/#!/en/map/3ca62f1d891d
which is running this commit: dc841ea (44 commits and 1 openwrt bump behind) (for images see here: https://github.com/freifunkMUC/site-ffm/actions/runs/7286708790?pr=339)
the latest commit on which I could reproduce this was c2dc338abfbebb34dcf62124dc09be85fa88f8ef (37 commits and 1 openwrt bump behind)

Devices on which this occured are the Cudy WR3000 v1 and FB7530
https://map.aachen.freifunk.net/#!/en/map/80afca06d558

There is currently no affected device known on current master

maurerle, 2024-01-01:
@rotanid the WR3000 which had this issue had working mesh-vpn (gluon-wireguard)
my FB7520 did only have wifi-mesh
and @grische 's FB7530 did have mesh-vpn (ffmuc-wireguard).

However, the local prefix6 uradvd address sent from the node itself, should normally work even without any gateway connection at all.
Even this did not work.

---

I updated some different devices to the latest firmware built from gluon master d517e0ff7c065fde70e098f2659a828c02f13ce2

- x86
- Netgear WAX202
- D-LINK COVR X1860
- RaspberryPi 3

I could only reproduce this on a 
- ASUS AX53U: https://map.aachen.freifunk.net/#!/en/map/50ebf6832c14

So this is still a blocker unfortunately..

neocturne, 2024-01-01:
Okay, so apparently netifd somehow does not fully configure the client interface - the interface never changes to the "up" status in `ifstatus`, and odhcp6c is not started for it.

Comparing the logread to a working node, it seems like netifd misses the "link up" event, possibly because too much is happening at the same time. Filtered for messages directly related to the status of the client bridge:

Working (tested on an x86 VM):
```
Mon Jan  1 17:50:29 2024 daemon.notice netifd: Interface 'client' is enabled
Mon Jan  1 17:50:30 2024 kern.info kernel: [    7.369688] IPv6: ADDRCONF(NETDEV_CHANGE): br-client: link becomes ready
Mon Jan  1 17:50:30 2024 daemon.notice netifd: bridge 'br-client' link is up
Mon Jan  1 17:50:30 2024 daemon.notice netifd: Interface 'client' has link connectivity
Mon Jan  1 17:50:30 2024 daemon.notice netifd: Interface 'client' is setting up now
Mon Jan  1 17:50:35 2024 daemon.notice netifd: Interface 'client' is now up
```

Broken (from @maurerle's log):
```
Thu Nov 30 07:33:38 2023 daemon.notice netifd: Interface 'client' is enabled
Thu Nov 30 07:33:38 2023 kern.info kernel: [   30.983954] IPv6: ADDRCONF(NETDEV_CHANGE): br-client: link becomes ready
```

I'm not yet sure if this hints at dropped netlink events, or an internal logic error in netifd.

neocturne, 2024-01-01:
Bisected the issue to netifd commit [516ab774cc16d4b04b3b17a067cbf2649f1adaeb](https://git.openwrt.org/?p=project/netifd.git;a=commitdiff;h=516ab774cc16d4b04b3b17a067cbf2649f1adaeb).

blocktrron, 2024-01-01:
@neocturne 

Does this mean in case of failure `IFF_UP` is not set while `IFF_LOWER_UP is not?

neocturne, 2024-01-01:
> @neocturne
> 
> Does this mean in case of failure `IFF_UP` is not set while `IFF_LOWER_UP is not?

Yes. During startup of netifd, `cb_rtnl_event` is called 3 times for br-client after the kernel reports "br-client: link becomes ready", twice with flags `0x00001002` and once with flags `0x00001003`. Checking `ip link` a moment later does show `IFF_LOWER_UP`, but it seems that no RTNL event is delivered for this change.

When manually setting `br-client` down and up again using `ip link`, eventually an event with flags `0x00011003` is delivered, and the setup succeeds.

blocktrron, 2024-01-01:
Interesting, so it worked in the past because the link-state was pulled from sysfs?

blocktrron, 2024-01-01:
It seems there is no netlink event for `NETDEV_CHANGE` event-type triggered: https://elixir.bootlin.com/linux/latest/source/net/core/rtnetlink.c#L6475

blocktrron, 2024-01-02:
Do you know where this event is emitted in the kernel? The event handler I've linked above at least does not seem to handle this case.

neocturne, 2024-01-02:
netifd patch that fixes the issue on my test device, with some explanation: https://patchwork.ozlabs.org/project/openwrt/patch/5c8c7673d95d24ac88468a8ab025fc66b9b15aee.1704207273.git.mschiffer@universe-factory.net/

Djfe, 2024-01-04:
nice, it just got accepted
https://github.com/openwrt/netifd/commit/4219e99eeec7514657f5838eb4b4b5eb28ee1271

neocturne, 2024-01-04:
Closed by #3142

closed 2024-01-04 by neocturne