freifunk-gluon/gluon #3154: ramips-mt7621: MT7915e - wifi crashes randomly

maurerle, 2024-01-09:
### Bug report

**What is the problem?**

Some devices in ramips-mt7621, (e.g. COVR-X1860, Multy WSM20, NWA50AX) are rarely affected by https://github.com/openwrt/openwrt/issues/11931 

but the devices work well for weeks most of the time.
An example of this issue can be seen here: 
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=0c0e76cf3bca&from=1704723079634&to=1704773255949
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=0c0e76cf2add&from=1699421650829&to=1701080735177
https://stats.ffmuc.net/d/hRIn3dRWk/mesh-nodes?orgId=1&var-nodeid=b8eca3e24c3f&from=1700938800000&to=1701032400000
The airtime is reported as 0 which is wrong.

A error log is found here: https://github.com/openwrt/openwrt/issues/11931#issuecomment-1828047258

End users can not connect to this broken wifi, but can still see it.

It seems that some devices are affected more often than others.
The DAP-X1860 which has the same chip but is supported within v22.03 never had this issue.

**What is the expected behaviour?**
Wifi should work reliable for devices with MT7915e driver

**Gluon Version:**
This has been seen with Wifi6 devices using the MT7915e wifi chip since Openwrt 23.05 (gluon v2023.2.x)
I have not seen this behavior on a device on v22.03 or gluon v2023.1.x yet?

**Site Configuration:**
FFAC, FFMUC

**Workarounds**

If this happens and you have SSH access through the WAN/VPN, you can run:
`rmmod mt7915e && modprobe mt7915e && wifi`
This restarts the wifi driver and fixes this issue temporarily.

maurerle, 2024-01-14:
This bug is also confirmed for the the D-Link DAP-X1860 a1 running with gluon v2023.1.1:
https://grafana.ffac.rocks/d/000000002/node?orgId=1&var-node=a42a954501f0&from=1704981600000&to=1705078800000

istrator2, 2024-05-22:
Hello everyone,

In a comparison of an **NWA50AX** with broken WiFi-driver and a running one, I think I found some "indicators of crash".

Easiest of them is to `cat /sys/kernel/debug/ieee80211/phy*/mt76/rf_regval` and if it fails, WiFi is probably broken.
[image]

So I made a cronjob with very little overhead, reloading the WiFi driver just if it's neccessary.

`cat /sys/kernel/debug/ieee80211/phy*/mt76/rf_regval || { logger -p 0 -t WiFi-Crash-Workaround "wifi seems crashed; reloading driver"; rmmod mt7915e && modprobe mt7915e && wifi; }`

Running once a minute should fix WiFi with little downtime and some dumb periodic reloading is not necessary.

I'm testing at the moment an extension of my cronjob, as first version is a little log-spammy. But I'm not sure about the error-level if (sucessfully) piped to trash.

`* * * * * { cat /sys/kernel/debug/ieee80211/phy*/mt76/rf_regval > /dev/null; } || { logger -p 0 -t WiFi-Crash-Workaround "wifi seems crashed; reloading driver"; rmmod mt7915e && modprobe mt7915e && wifi; }`

Maybe you can check this with other affected models if there is some similar behaviour.

istrator2, 2024-05-27:
...more on this.

My cronjob basically seems to work - but...

Let's have a look in your graph.
https://stats.ffmuc.net/d/hRIn3dRWk/mesh-nodes?orgId=1&var-nodeid=b8eca3e24c3f&from=1716757200000&to=1716764400000

We can see crash at ~23:15 and continued to work at ~00:50.
I did a logread to see what's going on here.

Until 23:17 everything's fine.

Sun May 26 23:18:04 first timeout message from timer.
Sun May 26 23:18:24 first time my cronjob triggers
Sun May 26 23:19:26 2024 **daemon.err rmmod: unloading the module failed**

So here's the Problem for the workaround.
Reload cannot be done at the point... -->WiFi is broken until rmmod is successfull. (Keeps trying a reload every minute)

```
Sun May 26 23:13:06 2024 daemon.info micrond[22224]: No new firmware available.
Sun May 26 23:17:41 2024 daemon.notice hostapd: client0: STA-OPMODE-N_SS-CHANGED 92:1f:f9:6d:d8:f0 1
Sun May 26 23:17:41 2024 daemon.notice hostapd: client0: STA-OPMODE-SMPS-MODE-CHANGED 92:1f:f9:6d:d8:f0 static
Sun May 26 23:17:41 2024 daemon.info hostapd: client1: STA 92:1f:f9:6d:d8:f0 IEEE 802.11: authenticated
Sun May 26 23:17:41 2024 daemon.notice hostapd: client1: STA-OPMODE-N_SS-CHANGED 92:1f:f9:6d:d8:f0 1
Sun May 26 23:17:41 2024 daemon.info hostapd: client1: STA 92:1f:f9:6d:d8:f0 IEEE 802.11: associated (aid 1)
Sun May 26 23:17:41 2024 daemon.notice hostapd: client1: AP-STA-CONNECTED 92:1f:f9:6d:d8:f0 auth_alg=open
Sun May 26 23:17:41 2024 daemon.notice hostapd: client0: Prune association for 92:1f:f9:6d:d8:f0
Sun May 26 23:17:41 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 92:1f:f9:6d:d8:f0
Sun May 26 23:17:41 2024 daemon.info hostapd: client1: STA 92:1f:f9:6d:d8:f0 RADIUS: starting accounting session E51B3F44083363D1
Sun May 26 23:18:04 2024 kern.err kernel: [2264201.675844] mt7915e 0000:02:00.0: Message 000025ed (seq 1) timeout
Sun May 26 23:18:24 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:18:24 2024 kern.err kernel: [2264222.155470] mt7915e 0000:02:00.0: Message 000102ed (seq 2) timeout
Sun May 26 23:18:45 2024 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Sun May 26 23:18:45 2024 daemon.notice hostapd: nl80211: nl80211_recv_beacons->nl_recvmsgs failed: -5
Sun May 26 23:18:45 2024 daemon.notice hostapd: client1: STA-OPMODE-SMPS-MODE-CHANGED 92:1f:f9:6d:d8:f0 off
Sun May 26 23:18:45 2024 kern.err kernel: [2264242.635087] mt7915e 0000:02:00.0: Message 0000aded (seq 3) timeout
Sun May 26 23:19:05 2024 kern.err kernel: [2264263.114711] mt7915e 0000:02:00.0: Message 00005aed (seq 4) timeout
Sun May 26 23:19:26 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:19:26 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Sun May 26 23:19:26 2024 kern.err kernel: [2264283.594359] mt7915e 0000:02:00.0: Message 000102ed (seq 5) timeout
Sun May 26 23:19:26 2024 daemon.err rmmod: unloading the module failed
Sun May 26 23:19:46 2024 kern.err kernel: [2264304.074023] mt7915e 0000:02:00.0: Message 0000aded (seq 6) timeout
Sun May 26 23:20:07 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:20:07 2024 kern.err kernel: [2264324.553596] mt7915e 0000:02:00.0: Message 000102ed (seq 7) timeout
Sun May 26 23:20:27 2024 kern.err kernel: [2264345.033277] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
Sun May 26 23:20:48 2024 kern.err kernel: [2264365.512843] mt7915e 0000:02:00.0: Message 0000aded (seq 9) timeout
Sun May 26 23:21:08 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:21:08 2024 kern.err kernel: [2264385.992492] mt7915e 0000:02:00.0: Message 000102ed (seq 10) timeout
Sun May 26 23:21:28 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:21:28 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Sun May 26 23:21:28 2024 kern.err kernel: [2264406.472100] mt7915e 0000:02:00.0: Message 000102ed (seq 11) timeout
Sun May 26 23:21:28 2024 daemon.err rmmod: unloading the module failed
Sun May 26 23:21:49 2024 kern.err kernel: [2264426.951725] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
Sun May 26 23:22:09 2024 kern.err kernel: [2264447.431355] mt7915e 0000:02:00.0: Message 0000aded (seq 13) timeout
Sun May 26 23:22:30 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:22:30 2024 kern.err kernel: [2264467.910966] mt7915e 0000:02:00.0: Message 000102ed (seq 14) timeout
Sun May 26 23:22:50 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:22:50 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Sun May 26 23:22:50 2024 kern.err kernel: [2264488.390591] mt7915e 0000:02:00.0: Message 000102ed (seq 15) timeout
Sun May 26 23:22:50 2024 daemon.err rmmod: unloading the module failed

(....)

Sun May 26 23:31:02 2024 kern.err kernel: [2264979.901724] mt7915e 0000:02:00.0: Message 000102ed (seq 9) timeout
Sun May 26 23:31:22 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:31:22 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Sun May 26 23:31:22 2024 kern.err kernel: [2265000.381347] mt7915e 0000:02:00.0: Message 000102ed (seq 10) timeout
Sun May 26 23:31:22 2024 daemon.err rmmod: unloading the module failed
Sun May 26 23:31:43 2024 kern.err kernel: [2265020.860993] mt7915e 0000:02:00.0: Message 000025ed (seq 11) timeout
[... cut ...]
Sun May 26 23:33:05 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:33:05 2024 kern.err kernel: [2265102.779502] mt7915e 0000:02:00.0: Message 000102ed (seq 15) timeout
Sun May 26 23:33:25 2024 daemon.err micrond[22224]: cat: read error: Operation timed out
Sun May 26 23:33:25 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Sun May 26 23:33:25 2024 kern.err kernel: [2265123.259156] mt7915e 0000:02:00.0: Message 000102ed (seq 1) timeout
Sun May 26 23:33:25 2024 daemon.err rmmod: unloading the module failed
(...)
```

Djfe, 2024-06-28:
maybe there needs to be a few seconds between the cat and unloading the kernelmodule?
add "sleep", maybe that fixes it?

istrator2, 2024-07-01:
I'll have a look at this idea.

I have modified my script to add an increasing delay after the _cat_.
Maybe then I can estimate how long the pause should be.

I have no hope that the sleep will help.

```
echo -n '* * * * * { DELAY_FILE="/tmp/wifi_delay"; if [ ! -f "$DELAY_FILE" ]; then echo 1 > "$DELAY_FILE"; fi; DELAY=$(cat "$DELAY_FILE"); { cat /sys/kernel/debug/ieee80211/phy*/mt76/rf_regval > /dev/null; } || { logger -p 0 -t WiFi-Crash-Workaround "wifi seems crashed; reloading driver"; ' > /usr/lib/micron.d/wifi-crash-increasing-delay-reload
echo 'logger -p 0 -t WiFi-Crash-Workaround "Delaying rmmod by $DELAY seconds"; sleep $DELAY; rmmod mt7915e && modprobe mt7915e && wifi; NEW_DELAY=$(( DELAY + 1 )); if [ "$NEW_DELAY" -gt 55 ]; then NEW_DELAY=55; fi; echo $NEW_DELAY > "$DELAY_FILE"; }; }' >> /usr/lib/micron.d/wifi-crash-increasing-delay-reload
/etc/init.d/micrond restart
```

maurerle, 2024-07-05:
As it is the `rmmod: unloading the module failed` resulting from `rmmod mt7915e`, a `wifi down` fails (`Command failed: Request timed out`), and nothing else helps - I think it is correct to reboot the device in this case.

See attached logread from a Cudy WR3000 filogic device, on which I could reproduce that restarting the firmware did not help.

<details>

```
Fri Jul  5 15:38:25 2024 daemon.info hostapd: client1: STA b6:4d:67:ea:c8:74 IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:38:25 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED f2:e6:d6:b2:8e:4b
Fri Jul  5 15:38:25 2024 daemon.info hostapd: client0: STA f2:e6:d6:b2:8e:4b IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:38:26 2024 daemon.info hostapd: client1: STA b6:4d:67:ea:c8:74 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:38:26 2024 daemon.info hostapd: client0: STA f2:e6:d6:b2:8e:4b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:38:32 2024 daemon.notice hostapd: client0: AP-STA-POLL-OK d2:25:ef:ad:44:68
Fri Jul  5 15:38:45 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 4a:4d:f7:12:55:73
Fri Jul  5 15:38:45 2024 daemon.info hostapd: client0: STA 4a:4d:f7:12:55:73 IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:38:45 2024 daemon.info hostapd: client0: STA d6:92:19:02:ff:27 IEEE 802.11: authenticated
Fri Jul  5 15:38:45 2024 daemon.info hostapd: client0: STA d6:92:19:02:ff:27 IEEE 802.11: associated (aid 2)
Fri Jul  5 15:38:45 2024 daemon.notice hostapd: client0: AP-STA-CONNECTED d6:92:19:02:ff:27 auth_alg=open
Fri Jul  5 15:38:45 2024 daemon.info hostapd: client0: STA d6:92:19:02:ff:27 RADIUS: starting accounting session FE18AB34FB36E96A
Fri Jul  5 15:38:46 2024 daemon.info hostapd: client0: STA 4a:4d:f7:12:55:73 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:38:49 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED d6:92:19:02:ff:27
Fri Jul  5 15:39:28 2024 daemon.info hostapd: client0: STA 42:05:05:b6:41:68 IEEE 802.11: authenticated
Fri Jul  5 15:39:28 2024 daemon.info hostapd: client0: STA 42:05:05:b6:41:68 IEEE 802.11: associated (aid 1)
Fri Jul  5 15:39:28 2024 daemon.notice hostapd: client0: AP-STA-CONNECTED 42:05:05:b6:41:68 auth_alg=open
Fri Jul  5 15:39:28 2024 daemon.info hostapd: client0: STA 42:05:05:b6:41:68 RADIUS: starting accounting session 2EC29E5B07E62C09
Fri Jul  5 15:39:57 2024 daemon.notice hostapd: client1: AP-STA-POLL-OK c2:70:e9:6d:83:87
Fri Jul  5 15:39:59 2024 daemon.info hostapd: client0: STA b6:58:65:37:b5:ea IEEE 802.11: authenticated
Fri Jul  5 15:39:59 2024 daemon.info hostapd: client0: STA b6:58:65:37:b5:ea IEEE 802.11: associated (aid 2)
Fri Jul  5 15:39:59 2024 daemon.notice hostapd: client0: AP-STA-CONNECTED b6:58:65:37:b5:ea auth_alg=open
Fri Jul  5 15:39:59 2024 daemon.info hostapd: client0: STA b6:58:65:37:b5:ea RADIUS: starting accounting session E4231284E26192C1
Fri Jul  5 15:40:08 2024 daemon.info hostapd: client1: STA fc:a9:f5:8f:be:b5 IEEE 802.11: authenticated
Fri Jul  5 15:40:08 2024 daemon.info hostapd: client1: STA fc:a9:f5:8f:be:b5 IEEE 802.11: associated (aid 4)
Fri Jul  5 15:40:08 2024 daemon.notice hostapd: client1: AP-STA-CONNECTED fc:a9:f5:8f:be:b5 auth_alg=open
Fri Jul  5 15:40:08 2024 daemon.info hostapd: client1: STA fc:a9:f5:8f:be:b5 RADIUS: starting accounting session BF0E1B07D317BAFF
Fri Jul  5 15:40:19 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 4a:1f:0b:6a:a2:e2
Fri Jul  5 15:40:19 2024 daemon.info hostapd: client0: STA 4a:1f:0b:6a:a2:e2 IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:40:20 2024 daemon.info hostapd: client0: STA 4a:1f:0b:6a:a2:e2 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:40:21 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED d6:72:0b:06:55:20
Fri Jul  5 15:40:21 2024 daemon.info hostapd: client0: STA d6:72:0b:06:55:20 IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:40:22 2024 daemon.info hostapd: client0: STA d6:72:0b:06:55:20 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:40:29 2024 daemon.notice hostapd: client0: AP-STA-POLL-OK 12:ad:92:87:a8:4d
Fri Jul  5 15:40:41 2024 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 7a:d0:a0:04:e3:ad
Fri Jul  5 15:40:41 2024 daemon.info hostapd: client0: STA 7a:d0:a0:04:e3:ad IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:40:42 2024 daemon.info hostapd: client0: STA 7a:d0:a0:04:e3:ad IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:40:48 2024 daemon.notice hostapd: client1: AP-STA-DISCONNECTED ea:a4:95:b9:69:2a
Fri Jul  5 15:40:48 2024 daemon.info hostapd: client1: STA ea:a4:95:b9:69:2a IEEE 802.11: disassociated due to inactivity
Fri Jul  5 15:40:49 2024 daemon.info hostapd: client1: STA ea:a4:95:b9:69:2a IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jul  5 15:40:49 2024 daemon.notice hostapd: client1: AP-STA-POLL-OK ae:4e:81:74:f1:c5
Fri Jul  5 15:40:49 2024 daemon.notice hostapd: client0: AP-STA-POLL-OK 16:c9:7c:28:e5:65
Fri Jul  5 15:40:49 2024 daemon.notice hostapd: client0: AP-STA-POLL-OK 26:f5:18:81:75:b4
Fri Jul  5 15:40:59 2024 daemon.info hostapd: client1: STA 04:d6:aa:15:5d:03 IEEE 802.11: authenticated
Fri Jul  5 15:40:59 2024 daemon.notice hostapd: client1: STA-OPMODE-N_SS-CHANGED 04:d6:aa:15:5d:03 2
[... cut ...]
Fri Jul  5 16:05:56 2024 kern.err kernel: [13505.016417] mt798x-wmac 18000000.wifi: Message 000026ed (seq 14) timeout
Fri Jul  5 16:06:16 2024 kern.err kernel: [13525.491434] mt798x-wmac 18000000.wifi: Message 00005aed (seq 15) timeout
Fri Jul  5 16:06:37 2024 kern.err kernel: [13545.966463] mt798x-wmac 18000000.wifi: Message 000102ed (seq 1) timeout
Fri Jul  5 16:06:57 2024 kern.err kernel: [13566.441479] mt798x-wmac 18000000.wifi: Message 00005aed (seq 2) timeout
Fri Jul  5 16:07:18 2024 kern.err kernel: [13586.916500] mt798x-wmac 18000000.wifi: Message 00005aed (seq 3) timeout
```

</details>

istrator2, 2024-07-05:
Yeah. I think so.

I'll make them rebooting now.
```
echo '* * * * * { cat /sys/kernel/debug/ieee80211/phy*/mt76/rf_regval > /dev/null; } || { logger -p 0 -t WiFi-Crash-Workaround "wifi seems crashed, initiate reboot now"; reboot; }' > /usr/lib/micron.d/wifi-crashed-reboot
/etc/init.d/micrond restart
```

You can have a look at the logfile.
Even at 11 Seconds delay after cat it's not working.

```
Fri Jul  5 15:39:43 2024 kern.err kernel: [319792.081221] mt7915e 0000:02:00.0: Message 000025ed (seq 3) timeout
Fri Jul  5 15:40:03 2024 kern.err kernel: [319812.560851] mt7915e 0000:02:00.0: Message 0000aded (seq 4) timeout
Fri Jul  5 15:40:24 2024 kern.err kernel: [319833.040684] mt7915e 0000:02:00.0: Message 00005aed (seq 5) timeout
Fri Jul  5 15:40:44 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:40:44 2024 kern.err kernel: [319853.520503] mt7915e 0000:02:00.0: Message 000102ed (seq 6) timeout
Fri Jul  5 15:41:05 2024 kern.err kernel: [319874.000331] mt7915e 0000:02:00.0: Message 0000aded (seq 7) timeout
Fri Jul  5 15:41:25 2024 kern.err kernel: [319894.480175] mt7915e 0000:02:00.0: Message 00005aed (seq 8) timeout
Fri Jul  5 15:41:45 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:41:46 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Fri Jul  5 15:41:46 2024 user.emerg WiFi-Crash-Workaround: Delaying rmmod by 1 seconds
Fri Jul  5 15:41:46 2024 kern.err kernel: [319914.959996] mt7915e 0000:02:00.0: Message 000102ed (seq 9) timeout
Fri Jul  5 15:41:47 2024 daemon.err rmmod: unloading the module failed
Fri Jul  5 15:42:06 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:42:06 2024 kern.err kernel: [319935.439870] mt7915e 0000:02:00.0: Message 000102ed (seq 10) timeout
Fri Jul  5 15:42:26 2024 kern.err kernel: [319955.919613] mt7915e 0000:02:00.0: Message 0000aded (seq 11) timeout
Fri Jul  5 15:42:47 2024 kern.err kernel: [319976.399432] mt7915e 0000:02:00.0: Message 00005aed (seq 12) timeout
Fri Jul  5 15:43:07 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:43:07 2024 kern.err kernel: [319996.879262] mt7915e 0000:02:00.0: Message 000102ed (seq 13) timeout
Fri Jul  5 15:43:28 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:43:28 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Fri Jul  5 15:43:28 2024 user.emerg WiFi-Crash-Workaround: Delaying rmmod by 1 seconds
Fri Jul  5 15:43:28 2024 kern.err kernel: [320017.359090] mt7915e 0000:02:00.0: Message 000102ed (seq 14) timeout
Fri Jul  5 15:43:29 2024 daemon.err rmmod: unloading the module failed
Fri Jul  5 15:43:48 2024 kern.err kernel: [320037.838905] mt7915e 0000:02:00.0: Message 0000aded (seq 15) timeout
Fri Jul  5 15:44:09 2024 kern.err kernel: [320058.318723] mt7915e 0000:02:00.0: Message 00005aed (seq 1) timeout
Fri Jul  5 15:44:29 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:44:29 2024 kern.err kernel: [320078.798549] mt7915e 0000:02:00.0: Message 000102ed (seq 2) timeout
Fri Jul  5 15:44:35 2024 user.notice ipv6_ra_filter: Failed to get batman gateway. Dropping any router advertisement.
Fri Jul  5 15:44:50 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:44:50 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Fri Jul  5 15:44:50 2024 user.emerg WiFi-Crash-Workaround: Delaying rmmod by 2 seconds
Fri Jul  5 15:44:50 2024 kern.err kernel: [320099.278361] mt7915e 0000:02:00.0: Message 000102ed (seq 3) timeout
Fri Jul  5 15:44:52 2024 daemon.err rmmod: unloading the module failed
Fri Jul  5 15:45:10 2024 kern.err kernel: [320119.758181] mt7915e 0000:02:00.0: Message 0000aded (seq 4) timeout
Fri Jul  5 15:45:18 2024 user.notice ipv6_ra_filter: Failed to get batman gateway. Dropping any router advertisement.
Fri Jul  5 15:45:31 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:45:31 2024 kern.err kernel: [320140.238020] mt7915e 0000:02:00.0: Message 000102ed (seq 5) timeout
Fri Jul  5 15:45:51 2024 kern.err kernel: [320160.717829] mt7915e 0000:02:00.0: Message 00005aed (seq 6) timeout
Fri Jul  5 15:46:01 2024 user.notice ipv6_ra_filter: Failed to get batman gateway. Dropping any router advertisement.
Fri Jul  5 15:46:12 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:46:12 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Fri Jul  5 15:46:12 2024 user.emerg WiFi-Crash-Workaround: Delaying rmmod by 2 seconds
Fri Jul  5 15:46:12 2024 kern.err kernel: [320181.197662] mt7915e 0000:02:00.0: Message 000102ed (seq 7) timeout
Fri Jul  5 15:46:14 2024 daemon.err rmmod: unloading the module failed
Fri Jul  5 15:46:32 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:46:32 2024 kern.err kernel: [320201.677467] mt7915e 0000:02:00.0: Message 000102ed (seq 8) timeout
Fri Jul  5 15:46:53 2024 kern.err kernel: [320222.157296] mt7915e 0000:02:00.0: Message 0000aded (seq 9) timeout
Fri Jul  5 15:47:13 2024 daemon.err micrond[5034]: cat: read error: Operation timed out
Fri Jul  5 15:47:13 2024 user.emerg WiFi-Crash-Workaround: wifi seems crashed; reloading driver
Fri Jul  5 15:47:13 2024 user.emerg WiFi-Crash-Workaround: Delaying rmmod by 2 seconds
[... cut ...]
Fri Jul  5 16:19:59 2024 kern.warn kernel: [322208.700750] [<83315a4c>] ___ieee80211_stop_tx_ba_session+0x2ac/0x2cc [mac80211]
Fri Jul  5 16:19:59 2024 kern.warn kernel: [322208.700870] [<83313ff0>] ieee80211_sta_tear_down_BA_sessions+0x88/0x158 [mac80211]
Fri Jul  5 16:19:59 2024 kern.warn kernel: [322208.700978] [<83307244>] ieee80211_sta_block_awake+0x8e8/0x1bd8 [mac80211]
Fri Jul  5 16:19:59 2024 kern.warn kernel: [322208.701085]
Fri Jul  5 16:19:59 2024 kern.warn kernel: [322208.701091] ---[ end trace b612ebfc7f27f777 ]---
Fri Jul  5 16:20:00 2024 daemon.err micrond[5034]: autoupdater is disabled
```

istrator2, 2024-07-10:
Short Update on my reboot-conjob.

If you look at my log (it's remote; not node-logread)
I guess crash happened very soon after first mt7915e timeout message, because no more were sent after that.

You can see here the little gap.
So there's every few days about ~4 minutes no WiFi.

https://stats.ffmuc.net/d/hRIn3dRWk/mesh-nodes?orgId=1&var-nodeid=b8eca3e24c3f&from=1720586659214&to=1720593578105

```
2024-07-10T04:16:00+02:00 FreifunkWertingenistraAP02 micrond[2394]: autoupdater is disabled
2024-07-10T05:16:00+02:00 FreifunkWertingenistraAP02 micrond[2394]: autoupdater is disabled
2024-07-10T06:16:00+02:00 FreifunkWertingenistraAP02 micrond[2394]: autoupdater is disabled
2024-07-10T06:36:17+02:00 FreifunkWertingenistraAP02 ipv6_ra_filter: Successfully set IPv6 gateway to f2:00:27:14:00:00.
2024-07-10T06:37:00+02:00 FreifunkWertingenistraAP02 ipv6_ra_filter: Successfully set IPv6 gateway to f2:00:26:14:00:00.
2024-07-10T07:15:11+02:00 FreifunkWertingenistraAP02 hostapd: client1: STA-OPMODE-N_SS-CHANGED 92:1f:f9:6d:d8:f0 1
2024-07-10T07:15:11+02:00 FreifunkWertingenistraAP02 hostapd: client1: STA-OPMODE-SMPS-MODE-CHANGED 92:1f:f9:6d:d8:f0 static
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client0: STA 92:1f:f9:6d:d8:f0 IEEE 802.11: authenticated
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client0: STA 92:1f:f9:6d:d8:f0 IEEE 802.11: associated (aid 1)
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client0: AP-STA-CONNECTED 92:1f:f9:6d:d8:f0 auth_alg=open
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client1: Prune association for 92:1f:f9:6d:d8:f0
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client1: AP-STA-DISCONNECTED 92:1f:f9:6d:d8:f0
2024-07-10T07:15:17+02:00 FreifunkWertingenistraAP02 hostapd: client0: STA 92:1f:f9:6d:d8:f0 RADIUS: starting accounting session B09ACBBF26410A57
2024-07-10T07:15:39+02:00 FreifunkWertingenistraAP02 kernel: [300343.323595] mt7915e 0000:02:00.0: Message 000025ed (seq 9) timeout
2024-06-21T20:16:23+02:00 FreifunkWertingenistraAP02 logread[1144]: Logread connected to ffwer-jumphost.istranet.org:514 via tcp
2024-07-10T07:23:43+02:00 FreifunkWertingenistraAP02 firewall: Reloading firewall due to ifupdate of client (br-client)
2024-07-10T08:05:09+02:00 FreifunkWertingenistraAP02 ipv6_ra_filter: Successfully set IPv6 gateway to f2:00:27:14:00:00.
2024-07-10T08:06:33+02:00 FreifunkWertingenistraAP02 ipv6_ra_filter: Successfully set IPv6 gateway to f2:00:26:14:00:00.
2024-07-10T08:16:00+02:00 FreifunkWertingenistraAP02 micrond[2404]: autoupdater is disabled
```

herbetom, 2024-11-23:
Using #3370 (OpenWrt 24.10, to be eaxct [1bbea11](https://github.com/freifunk-gluon/gluon/commit/1bbea11e51a633c8fe45879a2aa7de0107ec7eb5)) on NWA55AXE (ramips-mt7621) i think there is an improvement.

Now the log looks as following:

```
[215935.084038] mt7915e 0000:02:00.0: Retry message 000025ed (seq 10)
[215937.163959] mt7915e 0000:02:00.0: Message 000025ed (seq 10) timeout
[215957.963708] mt7915e 0000:02:00.0: Retry message 00000010 (seq 13)
[215958.055519] mt7915e 0000:02:00.0: HW/SW Version: 0x8a108a10, Build Time: 20240429200716a
[215958.055519] 
[215978.443383] mt7915e 0000:02:00.0: Retry message 00000005 (seq 14)
[215998.933107] mt7915e 0000:02:00.0: Retry message 00000001 (seq 1)
[216019.402870] mt7915e 0000:02:00.0: Message 00000001 (seq 1) timeout
[216019.402949] mt7915e 0000:02:00.0: Download request failed
[216019.402987] mt7915e 0000:02:00.0: Failed to release patch semaphore
[216019.562101] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216019.712002] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216019.861904] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.011911] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.153111] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.301876] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.452064] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.601891] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.752086] mt7915e 0000:02:00.0: Failed to get patch semaphore
[216020.752171] mt7915e 0000:02:00.0: chip full reset failed
[216020.752208] ieee80211 phy0: Hardware restart was requested
[216020.752263] ieee80211 phy1: Hardware restart was requested
```

Full dmesg of two nodes at two locations running into it: https://gist.github.com/herbetom/cc708b05398e2a521361ae79490f5cfc

I'm not seeing clients afterwards so the requested hardware restart presumably fails somehow. But should be an improvement nonetheless and something that can hopefully be improved upon.

Djfe, 2024-11-23:
yep, nbd improved everything so downtime is only 10 seconds or so, if the mcu crashes/requires a recovery.
Also he fixed roaming between 2.4 and 5ghz on the same AP for dbdc APs. (there was an issue)

I already saw the issue you just experienced and have reported it to nbd last month. Last time I checked he doesn't have the time to look further into it.
but at least in October he found the time to already improve this a lot. even though the firmware still crashes it usually recovers on it's own now.
though if it doesn't like here that means the device is down if it's a wifi mesh node

rotanid, 2025-01-10:
@maurerle @istrator2 @Djfe @herbetom is this issue still relevant with the latest mt76 updates in gluon main (openwrt 24.10) ?

herbetom, 2025-01-10:
Just updated an NWA55AXE ([meshviewer](https://meshviewer.darmstadt.freifunk.net/#!/de/map/b8eca3e12ec0)) to a current enough gluon commit ([site-commit](https://github.com/freifunk-darmstadt/site-ffda/commit/e7ed0183f2af33188c6188cb1fd72e30841f2c69#diff-81ebb07af8366299fb17b327fbc00d8b3662f582fec2856fb041980a924f65acR5) | [gluon-commit](https://github.com/freifunk-gluon/gluon/commits/a475b1919cead4c7d476a321303e6610ad879157)) to answer that question.

We'll see how it behaves :)

Djfe, 2025-01-10:
@rotanid we are currently evaluating.
This might take 1-2 weeks

AiyionPrime, 2025-03-05:
Any results on this? :)

herbetom, 2025-03-05:
The test from January 10th wasn't successful. It still broke after some time.

Haven't yet build firmware with #3436 but that's the next step.

Djfe, 2025-03-05:
yeah, it's definitely better with this and it felt like the device recovered every mcu timeout on it's own.

What's left is the other issue where Pings go to several seconds (packets are very delayed) and opening fdac::ac straight out fails.

The last thing blocktrron looked at was tx packets getting stuck and figuring out why (this was specific to mt7981 and maybe other filogic devices). This lead to crippling TQ values on wifi mesh (one of the symptoms)

I'm waiting for the next update of commits to test (at some point).

This PR (to me) feels ready to be upstreamed but it's not my call.

rotanid, 2025-03-09:
please test with gluon master branch again now that #3436 is merged

Djfe, 2025-03-09:
I can deny that this patch fixes said issues. but it improves other parts of filogic too.
#3436 is about fixing the code that automatically restarts the mcu (wifi chip) on crashes. There was for example one issue where the restart routine can't be completed due to a semaphore. Also the PR adjusts certain timeouts to be more ideal.

This issue on filogic still remains and has a different cause.
We tested the branch of #3436 and a related branch for debugging purposes in January. The cause of this issue might be related to tx packets getting stuck, but we don't have a full diagnosis, yet.

maurerle, 2025-04-15:
I can confirm using gluon master for a while on my d-link covr-x1860-a1 did never have this problem again, even without the `ffac-mt7915-hotfix`.
This is not needed anymore with the latest gluon main

closed 2025-04-15 by maurerle

stweil, 2025-04-16:
> Some devices in ramips-mt7621, (e.g. COVR-X1860, Multy WSM20, NWA50AX, NWA55AXE) are rarely affected by [openwrt/openwrt#11931](https://github.com/openwrt/openwrt/issues/11931)

[Our NWA50AX](https://stats.ffrn.de/d/000000002/nodes?var-node=b8eca3e0f050&theme=light&orgId=1&from=now-1y&to=now&timezone=browser&var-hostname=Ladenburg-INTAKT-001) is not so rarely affected, but had this issue very often in the past. @herbetom, I am curious how new firmware with the mentioned fix will change the situation.

Djfe, 2025-04-16:
agreed. I asked nbd168 on IRC whether it's possible to backport master mt76 to OpenWrt-23.05 so we could also get all fixes and new firmware in the (soon) old stable release, too.

Else it might be a nice addition to backport all necessary commits/fixes by him from October, December and January and blocktrron's fixes, too. But that would require more work.
Gluon v2025.1 will be released with all fixes included 👌

#3483 might be a reason to not roll out current master on your affected APs just yet.

herbetom, 2025-04-16:
@stweil The mentionend Node is using 2.3.x-20250325. That's a FFRN nightly Firmware which has #3436 already included:

https://github.com/Freifunk-Rhein-Neckar/site-ffrn/blob/v2.3.x-20250325/.github/build-info.json#L5

https://github.com/freifunk-gluon/gluon/commits/28f2ee9a923bd9253ca2663d551e585590da2561/

It was automatically installed at 2025-03-27 04:32. Since then the device had only one reboot. Of course i don't know if it was an internal (e.g. software crashing) or external (e.g. power) reason.
But i woulndn't consider it that bad even if it was an internal reason as it doesn't look like clients lost the connection for a long period of time. The important part in my eyes would be that it automatically corrected itself. Reducing the impact is of course good and in an ideal world a crash which results in an reboot wouldn't happen. But i'm not too worried at the moment and to me it currently looks good from afar.