freifunk-gluon/gluon #3165: 5GHz outdoor operation 40MHz - acs error

Djfe, 2024-01-18:
### Bug report

**What is the problem?**
I can't get this router to run on 5GHz in outdoor mode (40MHz)
http://[2a03:2260:3006:112:a2a3:f0ff:fe48:9b70]/cgi-bin/status
https://pastebin.com/CLkyT79s

while this one works fine with the same config:
http://[2a03:2260:3006:112:a62a:95ff:fe45:1f0]/cgi-bin/status

I only just set this new config after updating the routers to our newest release.
("hinten" was installed with a pre-release 2023.2 while the working one "theke" was just updated from 2023.1.1 to 2023.2). If it actually matters which versions they ran before exactly, then I'll look it up for you.

**Part of the problem**
ACS errors in logread (see pastebin for the log excerpt):
DFS start_dfs_cac() fails
Is this a bug or am I using it wrong?

I'm not sure whether it's got anything to do with it, but we disabled channels 120-128 in Aachen. (see site.conf linked below)
Does that prevent hostapd from setting this frequency which then prevents the dfs scan on the randomly chosen frequency?

**Gluon Version:**
branch 2023.2 https://github.com/freifunk-gluon/gluon/commit/bfbefa4580d1162b2766c6e685d85b970d0c9680
v2023.2-11-gbfbefa45

**Site Configuration:**
https://github.com/ffac/site/blob/v2023.2.x/site.conf#L61

**Custom patches:**
https://github.com/ffac/site/tree/v2023.2.x/patches

blocktrron, 2024-01-18:
>  I can't get this router to run on 5GHz in outdoor mode (40MHz)

Please provide more information. This device is not reachable

>  If it actually matters which versions they ran before exactly, then I'll look it up for you.

Doesn't matter

> Is this a bug or am I using it wrong?

Depends on the configuration. Which is the most important information missing in this ticket after the exact model of the non-working device.

> Does that prevent hostapd from setting this frequency which then prevents the dfs scan on the randomly chosen frequency?

Highly doubt that. If it tries to start a CAC, it has already selected channels in hostapd, while the request for starting the CAC either fails in mac80211 or the driver.

blocktrron, 2024-01-18:
Fix in #3164

Djfe, 2024-01-19:
the device is installed in a bar and the power is cut daily by the circuit breaker. It should be online again by noon.
Thanks for already figuring out that there is a fix, we will roll it out once the pr is merged.

Djfe, 2024-01-19:
*Thanks for implementing a fix so fast ❤️

Djfe, 2024-01-19:
Here are the links to our map I should've posted instead
affected was: https://map.aachen.freifunk.net/#!/de/map/a0a3f0489b70
working: https://map.aachen.freifunk.net/#!/de/map/a42a954501f0

closed 2024-01-19 by blocktrron

Djfe, 2024-01-19:
I just flashed the version of your branch to the affected device and the problem was solved 👍

PR #3164 (2024-01-19): modules: update to latest HEAD
Fixes Ethernet RX issues on MediaTek ARM SOCs.

Closes #3165