freifunk-gluon/gluon #3173: v2023.1.x: FUTRO running with x86-generic image does not run wgpeerselector after some boots

maurerle, 2024-01-26:
While migrating our FFAC domain, we have seen some futros behave weird.
Some did not come online, while most did not have any problem.
In total we have about 100 futros and more than 70 are using the x86-generic image (even though the hardware would support x86-64 if i am not mistaken)

I could reproduce this issue on two FUJITSU SIEMENS FUTRO S550 - and suspect that a handful of our devices are offline because of this - and I don't know how many devices would die after a reboot.

## Problem:

After booting, the `wgpeerselector` is not started - so something with this invocation did not work:
https://github.com/ffac/packages/blob/master/net/wgpeerselector/files/lib/netifd/proto/wgpeerselector.sh

Therefore, no mesh-vpn is established.
On some firmwares, it is the boot directly after a sysupgrade, while on others, it occurs after rebooting the first time.

I tried to run `wgpeerselector -i wg_mesh --group gluon-mesh-vpn -v` on a broken node, which did not help.

## Fixes
Flash the x86-64 image for your device on top.

use an image built from this commit or earlier: 13a6617f2a90a1b15febe23674a8f4c7092e00a1

## Affected versions

1b3b12103cdca576ed263f3e2163b3539f293612
v2023.1 (the tag)
up to
1b3b12103cdca576ed263f3e2163b3539f293612 (nearly v2023.1.2)

I did not test v2023.2 yet

maurerle, 2024-01-29:
In FFAC we updated all x86-generic images with a x86-64 image, which worked well, as the hardware is generelly supported.
Though I would not recommend this step generally, as it will brick true i386 devices if there are any.
We did not seem to have any.

This issue will not be relevant anymore when v2023.1.x runs out of support (probably even if the v2023.1.x branch gets a kernel-bump again) and I don't mind closing it.

But if someone wants to keep this open, just reopen it.

maurerle, 2024-01-29:
I have just seen this issue also occur on a v2023.2.x firmware built bfbefa4580d1162b2766c6e685d85b970d0c9680 when testing..

This is not far behind v2023.2.x
https://github.com/freifunk-gluon/gluon/compare/bfbefa4580d1162b2766c6e685d85b970d0c9680...v2023.2.x

neocturne, 2025-03-30:
Hmm, bfbefa4580d1162b2766c6e685d85b970d0c9680 should have included the netifd bugfix that I would have expected to solve this. Can this still be reproduced on current Gluon? We'll need more information on the netifd state (`ifstatus` on the wgpeerselector interface)

maurerle, 2025-04-15:
We migrated our futro devices away from x86-generic to x86-64, I can not reproduce this issue therefore.
Maybe someone from FFH can step in, as they also use wgpeerselector and recently migrated various offloaders @AiyionPrime

AiyionPrime, 2025-06-21:
I do not own any FUTRO devices and worked on getting rid of all of them in favor of more modern hardware.

It appears I was sucessful, there are merely two devices left; sorry :/

closed 2025-11-18 by maurerle